MichaIng/DietPi #3576: Update to 6.30 failed (from 29.2)

Paulisper, 2020-05-30:
Sorry for posting this.
My Pi 3 could be updated to 6.30 without any flaw.
But my Pi 2 failed.
I am posting this because there was the request at the end of the failed update.
Thank you so much for coding and developing dietpi.
Kind regards,
Urban

#### Details:
- Date           | Sat May 30 17:52:14 CEST 2020
- DietPi version | v6.29.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi 2 Model B (armv7l) (ID=2)
- Kernel version | Linux DietPi7 4.19.66-v7+ #1253 SMP Thu Aug 15 11:49:46 BST 2019 armv7l GNU/Linux
- Distro         | stretch (ID=4)
- Command        | apt-get -q update
- Exit code      | 100
- Software title | DietPi-Update
#### Steps to reproduce:

1. dietpi-update
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- c below
#### Extra details:

- self explaining, I guess. Sorry for not searching for a solution. 
I went back to 6.29.2,
#### Additional logs:
```
Hit:1 http://raspbian.raspberrypi.org/raspbian stretch InRelease
Ign:2 https://archive.raspberrypi.org/debian stretch InRelease
Err:3 https://archive.raspberrypi.org/debian stretch Release
  server certificate verification failed. CAfile: /etc/ssl/certs/ca-certificates.crt CRLfile: none
Reading package lists...
E: The repository 'https://archive.raspberrypi.org/debian stretch Release' does no longer have a Release file.
```

Joulinar, 2020-05-30:
Hi,

Many thanks for you message. Yes this is known but nothing we can do against it. 

See #3575

Joulinar, 2020-05-30:
issue fixed

closed 2020-05-30 by Joulinar