MichaIng/DietPi #3578: DietPi-Imager | Loop device support for Clonezilla

MichaIng, 2020-06-01:
@sal666 
FTR: https://gitlab.com/stevenshiau/clonezilla/-/merge_requests/43
When allowing loop devices for Clonezilla in DietPi-Imager, it runs through nicely with this Clonezilla function additions: https://dietpi.com/downloads/images/testing/DietPi_NativePC-UEFI-x86_64-Buster.7z
I'm doing a test install on UEFI-capable VM with this, finally on native PC. This allows automated builds without involving any real machine or disk.

MichaIng, 2020-06-01:
🈴 Restoring fails due to "invalid partition table format".
This is strange since the mentioned file clearly states "gpt":
```
Model: Loopback device (loopback)
Disk /dev/loop0: 2299904s
Sector size (logical/physical): 512B/512B
Partition Table: gpt
Disk Flags: 

Number  Start    End       Size      File system  Name  Flags
 1      2048s    194559s   192512s   fat32              boot, esp
 2      194560s  2297928s  2103369s  ext4
```
- Probably the restoring functions as well need some rework to accept "loop[0-9]" here.

MichaIng, 2020-06-01:
🈯 Bingo! Applying the change to the live disk as well solves the issue.

MichaIng, 2020-06-11:
Looks good, might be merged after the coming release: https://gitlab.com/stevenshiau/clonezilla/-/merge_requests/43#note_358529644

MichaIng, 2020-09-14:
Self-reminder:
- Create new UEFI installer image and by this test the new Clonezilla version with integrated loop device support: https://sourceforge.net/projects/clonezilla/files/clonezilla_live_alternative_testing/20200910-focal/
- Many thanks to @stevenshiau for merging this quite quickly and even providing an early test image. Sorry that I was not able to test this until now. I'll do later with the current "official" testing image and provide feedback.

MichaIng, 2020-09-17:
🈯 Works like a charm, any `x86_64` image can be reliably directly created a Clonezilla installer with, awesome!

sal666, 2020-09-17:
Awesome!! Great work!!

MichaIng, 2020-10-17:
Updated milestone, just to keep track until the Clonezilla update has been released.

MichaIng, 2020-11-11:
Release is there: https://sourceforge.net/projects/clonezilla/files/clonezilla_live_alternative/20201102-groovy/

closed 2020-11-11 by MichaIng