MichaIng/DietPi #3579: no network on Pine64 LAN at startup

pcossu61, 2020-06-02:
#### Details:
- Date           | Tue  2 Jun 21:09:00 CEST 2020
- Bug report     | 257e2287-0cd2-4ba3-b68a-e17083da833f
- DietPi version | v6.28.0 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Armbian
- SBC device     | Pine A64 (aarch64) (ID=40)
- Kernel version | Linux DietPi 5.4.7-sunxi64 #19.11.6 SMP Sat Jan 4 19:40:10 CET 2020 aarch64 GNU/Linux
- Distro         | buster (ID=5)
- Command        | ping -c 1 -W 5 1.1.1.1
- Exit code      | 1
- Software title | DietPi-Login

#### Steps to reproduce:

1. Dietpi on login gives the indicated error
2. Configuring the devices everything works but soon afterwards same error

#### Expected behaviour:

- i need the Pine64 as server with internet access, at startup acces should be available immediately

#### Actual behaviour:

- no connection

#### Extra details:

- no clue

#### Additional logs:
```
PING 1.1.1.1 (1.1.1.1) 56(84) bytes of data.

--- 1.1.1.1 ping statistics ---
1 packets transmitted, 0 received, 100% packet loss, time 0ms
```

Joulinar, 2020-06-02:
Hi,

many thanks for your report. Can you ping anything else on the internet? Or does it fails on `ping 1.1.1.1` only

pcossu61, 2020-06-03:
Hello

I tried with 1.1.1.1 and 8.8.8.8 with no success

Opendns 208.67.222.222 instead is working... !

[FAILED] DietPi-Login | Unable to continue, DietPi-Login will now terminate.
root@DietPiBusterweewx:~# ping -c 1 -W 5 8.8.8.8
PING 8.8.8.8 (8.8.8.8) 56(84) bytes of data.

--- 8.8.8.8 ping statistics ---
1 packets transmitted, 0 received, 100% packet loss, time 0ms

root@DietPiBusterweewx:~# ping -c 1 -W 5 1.1.1.1
PING 1.1.1.1 (1.1.1.1) 56(84) bytes of data.

--- 1.1.1.1 ping statistics ---
1 packets transmitted, 0 received, 100% packet loss, time 0ms

root@DietPiBusterweewx:~# ping -c 1 -W 5 208.67.222.222
PING 208.67.222.222 (208.67.222.222) 56(84) bytes of data.
64 bytes from 208.67.222.222: icmp_seq=1 ttl=57 time=53.1 ms

--- 208.67.222.222 ping statistics ---
1 packets transmitted, 1 received, 0% packet loss, time 0ms
rtt min/avg/max/mdev = 53.143/53.143/53.143/0.000 ms

Joulinar, 2020-06-03:
hmm could be Google DNS as well as Cloudflare DNS are blocked on your network or by your ISP. But it seems Cisco DNS is working. Means, it might be needed to adjust connectivity checks as follow:

Do `nano /DietPi/dietpi.txt` and search for the following 2 lines:

```
# General connection and DNS testing
# - IP to ping when checking network connectivity. Default: 1.1.1.1 (Cloudflare DNS, should be very fast world-wide)
CONFIG_CHECK_CONNECTION_IP=1.1.1.1
# - Domain to ping when checking DNS resolver. Default: one.one.one.one (Cloudflare DNS domain, see above)
CONFIG_CHECK_DNS_DOMAIN=one.one.one.one
```

you can change it to `208.67.222.222` and `resolver1.opendns.com`

MichaIng, 2020-06-24:
I mark this as closed. Feel free to reopen if issue is still present.

closed 2020-06-24 by MichaIng