MichaIng/DietPi #3580: default smb.conf :  create mask = 0775 --> create mask = 0664

maartenlangeveld, 2020-06-03:
When installing samba through dietpi-software `/etc/samba/smb.conf` config file is created.
It contains `create mask = 0775` where I guess `create mask = 0664` fit better.

current smb.conf mask entries
```
        create mask = 0775
        directory mask = 0775
```

preferred smb.conf mask entries
```
        create mask = 0664
        directory mask = 0775
```

maartenlangeveld, 2020-06-03:
It only applies to new files created on the client... but since I have change the smb.conf config file myself it''s not a big deal to change the default values.

MichaIng, 2020-06-03:
Ah yes I was not thinking through. Since this is only about the server-side permissions, when required one can add execute permissions manually per-file anyway. Only for the client-side it is an argument to keep execute permissions by default since one cannot change per-file permissions on a samba mount. So your suggestion makes sense 👍.

MichaIng, 2020-06-03:
Done: https://github.com/MichaIng/DietPi/commit/d9daf32ccb479755e23596098b661f1dca2a277a
Changelog: https://github.com/MichaIng/DietPi/commit/7c4284b5a285000112c86f4a7577b17cca7e80a4

closed 2020-06-04 by maartenlangeveld