MichaIng/DietPi #3584: Checking DNS resolver issues

yslupdates, 2020-06-05:
Hey I'm having issues setting up my Pi Zero W on DietPi. I flash it and immediately plug into my Pi, ethernet thru micro usb. Nothing configured in dietpi.txt so defaults. I was able to ping other sites just fine, the connection seems to be online but I just can't get past the Checking DNS resolver part.

```
#### Details:
- Date           | Fri  5 Jun 06:50:37 BST 2020
- Bug report     | 17dbc259-84d6-4544-b1fa-c315bb9bd44e
- DietPi version | v6.28.0 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi Zero W (armv6l) (ID=1)
- Kernel version | Linux DietPi 4.19.75+ #1270 Tue Sep 24 18:38:54 BST 2019 armv6l GNU/Linux
- Distro         | buster (ID=5)
- Command        | ping -c 1 -W 5 one.one.one.one
- Exit code      | 1
- Software title | DietPi-Login

#### Steps to reproduce:
<!-- Explain how to reproduce the issue -->
1. ... Flash new image of DietPi
2. ... Plug in and setup thru SSH

#### Expected behaviour:
<!-- What SHOULD happen? -->
- ... DNS resolver works

#### Actual behaviour:
<!-- What IS happening? -->
- ... DNS resolve isn't working

#### Extra details:
<!-- Please post any extra details that might help solve the issue -->
- ...

#### Additional logs:
```
PING one.one.one.one(one.one.one.one (2606:4700:4700::1111)) 56 data bytes

--- one.one.one.one ping statistics ---
1 packets transmitted, 0 received, 100% packet loss, time 0ms
```

```

Joulinar, 2020-06-05:
Hi,

many thanks for your report. Pls can you try to deactivate IPv6 if possible?

StephanStS, 2020-06-05:
@yslupdates: Is there any Pi-Hole present in your setup? If yes, check Pi-Hole for too strong filtering or disable Pi-Hole during the first run.

yslupdates, 2020-06-05:
Turning off IPv6 in dietpi-config worked! Can turn it back on after I’m done though?

And yes, I do have a 2nd pihole running but I disabled it before installing to no avail!

yslupdates, 2020-06-06:
Quick update, I tried doing dietpi installation over wifi and that worked smoothly. No DNS resolver checking issues on wifi. I just did that, then plugged in ethernet and switched and then setup static IP just fine. Although I did have the same issue, but now with trying to run dietpi-software. I can install pi-hole the normal bash script way they recommend, but I can't access dietpi-software. I was able to change the command from saying one.one.one.one to 1.1.1.1 and that worked. I've disabled my other pi-hole this whole time.

`│ Checking DNS resolver                                                        │
│  - Command: ping -c 1 -W 5 one.one.one.one                                   │
│  - Exit code: 1                                                              │
│  - DietPi version: v6.30.0 (MichaIng/master) | HW_MODEL: 1 | HW_ARCH: 1 |    │
│ DISTRO: 5                                                                    │
│  - Image creator: DietPi Core Team                                           │
│  - Pre-image: Raspbian Lite                                                  │
│  - Error log:                                                                │
│ PING one.one.one.one(one.one.one.one (2606:4700:4700::1001)) 56 data bytes   │
│                                                                              │
│ --- one.one.one.one ping statistics ---                                      │
│ 1 packets transmitted, 0 received, 100% packet loss, time 0ms `

Joulinar, 2020-06-06:
Hi,

I guess your issue is with IPv6. Seems you have issues connection to the IPv6 address of Cloudflare. You could check if other IPv6 address are working fine.

faxesystem, 2020-06-06:
Having the same problem. My pi is connected via ethernet. I can ping one.one.one.one from all other systems in my network without problems and get the IPv6 address resolved. But dietpi seems to have a problem with it.

MichaIng, 2020-06-24:
I mark this issue as closed. Feel free to reopen if required.

closed 2020-06-24 by MichaIng

Ben-L-E, 2021-04-26:
I also encounter this issue on a fresh install because https://cleanbrowsing.org/ blocks dns9.quad9.net on our test network:

```
[FAILED] DietPi-Software | Checking DNS resolver
 - Command: ping -c 1 -W 10 dns9.quad9.net
```

Would it be possible to replace or add a family-friendly backup test, e.g. `ping -c 1 -W 10 doh.familyshield.opendns.com` ?

Joulinar, 2021-04-26:
You can change the DNS as well as ping check to a value fitting your need. This is possible inside `dietpi.txt`