MichaIng/DietPi #3585: Update impossible to 6.30

AlexanderShniperson, 2020-06-05:
### Creating a bug report/issue

after run `dietpi-update` i see
```
- Command: G_AGI gpac motion                                                           
│  - Exit code: 100                                                                       
│  - DietPi version: v6.28.0 (MichaIng/master) | HW_MODEL:1 | HW_ARCH:1 | DISTRO:5        
│  - Image creator: DietPi Core Team                                                      
│  - Pre-image: Raspbian Lite                                                             
│                                                                                         
│ Log file content:                                                                       
│ E: Failed to fetch                                                                      
│ http://mirror.yandex.ru/mirrors/raspbian/raspbian/pool/main/p/pango1.0/libpango-1.0-0_  
│ 1.42.4-8~deb10u1_armhf.deb  404  Not Found [IP: 213.180.204.183 80]                     
│ E: Failed to fetch                                                                      
│ http://mirror.yandex.ru/mirrors/raspbian/raspbian/pool/main/p/pango1.0/libpangoft2-1.0  
│ -0_1.42.4-8~deb10u1_armhf.deb  404  Not Found [IP: 213.180.204.183 80]                  
│ E: Failed to fetch 
│ http://mirror.yandex.ru/mirrors/raspbian/raspbian/pool/main/p/pango1.0/libpangocairo-1  
│ .0-0_1.42.4-8~deb10u1_armhf.deb  404  Not Found [IP: 213.180.204.183 80]                
│ E: Failed to fetch                                                                      
│ http://mirror.yandex.ru/mirrors/raspbian/raspbian/pool/main/libs/libssh/libssh-gcrypt-  
│ 4_0.8.7-1+deb10u1_armhf.deb  404  Not Found [IP: 213.180.204.183 80]                    
│ E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?   
│                                                                                         
│ If problems persist, please report at https://github.com/MichaIng/DietPi/issues for     
│ investigation.                                                                          
│                                                                                         
│ Unable to continue, DietPi-Software will now terminate.                                 
│    
```

Joulinar, 2020-06-05:
Hi,

many thanks for your report. Looks like you have issues fetching some software packages from a 3rd party vendor. Pls can you try running `apt update && apt upgrade` manually.

AlexanderShniperson, 2020-06-09:
i don't know what was done different, but i restored my image and did `dietpi-update` and this error gone
issue should be closed, thanks for reply

Joulinar, 2020-06-09:
Many thanks for your feedback. Maybe a temp issue at the 3rd party vendor side. I will go to close the topic. Feel free to reopen if needed.

closed 2020-06-09 by MichaIng