MichaIng/DietPi #3586: Rpi 4 4Gb | Dietpi v6.30 with Buster | LxQt Desktop not work with "Dietpi" user

Pain-Patate, 2020-06-06:
### Creating a bug report/issue

#### Required Information
- DietPi version | `6.30` | It's a [6.28](https://dietpi.com/downloads/images) update online to 6.3
- Distro version | `10.4 | Buster`
- Kernel version | `Linux DietPi: 4.19.118-v71`
- SBC model | `RPi4 4Gb`
- Power supply used | `Official Raspberry pi allimentation 5,1V - 3A`
- SDcard used | `SanDisk`

#### Steps to reproduce
1. [Downloaded](https://dietpi.com/downloads/images/DietPi_RPi-ARMv6-Buster.7z), unzipped, checksum sha256sum OK, put ond SD card
 2.  Rpi4 turn ON and link to internet
 2.  Installation in progress. I modify somes options in **Networking Options: Adapters**, **Language / Regional Options**
 3. In **Software Additional**, I installed **Fail2Ban** and **LxQt**, only this programs.
 4. After some reboot system, I use `startx` to start LxQt Desktop and :
- **with DietPi user : LxQt don't start**
[image]
- **with Root user : LxQt start**
[image]

- I have a rock64 and with your OS in version 6.3,  in Stretch, LxQt work on user and Root without problem.

Maybe this desktop have problems on Buster version !!

Thanks to all and good weekend

Joulinar, 2020-06-06:
Hi,

many thanks for your request. Not sure why but I'm getting a different error if I try to execute `startx` as user `dietpi`.

```
Fatal server error:
(EE) parse_vt_settings: Cannot open /dev/tty0 (Permission denied)
```

But this is somehow expected as access to `/dev/tty0` is allowed for `root` only.

```
root@DietPi3:~# ls -la /dev/tty0
crw--w---- 1 root tty 4, 0 Jun  6 21:22 /dev/tty0
root@DietPi3:~#
```

can you try using `sudo startx` as user `dietpi`

Pain-Patate, 2020-06-06:
> can you try using `sudo startx` as user `dietpi`
I did it. The LxQt start but the system is in root session (terminal in root, explorer folder too)

Joulinar, 2020-06-06:
as I said, `/dev/tty0` is accessible for root only. Therefore `dietpi` user should not be able to open a xsession.

Anyway you get a different error massage `(EE) no screens found(EE)` than I would expect. Therefore, pls can you past the entire log file

cat /home/dietpi/.local/share/xorg/Xorg.0.log

Pain-Patate, 2020-06-07:
> cat /home/dietpi/.local/share/xorg/Xorg.0.log
[Xorg.0.log](https://github.com/MichaIng/DietPi/files/4742233/Xorg.0.log)

Joulinar, 2020-06-07:
ok to bypass the permission issue of non-root user, you could do following
```
apt update && apt install xserver-xorg-legacy
nano /etc/X11/Xwrapper.config
```
adjust to `allowed_users=anybody`

On my RPi3B+ I'm able to execute `startx` as `dietpi` user now .

MichaIng, 2020-06-08:
My first idea was:
```
usermod -aG tty dietpi
```
Makes sense that it worked with Stretch since we do not do this anymore for a while but did in the past by default. However I aim to remove the `dietpi` user anyway and switch to optional new non-root user creation on first boot, so such permissions shall not be applied automatically when choosing the autostart option.

> ```
> open /dev/fb0: Permission denied
> ...
> (EE) no screens found
> ```
I guess the second is a result of the first, and both should be solved with:
```
usermod -aG video dietpi
```

However indeed `/etc/X11/Xwrapper.config` seems to be the best solution to not break the basic security intention. I'll verify that this allows non-root X11 access in all cases (all desktops, Chromium) and in case implement into `dietpi-autostart` to be applied for the selected autostart user only.

Pain-Patate, 2020-06-08:
`apt update && apt install xserver-xorg-legacy`
`nano /etc/X11/Xwrapper.config` | I modified this file with :  allowed_users=anybody

Great, `startx` in DietPi user, but I have 2 desktop, strange : 

`startx` in **DietPi user** :
[image]

`startx` in **Root user** :
[image]

Joulinar, 2020-06-08:
well this is working as expected. by default, DietPi did not setup any desktop environment for non-root user. Therefore you get the nearly empty screen for user `dietpi`. But I guess @MichaIng could explain it better than I 😃

MichaIng, 2020-06-09:
Yes that is true, although I'm wondering about the changed colour scheme. Actually the I like the dark one more 😄. I'll have to verify the configs we apply to the desktops (if any), makes sense to apply the same defaults for all users.

Pain-Patate, 2020-06-09:
Okey, Thank you both for your help

Pain-Patate, 2020-06-17:
Hi, 
I installed a new system in my Raspberry Pi .. installed LxQt with your Dietpi-Software and done this : 
`apt update && apt install xserver-xorg-legacy`
I tried `startx` and the LxQt Desktop work.;-)

Is it important to make `nano /etc/X11/Xwrapper.config`, to modify the line : `allowed_users=anybody` ?

Joulinar, 2020-06-17:
Which user you used to run `startx`?

Pain-Patate, 2020-06-17:
With user **Dietpi**

MichaIng, 2020-06-17:
Actually.... the legacy package seems to start X11 with root permissions set setuid if I understand it right, so it is definitely NOT a good solution but just a backwards-compatibility wrapper.

So if one really wants to initiate X11 as non-root users, the `video` group is the way to go. `tty` group should only be required if you want to start X11 on a different TTY, e.g. you login via SSH and run `startx` there to start X11 on the locally attached screen.

But, actually, I guess for desktops the intended way is to have a login manager like LightDM which starts X11 as a service and allows your unprivileged user to login without the need to start X11 manually.

Pain-Patate, 2020-06-21:
After `sudo apt install xserver-xorg-legacy` without `nano /etc/X11/Xwrapper.config`, without the modification `allowed_users=anybody `:
In DietPi user, it seems to me that I'm in user mode, without access because, I can 't to mount an usb key, i can't to see the root folder, i must to put `sudo` to install software

MichaIng, 2020-06-22:
> I'm in user mode with no access root because

Only the X server is started with root privileges, the desktop session itself has regular user permissions of course. However I'd need to run some tests to verify the exact differences.

MichaIng, 2021-02-15:
Even `xserver-xorg-legacy` is not reliable option. It tries to auto-detect whether the X server requires root or not. New GPUs/drivers allow to grant permissions via video/render group, but on older GPUs/drivers (legacy) this is not possible.

However, this detection may fail. `/etc/X11/Xwrapper.config` allows to override the auto-detection to force starting X server as root, but that overrides the security intentions, so isn't a good solution to ship.

But `LightDM` can be easily configured to login any user and even run custom X applications. So this is how we'll implement it + all other X application autostarts.

C0D3-M4513R, 2021-03-05:
link to lightdm autologin tutorial:
https://www.maketecheasier.com/enable-autologin-lightdm/

scgbckbone, 2021-03-11:
any estimate when this could be out ?

C0D3-M4513R, 2021-03-11:
> any estimate when this could be out ?

Do NOT ask for eta's. When a pr is linked, it prob. means, that someone is working on this issue. If this issue is closed, it is prob. either not getting implemented, or already implemented. If you really want this feature, make a pr.

ravenclaw900, 2021-03-11:
To be honest, though, it will probably be out next release (See the v7.1 milestone on the issue).

MichaIng, 2021-05-18:
PR up: #4390

closed 2021-05-29 by MichaIng