MichaIng/DietPi #3587: Netflix and Spotify on a Raspberry Pi 4 with Latest Default Chromium (widevine DRM)

WolfganP, 2020-06-06:
Just as a placeholder for those using their dietpi systems as a mediacenter and have difficulties with DRM official content (widevine protected) via Chromium:

https://blog.vpetkov.net/2019/07/12/netflix-and-spotify-on-a-raspberry-pi-4-with-latest-default-chromium/ (basically the solution is to add DRM support to latest Chromium by installing the widevine ARM lib from the official Chrome OS repository)

The author also impleemented a one line script for those wanting a quick try; https://blog.vpetkov.net/2020/03/30/raspberry-pi-netflix-one-line-easy-install-along-with-hulu-amazon-prime-disney-plus-hbo-spotify-pandora-and-many-others/

Enjoy!

MichaIng, 2020-06-08:
Many thanks for your info. We should probably add this to our Chromium online docs: https://dietpi.com/phpbb/viewtopic.php?p=3011#p3011

MichaIng, 2020-07-19:
I added it (a bid rough) to our online docs: https://dietpi.com/phpbb/viewtopic.php?p=3011#p3011
The script works fine when a desktop and chromium is installed via dietpi-software, the other article gives useful background information.
Probably we can implement it somehow into our scripts 🤔.