MichaIng/DietPi #3588: Rpi 4 4Gb | Dietpi v6.30 with Buster | DietPi-RAMlog installation pb, maybe

Pain-Patate, 2020-06-06:
### Creating a bug report/issue

#### Required Information
- DietPi version | `6.30` | It's a [6.28](https://dietpi.com/downloads/images/DietPi_RPi-ARMv6-Buster.7z) update online to 6.3
- Distro version | `10.4 | Buster`
- Kernel version | `Linux DietPi: 4.19.118-v71`
- SBC model | `RPi4 4Gb`
- Power supply used | `Official Raspberry pi allimentation 5,1V - 3A`
- SDcard used | `SanDisk`

#### Steps to reproduce
1. [Downloaded](https://dietpi.com/downloads/images/DietPi_RPi-ARMv6-Buster.7z), unzipped, checksum sha256sum OK, put ond SD card
 2.  Rpi4 turn ON and link to internet
 2.  Installation in progress. I modify somes options in **Networking Options: Adapters**, **Language / Regional Options**
 3. In **Software Additional**, I choice **Fail2Ban** and **LxQt** and  [NONE] in **Log System**
[image]

I go to **Install** an click on
[image]

 4. After some reboot system and installation, I check  **Software Additional** 
[image]

**DietPi-RAMlog is installed. Is it normal ?**

Thanks to all and have goods days

Joulinar, 2020-06-06:
Hi,

many thanks for your report. Not sure what happen on your system but as you can see on your 2nd screen, `DietPi-RAMlog` was selected to be installed. Therefore it's logic that it is installed. I guess you would need to set it to `None` again using `dietpi-software`. I did a small test in my RPi3B and couldn't replicate the issue. For me `DietPi-RAMlog` was removed and I installed `LXQt`.

Pain-Patate, 2020-06-06:
Hi, 

Speed answer :-). In fact, I didn't choose to install `DietPi-RAMlog` and he did it anyway. It's strange

Pain-Patate, 2020-06-07:
I tried again on a new system, and it did the same thing to me.
I put **[None]** on ### Log System. 
Next, the system display :
[image]
In **DietPi-Software**, after some reboot system and installation :
[image]
In **DietPi-Software**, **Uninstall** :
[image]

I can uninstall **DietPi-RAMlog** while there is **[None]** on **Log System** (**DietPi-Sofware**)

MichaIng, 2020-06-08:
Verified and fixed with: https://github.com/MichaIng/DietPi/commit/fe170d5bec55974d3bcd2167085bd66ae8f677a9

Pain-Patate, 2020-06-08:
> 
> 
> Verified and fixed with: [fe170d5](https://github.com/MichaIng/DietPi/commit/fe170d5bec55974d3bcd2167085bd66ae8f677a9)
> Changelog: [f68580a](https://github.com/MichaIng/DietPi/commit/f68580a74af77334631f64b4c57c96b5a588ff5e)

Great. :-)

closed 2020-06-30 by MichaIng