MichaIng/DietPi #3589: Certbot - Update, and add DNS challenges

Dubz, 2020-06-07:
### Creating a software request

#### Formal software information
- Software title | Certbot
- Short description | LetsEncrypt cert automation
- Official URL (if available) | 
- GitHub URL (if available) | 
- (Official) install documentation | 

#### How can DietPi make the installation easier or compatible, than following the install instructions or do APT installation, if available?
I am unable to import configs from other environments, since the parser errors out while trying to parse files from 1.0.0 while using 0.28.0
- ...

Additionally, having the DNS challenge integrated (mainly Cloudflare in my case) would be fantastic. I currently have to run Certbot with this on another machine, which can be a hassle, and is just another failure point.

Joulinar, 2020-06-07:
Hi,

many thanks for your request. Cerbot is installed via APT packages. The current stable package on Buster is `0.31.0-1`

https://packages.debian.org/search?lang=en&suite=all&searchon=names&keywords=Certbot

```
Package certbot
stretch (oldstable) (web): automatically configure HTTPS using Let's Encrypt
0.28.0-1~deb9u2: all
stretch-updates (web): automatically configure HTTPS using Let's Encrypt
0.28.0-1~deb9u2: all
stretch-backports (web): automatically configure HTTPS using Let's Encrypt
0.28.0-1~bpo9+1: all
buster (stable) (web): automatically configure HTTPS using Let's Encrypt
0.31.0-1: all
```

Dubz, 2020-06-07:
@Joulinar 

Thank you for the quick response and redirect to where this data is pulled. I greatly appreciate it!
Does this also already have some form of the DNS challenges then, or will those need to be implemented?

Joulinar, 2020-06-07:
DietPi is using the official Debian package. Not sure what feature is included within this version honestly.

MichaIng, 2020-06-08:
For DNS challenges, you either need to use a dedicated plugin for your DNS provider or the "manual" method. See: https://certbot.eff.org/docs/using.html#dns-plugins
For Cloudflare it's: `apt install python3-certbot-dns-cloudflare`

Debian/Raspbian currently ships v1.5.0 in Bullseye (testing) repo: https://packages.debian.org/certbot
As long as there is no forcing reason, I'm afraid on Buster the version will stay on v0.31.0 😢.

However of course Certbot can be installed manually as well to get the latest version: https://certbot.eff.org/lets-encrypt/pip-other

closed 2020-06-18 by MichaIng