MichaIng/DietPi #3590: Dietpi-sofware | Docker | No internet in containers

nicolasbuch, 2020-06-07:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=30
G_DIETPI_VERSION_RC=0
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
buster

- Kernel version | `uname -a`
Linux DietPi 4.14.180+ #1 SMP PREEMPT Wed May 20 14:57:20 UTC 2020 armv7l GNU/Linux

- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
Odroid XU3/XU4/MC1/HC1/HC2 (armv7l)

- Power supply used | (EG: 5V 1A RAVpower)
5V 4.0A AC adapter

- SDcard used | (EG: SanDisk ultra)
Hardkernel eMMC 16gb

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
Docker

- Was the software title installed freshly or updated/migrated?
Freshly installed on fresh dietpi installation

- Can this issue be replicated on a fresh installation of DietPi?
Yes

- Bug report ID | `echo $G_HW_UUID`
3c0770df-9995-4f26-9eac-8974c2b08fd6

#### Steps to reproduce

1. Install a fresh dietpi on the odroid XU4
2. Install docker from dietpi-software optimised
3. `docker run -it debian su`
4. `ping google.com`

#### Expected behaviour

- PING google.com (172.217.168.206): 56 data bytes
- 64 bytes from 172.217.168.206: icmp_seq=0 ttl=51 time=15.201 ms

#### Actual behaviour

- ping: google.com: Temporary failure in name resolution

#### Extra details

- There is no internet in any of my docker containers. Fresh dietpi install on my XU4. Never had any issues like this before. 
- When I start the container i get the following output in my terminal:

```
[40.101080] docker0: port 2 (veth0524ee7) entered blocking state
[40.105610] docker0: port 2 (veth0524ee7) entered disabled state
[40.111793] device veth0524ee7) entered promiscuous mode
[40.844943] eth0: renamed from veth0524ee7)
[40.101080] docker0: port 2 (veth0524ee7) entered blocking state
[40.105610] docker0: port 2 (veth0524ee7) entered disabled state
```

MichaIng, 2020-06-08:
Many thanks for your report. Docker on RPi and XU4 should be basically the, same repo, same version (also since you use Buster on both). Could you please verify the docker versions and that as well pinging a raw IP fails:
```
# On host
dpkg -l docker-ce
# Within container
ping 9.9.9.9
```

nicolasbuch, 2020-06-08:
Hi @MichaIng 
first of all let me use this opportunity to thank you (and the team) for the work you put into this - dietpi is amazing so thank you for that.

I just flashed my XU4 with a fresh version of dietpi (same image i've been using for the past few days) to make sure that none of the changes i've done would interfere with the commands you wanted me to run. 

And just like magic it works now. I've done nothing differently. 
Over the past 2 days i must have flashed my XU4 around 10 times starting over trying to fix the problem. I even installed docker from source yesterday, with the exact same result - no internet in the containers.

Today, all i did was;

1. Install a fresh dietpi on the odroid XU4
2. Install docker from dietpi-software optimised
3. `docker run -it debian su`
4. `ping google.com`

--

The problem does seem to reappear when i try to run my VPN container. I swear it was there yesterday even without the VPN container. But now i'm starting to believe that the container is at fault. Even though it works on the Rasp PI 4. I must do some more digging.

MichaIng, 2020-06-08:
> The problem does seem to reappear when i try to run my VPN container.

Would make sense since in both cases (client and server) a VPN affects network. To be true I have not much experience about how Docker containers establish or affect network interfaces, but checking routes (`ip r`) and firewall rules (e.g. `iptables`) would be the first thing I would check.

nicolasbuch, 2020-06-09:
Hello again

I was unfortunately not able to make it work. Is it possible for me to download an older version of Dietpi to try out if it fixes my issues?

Joulinar, 2020-06-09:
no DietPi did not have real separat releases. There is the current version of scripts available only.

MichaIng, 2020-06-11:
...nor does the DietPi version affect Docker network.

This is either about the Docker version itself, the VPN (version or setup), the particular network setup you applied or the kernel.

Older kernel versions can be found here: http://fuzon.co.uk/meveric/pool/xu3/l/
Older Docker versions can be found here: https://download.docker.com/linux/debian/dists/buster/pool/stable/armhf/
And for general debugging follow the two approaches from my last post.

nicolasbuch, 2020-06-17:
Thank you for taking the time to help even though it seems like DietPi was indeed not the issue here. 

> This is either about the Docker version itself, the VPN (version or setup), the particular network setup you applied or the kernel.

- I have ruled out Docker as the issue, as downgrading to several older versions make no difference.
- Iptables output looked fine, nothing stand out here 
- The issue seems to be present on Armbian also.

I've opened up an issue on the VPN docker container repo that i was using and it seems like we have finally narrowed it down to; **flushing of NAT** within the container. Running either one of the following commands would immediately cause permanent dropout of internet connectivity;

```
iptables -t nat --flush
iptables -t nat --delete-chain
```

These commands have now been removed from the container startup script. But AFAIK flushing of NAT should not cause issues like this. Could be a bug in linux kernel on the XU4?

Link to VPN container bug report : https://github.com/qdm12/private-internet-access-docker/issues/171

MichaIng, 2020-06-17:
> The issue seems to be present on Armbian also.
Could be a bug in linux kernel on the XU4?

AFAIK the Armbian Linux 4.14 is based on official Hardkernel Linux 4.14 as our image, so would make sense if it is present on both. To narrow it down you could test the Armbian Linux 5.4 image: https://www.armbian.com/odroid-xu4/#kernels-archive-all

> But AFAIK flushing of NAT should not cause issues like this.

The table is empty on the host and on other Docker containers, right? ... actually just read https://github.com/qdm12/private-internet-access-docker/issues/171#issuecomment-644487117 and indeed Docker uses the NAT table to establish host <> container network it seems. But the open question stays why this obviously is an issue some systems only and on others not.

MichaIng, 2020-12-17:
I gonna mark this as closed. Linux 5.X is in the pipeline for Odroid XU4 our end, let's see if this solves things: https://github.com/MichaIng/DietPi/pull/3861

closed 2020-12-17 by MichaIng