MichaIng/DietPi #3591: Rpi4 4gb | problem to update. boot/config.txt with arm_64bit=1

Pain-Patate, 2020-06-07:
Hi, 

### Creating a bug report/issue

#### Required Information
- DietPi version | `6.30` | It's a [6.28](https://dietpi.com/downloads/images/DietPi_RPi-ARMv6-Buster.7z) update online to 6.3
- Distro version | `10.4 | Buster`
- Kernel version | `Linux DietPi: 4.19.118-v71`
- SBC model | `RPi4 4Gb`
- Power supply used | `Official Raspberry pi allimentation 5,1V - 3A`
- SDcard used | `SanDisk`

#### Steps to reproduce
1. [Downloaded](https://dietpi.com/downloads/images/DietPi_RPi-ARMv6-Buster.7z), unzipped, checksum sha256sum OK,
2. I modify the file **config.txt** in the boot in the SD card by adding : `arm_64bit=1# Force booting in 64bit mode`
3. I put the SD card in the Rpi4 and link to internet
4. Installation in progress and then, the installation stop :
[image]

I can to install this **vmtouch_aarch64.deb**

Thanks to all, and take care of yourself

Joulinar, 2020-06-07:
Pls don't use the 64bit flag for initial installation on your RPi4. Otherwise an incorrect package is pulled.

Pain-Patate, 2020-06-07:
So, i must to install with the DietPi 32 bits and then, modify the file **config.txt** in the boot folder with : `arm_64bit=1# Force booting in 64bit mode` ?

Joulinar, 2020-06-07:
Yes that would be the way. But keep in mind that the 64bit flag is just forching the kernel into 64bit mode while the user space will be 32bit still. 

Anyway the Raspberry Foundation is working on a real 64bit Raspbian OS version. But this is in an early beta stage and not yet supported by DiePi.

Pain-Patate, 2020-06-07:
Ah shit. If I understand, I can't to install program in 64 bits ?

Joulinar, 2020-06-07:
Well just try it out. Finish the initial setup. Afterwards change the `config.txt` to force 64bit kernel mode and try using your software.

Pain-Patate, 2020-06-07:
> 
> 
> Well just try it out. Finish the initial setup. Afterwards change the `config.txt` to force 64bit kernel mode and try using your software.

Okey.  The Kernel will be in 64 bits and that's it, no program in 64 bits ? (Sorry, I m a noob in linux and I'm learning)

Joulinar, 2020-06-07:
What kind of software you are looking for that requires 64bit explicitly?

Pain-Patate, 2020-06-07:
In fact, I want to install [Opensnitch](https://github.com/gustavo-iniguez-goya/opensnitch).
And I'm testing to install it with a [arm64 version](https://github.com/gustavo-iniguez-goya/opensnitch/issues/18#issuecomment-629817554)
I think I go to speak with him, to have the version with armhf. I will test in my Rock64 with arm64
[Opensnitch](https://github.com/gustavo-iniguez-goya/opensnitch), Fork of [Opensnitch](https://github.com/evilsocket/opensnitch), is a great firewall that runs in service

Joulinar, 2020-06-08:
Just to avoid any confusion. DietPi is not an own Operating System or Distribution. DietPi is a bundle of scripts on top of a standard Debian installation. In case of Raspberry Pi, base image is Raspberry Pi OS (previously called Raspbian). Currently Raspberry Pi OS is available as 32bit version only. Yes you can switch kernel into 64bit mode on RPi3 and 4 but this will not change the package architecture. For verification purposes, I installed an official Raspberry Pi OS 32bit image and added `arm_64bit=1` into `/boot/config.txt`. 

```
pi@raspberrypi:~ $ dpkg --print-architecture
armhf
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 4.19.118-v8+ #1311 SMP PREEMPT Mon Apr 27 14:32:38 BST 2020 aarch64 GNU/Linux
pi@raspberrypi:~ $
```

As you can see, kernel was switched to `aarch64`, however package architecture remains on `armhf`. 

Probably you would need to wait on Raspberry Pi Foundation to release an real 64bit OS.

https://www.raspberrypi.org/forums/viewtopic.php?f=117&t=275370

MichaIng, 2020-06-08:
Yes one needs to understand that a 64-bit kernel does not make a 64-bit OS. Actually using a 64-bit kernel on 32-bit OS even causes issues since many installers and config scripts (including ours) use `uname -m` to check which binary/package architectures they should setup. But a 64-bit binary will have issues calling a 32-bit library e.g. or a 64-bit package might want to pull 64-bit dependencies which either lead to doubled installs, conflicts or are simply not available in the 32-bit repo.

Long story short: As long as you do not use a real 64-bit OS on RPi, do not set `arm_64bit=1`, it will not have any benefits but only causes havoc earlier or later 😉.
__________
But aside from that, Raspberry Pi OS 64-bit (which is in fact regular Debian with arm64 repo) is in early beta and we're working on making DietPi compatible with it. Once done a official 64-bit image will be released.

Pain-Patate, 2020-06-08:
Thanks to you : @Joulinar and @MichaIng to your time and I understand better and will take your advices
And So, this issues is not really a problem and we must wait a OS in 64 bit. We can close this issues. Are you OK ?

Joulinar, 2020-06-09:
yep I'm going to close the issue, feel free to reopen if needed.

closed 2020-06-09 by Joulinar