MichaIng/DietPi #3592: Raspberry Pi 4 (4GB) running DietPi via Berryboot

apatel007, 2020-06-09:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=30
G_DIETPI_VERSION_RC=0
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
10.4

- Kernel version | `uname -a`
Linux dietpi4 4.19.75v64 #1 SMP PREEMPT Sat Oct 5 03:06:01 CEST 2019 aarch64 GNU/Linux

- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
RPi 4 Model B (aarch64)

- Power supply used | (EG: 5V 1A RAVpower)
The one that came with the Pi4 Kit made for it.

- SDcard used | (EG: SanDisk ultra)
64GB SanDisk Ultra

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
Fresh latest IMG taken from a berryboot image archive of DietPi4:
https://berryboot.alexgoldcheidt.com/images/

- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. ...Install image referenced above; run all updates.
2. ...Periodically getting "dpkg: warning: failed to open configuration file '/etc/dpkg/dpkg.cfg.d/dietpi-no_armbian' for reading: No such device or address" whenever running apt-get install or reinstall or remove on anything.

#### Expected behaviour

- ...That error should not come.

#### Actual behaviour

- ...I get the error above.

#### Extra details

- ...There is a file there:
user@host:/etc/dpkg/dpkg.cfg.d# ls -als
total 8
4 drwxr-xr-x 2 root root 4096 Jun  1 12:59 .
4 drwxr-xr-x 6 root root 4096 Jan  7 17:33 ..
0 crwxrwxrwx 1 root root 0, 0 Jun  1 12:59 dietpi-no_armbian

What is this dietpi-no_armbian c file for and why does it throw these errors? I appreciate any insight you may be able to provide. I did not see anything in the forums about this. Thanks in advance for whatever you can help with!

MichaIng, 2020-06-09:
Many thanks for your report.

Can you give the link of the exact download you found and flashed? I cannot find it although they seem to offer downloads of a large amount of different images.

What I cannot verify is if they are regularly updated and if they host the untouched official images or add/change something. I can only recommend to use our official download: https://dietpi.com/downloads/images/DietPi_RPi-ARMv6-Buster.7z

`dietpi-no_armbian` should not be present on any RPi image from us, although I think we accidentally added it on one batch of images. It is aimed for images based on Armbian kernel which require an Armbian package that pulls user space tools and services as well. This config prevents those services and tools from being installed, which would otherwise conflict with DietPi tools and services.

However it should not produce any error. The contained file exclusions are very specific and the directives do not require any "device or address". This should be the content: https://github.com/MichaIng/DietPi/blob/master/PREP_SYSTEM_FOR_DIETPI.sh#L1132-L1141

So what I would do:
- Verify that you don't have any Armbian repository installed: `cat /etc/apt/sources.list{,.d/*.list}`
- If not, remove the file: `rm /etc/dpkg/dpkg.cfg.d/dietpi-no_armbian`
- If there is indeed an Armbian repo installed, the image is wrong and manipulated since our RPi images are based on Raspbian Lite and only use official RPi and Raspbian repo and Armbian does not even support RPi at all 😉.

MichaIng, 2020-06-11:
Okay found and checked the DietPi Buster image and indeed the `dietpi-no_armbian` file is present there. Current DietPi RPi images do not contain it so the next update will resolve it. I just remember that even DietPi-Update should remove it via incremental patches, as long as it's no Armbian-based image.

However the errors are still strange. I just tested on other non-Armbian systems (my RPi and VMs) and could not replicate the output. apt/dpkg is just silent as expected in any case. However not worth it do dig deeper here as long as we don't get more reports about this.
_____________
Not sure about HDMI audio, of course in case of RPi4 one has to take are that there are two HDMI ports and by default I guess audio is sent to port 0 only.
Did you configure ALSA manually or via `dietpi-config`? In case can you paste:
```
cat /etc/asound.conf
```

MichaIng, 2020-07-05:
I mark this as closed. Issue is not replicable on DietPi an obsolete file has been removed on current images. Feel free to reopen if required.

closed 2020-07-05 by MichaIng