MichaIng/DietPi #3593: What is the default password of Nexcloud in DietPi? I can't login...

tifDev, 2020-06-09:
#### Required Information
- DietPi version 
VERSION_CORE=6
VERSION_SUB=30
VERSION_RC=0

- Distro version: buster

#### Steps to reproduce
1. Install dietpi in a RPI4 and update
2. Install nexcloud via dietpi-software
3. Installation finished, then reboot
4. Try to access in other computer (`my_rpi_ip/nexcloud`)
5. Login page appears but no password works

[image]

**What is the default password? Where can i see the password (is in the database)?**

Joulinar, 2020-06-09:
Hi,

many thanks for your report. Basically the password is your own DietPi master password, that was set during initial installation. As well you can have a look to our small user guide. 

https://dietpi.com/phpbb/viewtopic.php?p=3026#p3026

tifDev, 2020-06-09:
You mean root password?

tifDev, 2020-06-09:
Nop, root password doesn't work neither

MichaIng, 2020-06-09:
On first login you were able to choose two default passwords, one for new software installs (global software password) and one for the two UNIX users "root" and "dietpi". The global software password is the one that you need to enter here, it defaults to `dietpi` if you skipped setting a different one, or if you set it via `dietpi.txt` before first boot, that is the one used.

I don't want to share it publicly here but there is a way to recover it, if you cannot find/remember anymore. You can send me an email or send PM on DietPi forum and I'll share the recovery method.

Joulinar, 2020-06-09:
But I guess you can set a new one using NextCloud tool on CLI. Isn't it?

MichaIng, 2020-06-09:
Yes that is true. But that does not change the Nextcloud admin password anymore now.

MichaIng, 2020-06-09:
Ah you're right, that is the better method:
```
ncc user:resetpassword admin
```
And then reset DietPi global default password as well via `dietpi-config` > `Security Options` > `Change Passwords`
Select `cancel` when being asked if you want to change the UNIX user passwords, since this would affect root user login.

tifDev, 2020-06-09:
I know the root password, the other i also know it but i'm not 100% sure if it's correct. I tested both on nextcloud and none worked.
So you suggest me to run `ncc user:resetpassword admin` and then follow the steps you said to reset DietPi global default password, right?

MichaIng, 2020-06-09:
Yes exactly.

Joulinar, 2020-06-09:
basically both activities are independent from each other. `ncc user:resetpassword admin` will just reset the admin pw on NextCloud. However as you are not 100% sure on the global application pw, it make sent to rest as well. This will take effect for installations that will be done in future.

tifDev, 2020-06-09:
Thank you all. Now it's working.

closed 2020-06-09 by tifDev

Ricardo-Simoes, 2022-11-22:
_UPDATE_: According to **[NextCloud official docs](https://docs.nextcloud.com/server/latest/admin_manual/configuration_user/reset_admin_password.html)** the command to reset admin password is:
`sudo -u www-data php /var/www/nextcloud/occ user:resetpassword admin`

NOTE: It used to be **ncc**, but now it's **occ**.

MichaIng, 2022-11-22:
`ncc` is not and never was a Nextcloud command. It is a shell function created in DietPi only, as a shortcut for `sudo -u www-data php /var/www/nextcloud/occ`. See:
```sh
grep ncc /boot/dietpi/func/dietpi-globals
```
So both variants are identical on DietPi systems. I heard that NextCloudPi uses the same shortcut btw.

Ricardo-Simoes, 2022-11-22:
> `ncc` is not and never was a Nextcloud command. It is a shell function created in DietPi only,

You're absolutely correct. Please accept my apologies.
Edited my previous post to correct it.