MichaIng/DietPi #3594: How can i change keyboard layout from gb to pt (portuguese) on a Razer Ornata Chroma keyboard

tifDev, 2020-06-09:
- DietPi version 
VERSION_CORE=6
VERSION_SUB=30
VERSION_RC=0

- Distro version: buster

#### Steps to reproduce
1. Run `dietpi-config`
2. Choose option 5 : "Language/Regional Options"
3. Choose option "Keyboard"
4. "Select the model of the keyboard of this machine"
I get stucked here because i can't find my keyboard model (Razer Ornata Chroma)

**How can i solve this?**

MichaIng, 2020-06-09:
Many thanks for your question. I also do not find the native keyboard configuration too intuitive.

Of course not all existing keyboard models (hundreds of thousands) are listed, only a few well known ones. Please select `Generic 105-key PC (intl.)` instead which basically fits the common European layout for Windows keyboards. For details other other generic layouts see: https://en.wikipedia.org/wiki/IBM_PC_keyboard

On the next screen select `Others` at the bottom to get to the language selection. You'll find `Portuguese` there. Next you can again select `Portuguese` and just keep hitting return to select the defaults about AltGr and composite key, should basically work.

tifDev, 2020-06-09:
Thanks for your quick reply. Followed your steps and it's working.

closed 2020-06-09 by tifDev

justRegularStudent, 2023-10-09:
actually nothing really works that i get a german keyboard layout... is it really that hard to implement it?

MichaIng, 2023-10-09:
`dietpi-config` > Language/Regional Options > Keyboard does not work for you? The steps through the dialogues from there on for a German layout should be the same as for Portuguese, as I explained above.

justRegularStudent, 2023-10-10:
No sadly not. I also looked in the keyboard config and it is correctly setted...

MichaIng, 2023-10-12:
Yes that looks correct. After a reboot, this configuration should take effect. If this is not the case, which keys are wrong in particular?