MichaIng/DietPi #3595: NextCloud / mariadb issue (piZero)

jedisha, 2020-06-09:
Hi,
just getting set NC set up on a zero and had this issue:

`#### Details:
- Date           | Tue Jun  9 16:27:56 BST 2020
- DietPi version | v6.30.0 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi Zero (armv6l) (ID=1)
- Kernel version | Linux DietPi 4.19.118+ #1311 Mon Apr 27 14:16:15 BST 2020 armv6l GNU/Linux
- Distro         | buster (ID=5)
- Command        | systemctl restart mariadb
- Exit code      | 1
- Software title | DietPi-Software
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Job for mariadb.service failed because the control process exited with error code.
See "systemctl status mariadb.service" and "journalctl -xe" for details.
```
`

I recently changed location of swapfile and dietpi userdata as system was telling me that file space was all used. Now (perhaps unrelated?) My new install of NC isn't working. I uninstalled Nextcloud and re-installed, but still not working (didn't wipe files from system, just used dietpi-software to uninstall)

When I tried to access nextcloud this showed up:

> Internal Server Error

> The server encountered an internal error and was unable to complete your request.

> Please contact the server administrator if this error reappears multiple times, please include the 

> technical details below in your report.

> More details can be found in the server log.

Finally, I just upgraded dietpi version to latest.

Any help greatlt appreciated!

Joulinar, 2020-06-09:
Hi,

many thanks for your report. I guess MariaDB is not working due moving user data. Therefore you are not able to reach NextCloud web site as well. Pls can you post following

```
systemctl restart mariadb.service
journalctl -u mariadb
cat /var/log/mysql/error.log
```

MichaIng, 2020-06-09:
> I recently changed location of swapfile and dietpi userdata

> I guess MariaDB is not working due moving user data

Jep very likely. Please also paste the following command outputs:
```
readlink /var/lib/mysql
readlink -f /var/lib/mysql
```

jedisha, 2020-06-09:
`systemctl restart mariadb.service`
returned:
Job for mariadb.service failed because the control process exited with error code.
See "systemctl status mariadb.service" and "journalctl -xe" for details.
`journalctl -u mariadb`
returned:
-- Logs begin at Tue 2020-06-09 15:46:23 BST, end at Tue 2020-06-09 16:48:15 BST. --
Jun 09 15:47:27 DietPi systemd[1]: Starting MariaDB 10.3.22 database server...
Jun 09 15:47:29 DietPi mysqld[684]: 2020-06-09 15:47:29 0 [Note] /usr/sbin/mysqld (mysqld 10.3.22-MariaDB-0+d
eb10u1) starting as process 684 ...
Jun 09 15:47:29 DietPi mysqld[684]: 2020-06-09 15:47:29 0 [Warning] Can't create test file /var/lib/mysql/Die
tPi.lower-test
Jun 09 15:47:29 DietPi mysqld[684]: [90B blob data]
Jun 09 15:47:29 DietPi mysqld[684]: 2020-06-09 15:47:29 0 [ERROR] Aborting
Jun 09 15:47:29 DietPi systemd[1]: mariadb.service: Main process exited, code=exited,
 status=1/FAILURE
Jun 09 15:47:29 DietPi systemd[1]: mariadb.service: Failed with result 'exit-code'.

Jun 09 15:47:29 DietPi systemd[1]: Failed to start MariaDB 10.3.22 database server.

Jun 09 16:09:56 DietPi systemd[1]: Starting MariaDB 10.3.22 database server...
Jun 09 16:09:57 DietPi mysqld[1245]: 2020-06-09 16:09:57 0 [Note] /usr/sbin/mysqld (mysqld 10.3.22-MariaDB-0+
deb10u1) starting as process 1245 ...
Jun 09 16:09:57 DietPi mysqld[1245]: 2020-06-09 16:09:57 0 [Warning] Can't create test file /var/lib/mysql/Di
etPi.lower-test
Jun 09 16:09:57 DietPi mysqld[1245]: [90B blob data]
Jun 09 16:09:57 DietPi mysqld[1245]: 2020-06-09 16:09:57 0 [ERROR] Aborting
Jun 09 16:09:57 DietPi systemd[1]: mariadb.service: Main process exited, code=exited,
 status=1/FAILURE
Jun 09 16:09:57 DietPi systemd[1]: mariadb.service: Failed with result 'exit-code'.

Jun 09 16:09:57 DietPi systemd[1]: Failed to start MariaDB 10.3.22 database server.

Jun 09 16:11:37 DietPi systemd[1]: Starting MariaDB 10.3.22 database server...
Jun 09 16:11:39 DietPi mysqld[1448]: 2020-06-09 16:11:39 0 [Note] /usr/sbin/mysqld (mysqld 10.3.22-MariaDB-0+
deb10u1) starting as process 1448 ...
Jun 09 16:11:39 DietPi mysqld[1448]: 2020-06-09 16:11:39 0 [Warning] Can't create test file /var/lib/mysql/Di
etPi.lower-test
Jun 09 16:11:39 DietPi mysqld[1448]: [90B blob data]
Jun 09 16:11:39 DietPi mysqld[1448]: 2020-06-09 16:11:39 0 [ERROR] Aborting
Jun 09 16:11:39 DietPi systemd[1]: mariadb.service: Main process exited, code=exited,
 status=1/FAILURE
Jun 09 16:11:39 DietPi systemd[1]: mariadb.service: Failed with result 'exit-code'.

Jun 09 16:11:39 DietPi systemd[1]: Failed to start MariaDB 10.3.22 database server.

Jun 09 16:21:45 DietPi systemd[1]: Starting MariaDB 10.3.22 database server...
Jun 09 16:21:46 DietPi mysqld[5531]: 2020-06-09 16:21:46 0 [Note] /usr/sbin/mysqld (mysqld 10.3.22-MariaDB-0+
deb10u1) starting as process 5531 ...
Jun 09 16:21:46 DietPi mysqld[5531]: 2020-06-09 16:21:46 0 [Warning] Can't create test file /var/lib/mysql/Di
etPi.lower-test
Jun 09 16:21:46 DietPi mysqld[5531]: [90B blob data]
Jun 09 16:21:46 DietPi mysqld[5531]: 2020-06-09 16:21:46 0 [ERROR] Aborting
Jun 09 16:21:46 DietPi systemd[1]: mariadb.service: Main process exited, code=exited,
 status=1/FAILURE
Jun 09 16:21:46 DietPi systemd[1]: mariadb.service: Failed with result 'exit-code'.

Jun 09 16:21:46 DietPi systemd[1]: Failed to start MariaDB 10.3.22 database server.

Jun 09 16:27:40 DietPi systemd[1]: Starting MariaDB 10.3.22 database server...
Jun 09 16:27:41 DietPi mysqld[5623]: 2020-06-09 16:27:41 0 [Note] /usr/sbin/mysqld (mysqld 10.3.22-MariaDB-0+
deb10u1) starting as process 5623 ...
Jun 09 16:27:41 DietPi mysqld[5623]: 2020-06-09 16:27:41 0 [Warning] Can't create test file /var/lib/mysql/Di
etPi.lower-test
Jun 09 16:27:41 DietPi mysqld[5623]: [90B blob data]
Jun 09 16:27:41 DietPi mysqld[5623]: 2020-06-09 16:27:41 0 [ERROR] Aborting
Jun 09 16:27:41 DietPi systemd[1]: mariadb.service: Main process exited, code=exited,
 status=1/FAILURE
Jun 09 16:27:41 DietPi systemd[1]: mariadb.service: Failed with result 'exit-code'.
m
[... 53 lines cut ...]
m
Jun 09 16:48:14 DietPi systemd[1]: Starting MariaDB 10.3.22 database server...
Jun 09 16:48:15 DietPi mysqld[6361]: 2020-06-09 16:48:15 0 [Note] /usr/sbin/mysqld (mysqld 10.3.22-MariaDB-0+
deb10u1) starting as process 6361 ...
Jun 09 16:48:15 DietPi mysqld[6361]: 2020-06-09 16:48:15 0 [Warning] Can't create test file /var/lib/mysql/Di
etPi.lower-test
Jun 09 16:48:15 DietPi mysqld[6361]: [90B blob data]
Jun 09 16:48:15 DietPi mysqld[6361]: 2020-06-09 16:48:15 0 [ERROR] Aborting
Jun 09 16:48:15 DietPi systemd[1]: mariadb.service: Main process exited, code=exited,
 status=1/FAILURE
Jun 09 16:48:15 DietPi systemd[1]: mariadb.service: Failed with result 'exit-code'.
m
Jun 09 16:48:15 DietPi systemd[1]: Failed to start MariaDB 10.3.22 database server.
m

`readlink /var/lib/mysql`
returns:
`/mnt/dietpi_userdata/mysql`

`readlink -f /var/lib/mysql`
returns:
`/mnt/d25778c5-6af9-494b-997a-001c1a21ee6f/dietpi_userdata/mysql`
(mount point for external HDD)

(BTW - YOU GUYS ARE AWESOME! Thank you for the prompt reply!!)

jedisha, 2020-06-09:
@Joulinar Thanks for the edit! (Apologies!!!)

MichaIng, 2020-06-09:
Okay the symlinks are set correctly, lets check file system type, free space and permissions:
```
df -T /mnt/d25778c5-6af9-494b-997a-001c1a21ee6f
ls -al /mnt/d25778c5-6af9-494b-997a-001c1a21ee6f/dietpi_userdata/mysql
```

Joulinar, 2020-06-09:
yw, it's better readable this way 😉

jedisha, 2020-06-09:
`df -T /mnt/d25778c5-6af9-494b-997a-001c1a21ee6f`

returns:

 `root@DietPi:~# df -T /mnt/d25778c5-6af9-494b-997a-001c1a21ee6f
Filesystem     Type  1K-blocks     Used  Available Use% Mounted on
/dev/sda1      ext4 3844639728 18265836 3631006676   1% /mnt/d25778c5-6af9-494b-997a-001c1a21ee6f
`

`ls -al /mnt/d25778c5-6af9-494b-997a-001c1a21ee6f/dietpi_userdata/mysql`

returns:

`root@DietPi:~# ls -al /mnt/d25778c5-6af9-494b-997a-001c1a21ee6f/dietpi_userdata/mysql
total 176188
drwxr-xr-x  5 mysql  mysql      4096 Jun  9 13:39 .
drwxrwxr-x 10 dietpi dietpi     4096 Jun  9 13:12 ..
-rw-rw----  1 mysql  mysql     16384 Jun  9 13:39 aria_log.00000001
-rw-rw----  1 mysql  mysql        52 Jun  9 13:39 aria_log_control
-rw-r--r--  1 root   root          0 Mar 20 09:14 debian-10.3.flag
-rw-rw----  1 mysql  mysql      4123 Jun  9 13:39 ib_buffer_pool
-rw-rw----  1 mysql  mysql  79691776 Jun  9 13:39 ibdata1
-rw-rw----  1 mysql  mysql  50331648 Jun  9 13:39 ib_logfile0
-rw-rw----  1 mysql  mysql  50331648 Jun  6 13:05 ib_logfile1
-rw-rw----  1 mysql  mysql         0 Mar 20 09:14 multi-master.info
drwx------  2 mysql  mysql      4096 Mar 20 09:15 mysql
-rw-rw----  1 root   root         16 Mar 20 09:15 mysql_upgrade_info
drwx------  2 mysql  mysql     12288 Jun  8 15:56 nextcloud
drwx------  2 mysql  mysql      4096 Mar 20 09:15 performance_schema
`

jedisha, 2020-06-09:
@Joulinar how am i getting the code snippets wrong?!

Joulinar, 2020-06-09:
ok let's check if user `mysql` is able to write a test file to the directory

`sudo -u mysql touch /var/lib/mysql/test.file`

@jedisha 
to mark more lines as code snippets, just copy everything into the text box, mark the code section und hit the `insert code` button. This will place the snip into 3 backtick before and after the code section

MichaIng, 2020-06-09:
> how am i getting the code snippets wrong?!

When you have a multi line output, wrap two lines with ` ``` ` around it, which makes it best readable. The same is done when you paste the multi line output into the text field first, then select all lines and hit the "`Insert code`" button.
Singe ` signs are only good for single code lines or snippets 😉.

Okay file system and permissions are fine, plenty of free space, not sure currently why MariaDB cannot create the test file.

Do you have AppArmor or selinux installed?
```
dpkg -l apparmor selinux-basics
```

jedisha, 2020-06-09:
@MichaIng @Joulinar 

Thanks both so much! Am back home now and that pi is in my office, so I will be able to check this in the morning.

Thanks so far!

MichaIng, 2020-07-05:
I mark this as closed due to missing reply. Feel free to reopen if issue persists.

closed 2020-07-05 by MichaIng