MichaIng/DietPi #3597: Dietpi won't start! (oDroidXU4) Failed to load DietPi-Globals

jedisha, 2020-06-09:
Feeling v embarassed to come to the forum twice in a day with issues for two different machines! :-(

My oDroidXU4 was saying it was running out of disk space this morning as I tried to back it up before an update. I'm *fairly* sure I had already moved dietpi_userdata to my external SSD in the past. (The SSD has indeed run out of space and I've been migrating Plex media files etc over to a new, larger WD-red HDD.) I believe I attempted to move dietpi_userdata over to the WD-red in order to free up space on the SSD to complete backup and then update of system. 
HOWEVER... now all I get when I SSH into it is:

```

Using username "root".

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Mon Aug 27 01:09:40 2018 from 192.168.0.28
-bash: /boot/dietpi/func/dietpi-globals: No such file or directory
[FAILED] DietPi-Login | Failed to load DietPi-Globals. Skipping DietPi login scripts...
root@DietPi:~#

```

FTP also appears to work, however neither of my external drives appear to be mounted so cannot check contents of them...

Any help greatly appreciated!

tiaringhio, 2020-06-10:
I'm having the same problem, first the Pi told me it had no disk space available (which is weird because i'm sure there is some) then after rebooting it gives the following error:

`-bash: /boot/dietpi/func/dietpi-globals: No such file or directory`
`[FAILED] DietPi-Login | Failed to load DietPi-Globals. Skipping DietPi login scripts...`

The /boot folder is empty, i've done a couple of reboots but it doesn't seem to work. I can provide logs if needed.

tiaringhio, 2020-06-10:
Just wanted to update, the problem may be that RootFS is Read Only, how to solve this?
Follow the issue #3511 , it worked for me even after a reboot.
You will have to remount any external hdd you have mounted (you won't loose any data whatsoever)

MichaIng, 2020-07-05:
I mark this as closed due to missing reply. Feel free to reopen is required.

closed 2020-07-05 by MichaIng