MichaIng/DietPi #3598: Dietpi First Boot Setup failing

camow7, 2020-06-10:
#### Details:
- Date           | Wed Jun 10 01:48:49 BST 2020
- DietPi version | v6.28.0 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Armbian
- Hardware       | ROCK Pi 4 (aarch64) (ID=72)
- Kernel version | Linux DietPi 5.4.8-rockchip64 #19.11.7 SMP PREEMPT Sun Jan 5 23:30:10 CET 2020 aarch64 GNU/Linux
- Distro         | buster (ID=5)
- Command        | wget --spider -t 1 -T 5 https://raw.githubusercontent.com/MichaIng/DietPi/master/dietpi.txt
- Exit code      | 4
- Software title | DietPi-Set_software
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Spider mode enabled. Check if remote file exists.
--2020-06-10 01:48:43--  https://raw.githubusercontent.com/MichaIng/DietPi/master/dietpi.txt
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... failed: Connection timed out.
wget: unable to resolve host address ‘raw.githubusercontent.com’
```

camow7, 2020-06-10:
Resolved this by extending the `-T` time from 5 to 15 seconds as:
```
wget --spider -t 1 -T 15 https://raw.githubusercontent.com/MichaIng/DietPi/master/dietpi.txt
```

closed 2020-06-10 by camow7

MichaIng, 2020-06-11:
Alter `CONFIG_G_CHECK_URL_TIMEOUT` in `/boot/dietpi.txt` to make this change permanent.

Since >5 seconds is unexpected long (~0.5 seconds is usual), you might want to check if this long connections times are common on your system/network and why 😉.

Joulinar, 2020-06-11:
same could be achieved using `dietpi-config` > `8  : Network Options: Misc` > `G_CHECK_URL Timeout`