MichaIng/DietPi #3599: Dietpi

glecin, 2020-06-10:
#### Details:
- Date           | Tue Jun  9 20:28:48 MDT 2020
- DietPi version | v6.30.0 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Armbian
- Hardware       | ROCK64 (aarch64) (ID=43)
- Kernel version | Linux DietPi 5.4.43-rockchip64 #20.05.2 SMP PREEMPT Tue Jun 2 22:47:34 CEST 2020 aarch64 GNU/Linux
- Distro         | buster (ID=5)
- Command        | apt-get -qq --allow-change-held-packages install xserver-xorg xinit xcompmgr xterm dbus-x11 xfonts-base x11-xserver-utils x11-utils libgl1-mesa-dri mesa-utils mesa-utils-extra
- Exit code      | 100
- Software title | DietPi-Software
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
E: dpkg was interrupted, you must manually run 'dpkg --configure -a' to correct the problem.
```
---------------------------------------------------------------------
[FAILED] DietPi-Software | Unable to continue

Joulinar, 2020-06-10:
Hi,

many thanks for your report. Pls can you a couple of more lines from the error message. As well did you executed `dpkg --configure -a` as requested on the error message?

MichaIng, 2020-06-11:
At best run the following in your terminal and paste the full output here:
```
apt-get --allow-change-held-packages install xserver-xorg xinit xcompmgr xterm dbus-x11 xfonts-base x11-xserver-utils x11-utils libgl1-mesa-dri mesa-utils mesa-utils-extra
```

MichaIng, 2020-07-05:
I mark this issue as closed due to missing reply. Feel free to reopen if required.

closed 2020-07-05 by MichaIng