MichaIng/DietPi #3601: June 8th DietPi won't install via Wi-Fi

dsnyder0pc, 2020-06-13:
### Creating a bug report/issue

#### Required Information
- DietPi version | 6.30.0
- Distro version | 10.4
- Kernel version | did not check
- SBC model | RPi3 Model A+
- Power supply used | Powered USB Hub, 5V, 2.4A
- SDcard used | SanDisk ultra 32 GB

#### Additional Information (if applicable)
- Software title | First boot from Wi-Fi
- Was the software title installed freshly or updated/migrated? New Image
- Can this issue be replicated on a fresh installation of DietPi? Fails during first boot

#### Steps to reproduce

1. ...Flash a freshly formatted microSD card with the June 8th DietPi build for Raspberry Pi (md5=2a26658edbc5c88159be0519967696f6)
2. ...Configure dietpi.txt and dietpi-wifi.txt with the following changes:
```
- Settings updated in dietpi.txt:
         AUTO_SETUP_KEYBOARD_LAYOUT = us
   AUTO_SETUP_NET_WIFI_COUNTRY_CODE = US
                AUTO_SETUP_TIMEZONE = America/Los_Angeles
       CONFIG_BOOT_WAIT_FOR_NETWORK = 2
                   CONFIG_SOUNDCARD = usb-dac
                  AUTO_SETUP_LOCALE = en_US.UTF-8
         CONFIG_CHECK_CONNECTION_IP = 127.0.0.1
    AUTO_SETUP_NET_ETHERNET_ENABLED = 0
     AUTO_SETUP_INSTALL_SOFTWARE_ID = 5,121
            CONFIG_CHECK_DNS_DOMAIN = localhost
                    SURVEY_OPTED_IN = 0
               AUTO_SETUP_AUTOMATED = 1
            AUTO_SETUP_NET_HOSTNAME = name
        CONFIG_G_CHECK_URL_ATTEMPTS = 5
        AUTO_SETUP_NET_WIFI_ENABLED = 1
         CONFIG_G_CHECK_URL_TIMEOUT = 300
                    CONFIG_NTP_MODE = 0
       CONFIG_SERIAL_CONSOLE_ENABLE = 0

- Settings updated in dietpi-wifi.txt:
                          WIFI_SSID = '<my_ssid>'
                           WIFI_KEY = '<my_password>'
                        WIFI_KEYMGR = 'WPA-PSK'
```
3. ...eject the microSD card and insert it into the RPi3 A+
4. ...connect HDMI and power to the RPi3 A+
5. ...observe that it fails to connect to Wi-Fi and fails to boot

#### Expected behaviour

- ...the RPi should connect to the Wi-Fi network, updated itself, install additional packages, and become available for use. This worked fine almost 100% of the time with the January image. It consistently fails with the June 8th image. Back then, I used `CONFIG_WIFI_COUNTRY_CODE` instead of `AUTO_SETUP_NET_WIFI_COUNTRY_CODE` to set the Wi-Fi Country Code.

#### Actual behaviour

- ...the RPi fails to connect to Wi-Fi, can't access any update server, and the installation terminates with:
```[FAILED] Unknown install state/First run setup failed
An error has occurred either during first run update or installs.
```

#### Extra details

- Video of the failing boot here: https://youtu.be/zFgR9LAcfjk

dsnyder0pc, 2020-06-13:
**Diff showing my changes to dietpi.txt to enable Wi-Fi boot:**
```
--- /Users/dsnyder/tmp/dietpi.txt	2020-06-08 10:07:58.000000000 -0700
+++ dietpi.txt	2020-06-13 16:01:12.000000000 -0700
@@ -8,27 +8,27 @@
 #------------------------------------------------------------------------------------------------------
 ##### Language/Regional Options #####
 # Locale: eg: "en_GB.UTF-8" / "de_DE.UTF-8" | One entry and UTF-8 ONLY!
-AUTO_SETUP_LOCALE=C.UTF-8
+AUTO_SETUP_LOCALE=en_US.UTF-8
 
 # Keyboard Layout eg: "gb" / "us" / "de" / "fr"
-AUTO_SETUP_KEYBOARD_LAYOUT=gb
+AUTO_SETUP_KEYBOARD_LAYOUT=us
 
 # Timezone eg: "Europe/London" / "America/New_York" | Full list (TZ*): https://en.wikipedia.org/wiki/List_of_tz_database_time_zones
-AUTO_SETUP_TIMEZONE=Europe/London
+AUTO_SETUP_TIMEZONE=America/Los_Angeles
 
 ##### Networking Options #####
 # Enable Ethernet or WiFi adapter: 1=enable | 0=disable
 # - If both Ethernet and WiFi are enabled, WiFi will take priority and Ethernet will be disabled.
 # - If using WiFi, please edit dietpi-wifi.txt to pre-enter credentials.
-AUTO_SETUP_NET_ETHERNET_ENABLED=1
-AUTO_SETUP_NET_WIFI_ENABLED=0
+AUTO_SETUP_NET_ETHERNET_ENABLED=0
+AUTO_SETUP_NET_WIFI_ENABLED=1
 
 # Force ethernet speeds: 0=automatic speed | 10 = 10Mbit | 100 = 100Mbit etc
 # - This is mainly aimed at Pine A64 which may have an HW issue that causes unstable 1Gbit link.
 AUTO_SETUP_NET_ETH_FORCE_SPEED=0
 
 # WiFi country code: 2 uppercase character value (e.g. GB US DE JP): https://en.wikipedia.org/wiki/ISO_3166-1_alpha-2
-AUTO_SETUP_NET_WIFI_COUNTRY_CODE=GB
+AUTO_SETUP_NET_WIFI_COUNTRY_CODE=US
 
 # Enter your Static Network details below, if applicable.
 AUTO_SETUP_NET_USESTATIC=0
@@ -38,7 +38,7 @@
 AUTO_SETUP_NET_STATIC_DNS=9.9.9.9
 
 # Hostname
-AUTO_SETUP_NET_HOSTNAME=DietPi
+AUTO_SETUP_NET_HOSTNAME=name
 
 ##### Misc Options #####
 # Swapfile size to generate: 0=disable | 1=auto (2GB-RAM = size) | 2+=manual (MB)
@@ -93,7 +93,7 @@
 # On first login, run update, initial setup and software installs without any user input
 # - Setting this to "1" is required for below settings to take effect
 # - It is HIGHLY recommended to also set CONFIG_BOOT_WAIT_FOR_NETWORK=2, to force infinite wait for network connection during boot and prevent connection timeout errors.
-AUTO_SETUP_AUTOMATED=0
+AUTO_SETUP_AUTOMATED=1
 
 # Global Password to be applied for the system
 # - Affects user "root" and "dietpi" login passwords, and, all software installed by dietpi-software, that requires a login password
@@ -105,7 +105,8 @@
 # - Add as many entries as you wish, one each line.
 # - DietPi will automatically install all pre-reqs (e.g. ALSA/X11 for desktops etc)
 # - E.g. the following (without the leading "#") will install the LXDE desktop automatically on first boot:
-#AUTO_SETUP_INSTALL_SOFTWARE_ID=23
+AUTO_SETUP_INSTALL_SOFTWARE_ID=5
+AUTO_SETUP_INSTALL_SOFTWARE_ID=121
 
 #------------------------------------------------------------------------------------------------------
 # D I E T - P I
@@ -113,7 +114,7 @@
 #------------------------------------------------------------------------------------------------------
 # DietPi-Survey: 1=opt in | 0=opt out | -1=ask on first call
 # - https://dietpi.com/phpbb/viewtopic.php?p=34#p34
[... 279 lines cut ...]

# Chromium
SOFTWARE_CHROMIUM_RES_X=1280
SOFTWARE_CHROMIUM_RES_Y=720
SOFTWARE_CHROMIUM_AUTOSTART_URL=https://dietpi.com

# Home Assistant
# - Optional Python build dependencies and modules, possibly required for certain HA components
#	Space separated list (no quotation!), will be installed together with Home Assistant automatically, if present
SOFTWARE_HOMEASSISTANT_APT_DEPS=
#	Add Python modules with version string at best, e.g.: firstModule==1.2.3 secondModule==4.5.6
SOFTWARE_HOMEASSISTANT_PIP_DEPS=

#------------------------------------------------------------------------------------------------------
# D I E T - P I
# Dev settings
#------------------------------------------------------------------------------------------------------
DEV_GITBRANCH=master
DEV_GITOWNER=MichaIng

#------------------------------------------------------------------------------------------------------
# D I E T - P I
# Settings, automatically added by dietpi-update
#------------------------------------------------------------------------------------------------------
```

dsnyder0pc, 2020-06-13:
**Diff showing my changes to dietpi-wifi.txt to enable Wi-Fi boot (of course, I used the actual SSID and password):**

```
--- /Users/dsnyder/tmp/dietpi-wifi.txt	2020-06-08 10:08:00.000000000 -0700
+++ dietpi-wifi.txt	2020-06-13 16:01:12.000000000 -0700
@@ -1,9 +1,9 @@
 #---------------------------------------------------------------
 # Entry 0
 # - WiFi SSID: required, case sensitive
-aWIFI_SSID[0]=''
+aWIFI_SSID[0]='my_ssid'
 # - WiFi key: If no key/open, leave this blank
-aWIFI_KEY[0]=''
+aWIFI_KEY[0]='my_password'
 # - Key type: NONE (no key/open) | WPA-PSK | WEP | WPA-EAP (then use settings below)
 aWIFI_KEYMGR[0]='WPA-PSK'
 # - WPA-EAP options: Only fill if WPA-EAP is set above
```

**Complete dietpi-wifi.txt file here:**
```
#---------------------------------------------------------------
# Entry 0
# - WiFi SSID: required, case sensitive
aWIFI_SSID[0]='my_ssid'
# - WiFi key: If no key/open, leave this blank
aWIFI_KEY[0]='my_password'
# - Key type: NONE (no key/open) | WPA-PSK | WEP | WPA-EAP (then use settings below)
aWIFI_KEYMGR[0]='WPA-PSK'
# - WPA-EAP options: Only fill if WPA-EAP is set above
aWIFI_PROTO[0]=''
aWIFI_PAIRWISE[0]=''
aWIFI_AUTH_ALG[0]=''
aWIFI_EAP[0]=''
aWIFI_IDENTITY[0]=''
aWIFI_PASSWORD[0]=''
aWIFI_PHASE1[0]=''
aWIFI_PHASE2[0]=''
# - Path to the certificate file, e.g.: /boot/mycert.cer
aWIFI_CERT[0]=''
#---------------------------------------------------------------
# Entry 1
# - WiFi SSID: required, case sensitive
aWIFI_SSID[1]=''
# - WiFi key: If no key/open, leave this blank
aWIFI_KEY[1]=''
# - Key type: NONE (no key/open) | WPA-PSK | WEP | WPA-EAP (then use settings below)
aWIFI_KEYMGR[1]='WPA-PSK'
# - WPA-EAP options: Only fill if WPA-EAP is set above
aWIFI_PROTO[1]=''
aWIFI_PAIRWISE[1]=''
aWIFI_AUTH_ALG[1]=''
aWIFI_EAP[1]=''
aWIFI_IDENTITY[1]=''
aWIFI_PASSWORD[1]=''
aWIFI_PHASE1[1]=''
aWIFI_PHASE2[1]=''
# - Path to the certificate file, e.g.: /boot/mycert.cer
aWIFI_CERT[1]=''
#---------------------------------------------------------------
# Entry 2
# - WiFi SSID: required, case sensitive
aWIFI_SSID[2]=''
# - WiFi key: If no key/open, leave this blank
aWIFI_KEY[2]=''
# - Key type: NONE (no key/open) | WPA-PSK | WEP | WPA-EAP (then use settings below)
aWIFI_KEYMGR[2]='WPA-PSK'
# - WPA-EAP options: Only fill if WPA-EAP is set above
aWIFI_PROTO[2]=''
aWIFI_PAIRWISE[2]=''
aWIFI_AUTH_ALG[2]=''
aWIFI_EAP[2]=''
aWIFI_IDENTITY[2]=''
aWIFI_PASSWORD[2]=''
aWIFI_PHASE1[2]=''
aWIFI_PHASE2[2]=''
# - Path to the certificate file, e.g.: /boot/mycert.cer
aWIFI_CERT[2]=''
#---------------------------------------------------------------
# Entry 3
# - WiFi SSID: required, case sensitive
aWIFI_SSID[3]=''
# - WiFi key: If no key/open, leave this blank
aWIFI_KEY[3]=''
# - Key type: NONE (no key/open) | WPA-PSK | WEP | WPA-EAP (then use settings below)
aWIFI_KEYMGR[3]='WPA-PSK'
# - WPA-EAP options: Only fill if WPA-EAP is set above
aWIFI_PROTO[3]=''
aWIFI_PAIRWISE[3]=''
aWIFI_AUTH_ALG[3]=''
aWIFI_EAP[3]=''
aWIFI_IDENTITY[3]=''
aWIFI_PASSWORD[3]=''
aWIFI_PHASE1[3]=''
aWIFI_PHASE2[3]=''
# - Path to the certificate file, e.g.: /boot/mycert.cer
aWIFI_CERT[3]=''
#---------------------------------------------------------------
# Entry 4
# - WiFi SSID: required, case sensitive
aWIFI_SSID[4]=''
# - WiFi key: If no key/open, leave this blank
aWIFI_KEY[4]=''
# - Key type: NONE (no key/open) | WPA-PSK | WEP | WPA-EAP (then use settings below)
aWIFI_KEYMGR[4]='WPA-PSK'
# - WPA-EAP options: Only fill if WPA-EAP is set above
aWIFI_PROTO[4]=''
aWIFI_PAIRWISE[4]=''
aWIFI_AUTH_ALG[4]=''
aWIFI_EAP[4]=''
aWIFI_IDENTITY[4]=''
aWIFI_PASSWORD[4]=''
aWIFI_PHASE1[4]=''
aWIFI_PHASE2[4]=''
# - Path to the certificate file, e.g.: /boot/mycert.cer
aWIFI_CERT[4]=''
```

dsnyder0pc, 2020-06-13:
Verification of issue with default values for. CONFIG_CHECK_CONNECTION_IP and CONFIG_CHECK_DNS_DOMAIN: https://youtu.be/eUxdUlSOWuc

StephanStS, 2020-06-14:
@dsnyder0pc: Could this be similar to #3583?
Could you check [this area](https://github.com/MichaIng/DietPi/issues/3583#issuecomment-643415380) of the issue?

dsnyder0pc, 2020-06-14:
I re-tested with the Jan 26th image for RPi ARMv6 Buster. First attempt failed because I had set `CONFIG_NTP_MODE` to `0` as part of my troubleshooting of the June 8th build. SSL Cert check fails when checking the mirror with `certificate is not yet valid`. After I re-enabling NTP, the January 26th image boots up on Wi-Fi just fine. I'll add a video link here in a bit.

dsnyder0pc, 2020-06-14:
> @dsnyder0pc: Could this be similar to #3583?
> Could you check [this area](https://github.com/MichaIng/DietPi/issues/3583#issuecomment-643415380) of the issue?

@StephanStS - Looking…thanks.

Joulinar, 2020-06-14:
I guess that's the issue

`RTNETLINK answers: Operation not possible due to RF-kill`

loos like the interface is soft blocked. A workaround could be following

```
rfkill list all
rfkill unblock all
ip link set wlan0 up
```

dsnyder0pc, 2020-06-14:
> I guess that's the issue
> 
> `RTNETLINK answers: Operation not possible due to RF-kill`
> 
> loos like the interface is soft blocked. A workaround could be following. Once hit by the issue, just exit the setup and do:
> 
> ```
> rfkill list all
> rfkill unblock all
> ip link set wlan0 up
> /boot/dietpi/dietpi-login
> ```
> 
> On my RPi3B+ it's working

Helpful. I'll try putting those commands into `/boot/Automation_Custom_PreScript.sh` to see if that will get me past first boot on Wi-Fi. In case it's not obvious, the idea that I'm going after is a hands-off setup process once the microSD card is flashed and dietpi.txt/dietpi-wifi.txt are prepared. Thanks for suggestion.

Joulinar, 2020-06-14:
As well you can use the image from January still. 
The June image was just created to quickly add support for the new RPi4 8GB

In any case you will be updated to latest DietPi as well as all available updates for Raspberry OS will be applied.

dsnyder0pc, 2020-06-14:
> As well you can use the image from January still.
> The June image was just created to quickly add support for the new RPi4 8GB
> 
> In any case you will be updated to latest DietPi as well as all available updates for Raspberry OS will be applied.

Is it still possible for folks to download the January 26th image? I don't see it available anymore. I've stashed a copy for folks to use who are following my Roon Bridge setup instructions, but that's not ideal. I'll let you know how this preScript test goes.

Joulinar, 2020-06-14:
nope the image is not there on the official server. But if needed I can provide a private copy. Rgeading your script. The `rfkill unblock all` would need to be applied on quite an early state to allow WiFi to connect.

dsnyder0pc, 2020-06-14:
> nope the image is not there on the official server. But if needed I can provide a private copy. Regarding your script. The `rfkill unblock all` would need to be applied on quite an early state to allow WiFi to connect.

Got it. I have a copy of the Jan 26th image on my own server. My challenge is that I have a few dozen folks every month who use my [dietpi-setup script](https://github.com/dsnyder0pc/DietPi) to customize their DietPi installation to create [Roon Bridge](https://roonlabs.com/) endpoints. They are sad that this is not possible to do over Wi-Fi with the June 8th image. I just spent a few hours working with one of them over Zoom this afternoon. I'll send them to my private copy for now.

From what you are saying, it sounds like the `Automation_Custom_PreScript.sh` won't be run early enough. Comments in `dietpi.txt` suggested that it _might_ work, but I trust you.  :)

```
# Custom Script (pre-networking and pre-DietPi install)
# - Allows you to automatically execute a custom script before network is up on first boot.
# - Copy your script to /boot/Automation_Custom_PreScript.sh and it will be executed automatically.
# - Executed script log: /var/tmp/dietpi/logs/dietpi-automation_custom_prescript.log
```

Thanks for adding the tags and for your follow-up. Hoping this is something that can be sorted soon. I'm happy to help with testing.

dsnyder0pc, 2020-06-14:
@Joulinar - I have some partial good news. I modified my dietpi-setup script to drop a `Automation_Custom_PreScript.sh` file onto the microSD card after making its edits. Contents are just this:
```
rfkill list all
rfkill unblock all
ip link set wlan0 up
```
(diff [here](https://github.com/dsnyder0pc/DietPi/commit/bbe656feccec0e2b70e8e0df326c1a465de45e53))

Apparently, the preScript script actually does run early enough in the boot process to have the desired effect. :-). It worked for my RPi4. I'll test the RPi3 A+ in a bit (did the RPi4 first because iterations are a bit quicker).

Thanks so much for the suggestion!

dsnyder0pc, 2020-06-14:
@Joulinar - I have some partial good news. I modified my dietpi-setup script to drop a `Automation_Custom_PreScript.sh` file onto the microSD card after making its edits. Contents are just this:
```
rfkill list all
rfkill unblock all
ip link set wlan0 up
```
(diff [here](https://github.com/dsnyder0pc/DietPi/commit/bbe656feccec0e2b70e8e0df326c1a465de45e53))

Apparently, the preScript script actually does run early enough in the boot process to have the desired effect. :-). It worked for my RPi3 A+. Video here: https://youtu.be/vSymzrHXtLE

Thanks so much for the suggestion!

dsnyder0pc, 2020-06-14:
Success with RPi4 as well!  https://youtu.be/4thlkQfcXsA

[diff](https://github.com/dsnyder0pc/DietPi/commit/5ceb3ba170855d6716426fb0f206980d0b4393f9) updated slightly to only create the preScript if Wi-Fi is used.  :)

MichaIng, 2020-06-14:
Hey guys, many thanks for reporting. `rfkill` is driving me nuts... It's too bad that AFAIK removing the userland binary does not automatically unblock all devices since this is a kernel feature. Just checked it:
[image]

Since we never block (only unblock as failsafe step when enabling) WiFi, this must be the case for the current Raspberry Pi OS (32-bit) image already.

Okay I always wanted to assure that `rfkill unblock all` and then do that within DietPi-PREP, manually remove the store files for systemd-rfkill and remove the rfkill package then from our images. Let's see if removing the systemd-rfkill files is enough for a quick fix on the RPi image.

StephanStS, 2020-06-14:
> 
> @StephanStS - Not the same issue in my case. The RPi3 A+ has no support for wired Ethernet, so it's not possible to connect a LAN cable. First boot using Wi-Fi is the only way to install DietPi. :-\

@dsnyder0pc: For these issues I gain an adapter USB to Ethernet (one with USB-A, e.g. for my 3A+ and one with Micro-USB, e.g. for my Pi Zero W. They often helped quite much for the "first time setup until WiFi runs".

MichaIng, 2020-06-14:
#### New image is up without rfkill soft block in place: https://dietpi.com/downloads/images/DietPi_RPi-ARMv6-Buster.7z

dsnyder0pc, 2020-06-14:
> #### New image is up without rfkill soft block in place: https://dietpi.com/downloads/images/DietPi_RPi-ARMv6-Buster.7z

Cool. I'll test in a bit on my RPi4 and RPi 3 A+ and let you know how it goes. Thanks!

dsnyder0pc, 2020-06-14:
> #### New image is up without rfkill soft block in place: https://dietpi.com/downloads/images/DietPi_RPi-ARMv6-Buster.7z

Okay. I've re-run the tests. First time I tried without my workaround script in place, it failed, but I was not recording then. Not sure what happened. Likely something on my end. Here are videos of my test results in case you're curious about how things went in more detail.

https://www.youtube.com/playlist?list=PLMl09rJ6zKCnYsxyD67BP4Cy7rF8Nq9YK

Thanks again for your help. Good hunting!

MichaIng, 2020-06-15:
Many thanks for testing and creating this nice, I'd say professional, recordings. Since the second attempt without workaround had no blocked WiFi I guess the first time you might have gotten a Cloudflare-cached version of the old image. I forced a cache reset for this file but not sure how immediate this happens 😉.

Now that I see how long the time sync can take, I get the idea to bring it back to background. The problem is that simply using `&` allows the script to continue, but the service will still wait for it to finish before boot continues. Hopefully `nohup` and/or `disown` allows this. It can then fail silently, however in headless environment its just the same now and `dietpi-update` + `dietpi-software` rerun it anyway if not successful before.

closed 2020-07-09 by MichaIng