MichaIng/DietPi #3602: DietPi-Software | WireGuard - add dependency

Joulinar, 2020-06-14:
@MichaIng 
I guess we need to add `resolvconf` package as dependency for WireGuard. Becasue it seems to be required if WireGuard is working within client mode:

`Jun 14 21:35:46 DietPi3 wg-quick[540]: /usr/bin/wg-quick: line 32: resolvconf: command not found`

https://dietpi.com/phpbb/viewtopic.php?f=11&t=7783

As far as I can see `openresolv` is getting removed during PREP script execution

`Purging configuration files for openresolv (3.8.0-1) ...`

```
root@DietPi3:/etc/wireguard# dpkg -l *resolv*
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name           Version      Architecture Description
+++-==============-============-============-=================================
un  openresolv     <none>       <none>       (no description available)
un  resolvconf     <none>       <none>       (no description available)
root@DietPi3:/etc/wireguard#
```

MichaIng, 2020-06-14:
Indeed, it's just by wg-quick without being checked first but the package only has it as suggestion: https://packages.debian.org/buster-backports/wireguard-tools
I'll add it to dietpi-software when installing WireGuard as client.

MichaIng, 2020-06-14:
Done: https://github.com/MichaIng/DietPi/commit/8c33bbae8f9d583369951c33eaed83014008d823
Changelog: https://github.com/MichaIng/DietPi/commit/aeb8b5e3a0000ca73c114fc8b5841fb98195d8a3

Joulinar, 2020-06-14:
@MichaIng 
just a side note. Even if you go for a server setup, you could simply use it as client later on. Therefore it might be better to install `resolvconf` packe independent from usage typ. What do you thing?

MichaIng, 2020-06-14:
I'll add some note to the online docs instead. I'm not keen to install it for such edge case since it changes the way one must configure DNS nameservers significantly and I don't want to force this when not being sure that it is required in ~90% of the cases.

Joulinar, 2020-06-14:
ok agree, good argument.

MichaIng, 2020-06-15:
Done: https://dietpi.com/phpbb/viewtopic.php?p=16308#p16308
Formatting could be enhanced, but the note is inside the client setup part.

closed 2020-06-30 by MichaIng