MichaIng/DietPi #3604: DietPi-Software | HAProxy - update to latest version

Joulinar, 2020-06-14:
@MichaIng 
One more to be updated. Current version 2.1.3 seems to fail during `make`.

```
collect2: error: ld returned 1 exit status
make: *** [Makefile:857: haproxy] Error 1
strip: 'haproxy': No such file
Please run 'make' before 'make install'.
make: *** [Makefile:899: install-bin] Error 1
```
Changing to version 2.1.7 and `make` is finishing without issues.
https://github.com/MichaIng/DietPi/blob/d29a5af277aa1e6b76b4b3057968f0885894e61d/dietpi/dietpi-software#L4276

I was facing the issue during error investigation on our forum
https://dietpi.com/phpbb/viewtopic.php?p=25407#p25407

MichaIng, 2020-06-14:
Many thanks for forwarding. I'll have a look tomorrow.

MichaIng, 2020-06-17:
#### Done: https://github.com/MichaIng/DietPi/commit/8ab237f8d272f623bf982315ebdb6d696313ac77

Although I'm wondering how the last version build failed. I added quite a few build flags from Debian build system, I'll recheck those and test on RPi explicitly.

closed 2020-06-30 by MichaIng