MichaIng/DietPi #3606: Can't update from v6.28.0 to v6.30.0

MAT4150, 2020-06-15:
APT update
│  - Command: apt-get -q update
│  - Exit code: 100
│  - DietPi version: v6.28.0 (MichaIng/master) | HW_MODEL: 3 | HW_ARCH: 2 |
│ DISTRO: 4
│  - Error log:
│ Hit:1 http://raspbian.raspberrypi.org/raspbian stretch InRelease
│ Ign:2 http://download.opensuse.org/repositories/home:/strycore/xUbuntu_16.04
│ InRelease
│ Err:3 http://download.opensuse.org/repositories/home:/strycore/xUbuntu_16.04
│ Release
│   404  Not Found
│ Hit:4 http://deb.playonlinux.com trusty InRelease
│ Hit:5 https://archive.raspberrypi.org/debian stretch InRelease
│ Reading package lists...

Joulinar, 2020-06-15:
Hi,

Many thanks for your report. I guess you opened same issue at our forum? Correct?

MAT4150, 2020-06-15:
I did, yes. Wasn't sure which place was better to post. Very new to this!

Joulinar, 2020-06-15:
Doesn't matter. As already indicated, your source list would need to be corrected. Correctly I'm aware from my home. So will look into it later on. Maybe @MichaIng will be faster. We will see :wink:

MichaIng, 2020-06-15:
Many thanks for your report.

> http://download.opensuse.org/repositories/home:/strycore/xUbuntu_16.04

Where did you get this OpenSUSE/Ubuntu repository from? It does not exist. There are other even Raspbian/Debian repos available on this URL: http://download.opensuse.org/repositories/home:/strycore/

What I find nearly irritating is that I faced the nearly same error report on superuser.com just a few days ago. See my answer there (the one at the bottom): https://superuser.com/q/1557343/1183051

MichaIng, 2020-06-17:
Okay to bring this forward @MAT4150:
Do you use the **Lutris** platform to play games on your RPi? https://lutris.net/
And please run the following on your terminal paste the output here:
```
for i in /etc/apt/sources.list{,.d/*.list}
do
echo ${i}:
cat $i
done
```

MichaIng, 2020-07-05:
I mark this issue as closed due to missing reply. Offending sources.list entry is the reason, hence not a DietPi issue. However feel free to reopen if issue persists an we'll help solving it.

closed 2020-07-05 by MichaIng