MichaIng/DietPi #3607: Error while trying to install mono on my rPi

EmperiuM666, 2020-06-15:
#### Details:
- Date           | Mon Jun 15 17:56:06 BST 2020
- Bug report     | 3c4564b8-f5cd-427d-9095-87ebcb2b2192
- DietPi version | v6.30.0 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspberry Pi OS (32-bit) Lite
- Hardware       | RPi 2 Model B (armv7l) (ID=2)
- Kernel version | Linux DietPi 4.19.118-v7+ #1311 SMP Mon Apr 27 14:21:24 BST 2020 armv7l GNU/Linux
- Distro         | buster (ID=5)
- Command        | apt-get -qq --allow-change-held-packages install ./170.deb
- Exit code      | 100
- Software title | DietPi-Software
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
E: Unsupported file ./170.deb given on commandline
```

Joulinar, 2020-06-15:
Hi,

many thanks for your report. Are you sure you are trying to install `Mono`? Because according your error message, you are trying to install software ID 170. And this is related to `UnRAR`. Anyway I tried to install and it was working without issues. Pls can you try again and post some more lines.

 ```
DietPi-Software
─────────────────────────────────────────────────────
 Mode: Installing UnRAR: unarchiver for .rar files

[  OK  ] DietPi-Software | Checking URL: https://dietpi.com/downloads/binaries/rpi/unrar-armhf-buster.deb
[  OK  ] DietPi-Software | wget https://dietpi.com/downloads/binaries/rpi/unrar-armhf-buster.deb -O 170.deb
[ INFO ] DietPi-Software | APT install for: ./170.deb, please wait...
Selecting previously unselected package unrar.
(Reading database ... 21082 files and directories currently installed.)
Preparing to unpack /tmp/DietPi-Software/170.deb ...
Unpacking unrar (1:5.6.6-1) ...
Setting up unrar (1:5.6.6-1) ...
update-alternatives: using /usr/bin/unrar-nonfree to provide /usr/bin/unrar (unrar) in auto mode
[  OK  ] DietPi-Software | APT install for: ./170.deb
```

MichaIng, 2020-06-16:
To test the file directly:
```
cd /tmp
wget https://dietpi.com/downloads/binaries/rpi/unrar-armhf-buster.deb
apt-get install ./unrar-armhf-buster.deb
rm unrar-armhf-buster.deb
```

EmperiuM666, 2020-06-16:
@Joulinar, it was just after `Mono` and yes, it was `UnRar`, sorry.
@MichaIng it worked like you said.
`Reading package lists... Done
Building dependency tree       
Reading state information... Done
Note, selecting 'unrar' instead of './unrar-armhf-buster.deb'
The following NEW packages will be installed:
  unrar
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
Need to get 0 B/100 kB of archives.
After this operation, 313 kB of additional disk space will be used.
Get:1 /tmp/unrar-armhf-buster.deb unrar armhf 1:5.6.6-1 [100 kB]
Selecting previously unselected package unrar.
(Reading database ... 49833 files and directories currently installed.)
Preparing to unpack /tmp/unrar-armhf-buster.deb ...
Unpacking unrar (1:5.6.6-1) ...
Setting up unrar (1:5.6.6-1) ...
update-alternatives: using /usr/bin/unrar-nonfree to provide /usr/bin/unrar (unrar) in auto mode`

Joulinar, 2020-06-16:
@EmperiuM666 
This looks like installation went fine this time. Isn't it? At least no error message present.

EmperiuM666, 2020-06-16:
@Joulinar yes. probably it was any orphan thing on the previous install.
Sorry about open the bug.
Thank you

MichaIng, 2020-06-16:
No worries, sometimes we face temporary issues with network connection or server-side or the heavy mono compiling filled RAM which broke `/tmp` file write or something, but difficult to know that when running into such error. If you still have the original install attempt in your console scroll log we might find a hint, otherwise we keep it in mind just in case a similar report appears later 🙂.

closed 2020-06-17 by MichaIng