MichaIng/DietPi #3609: dietpi-cleaner: mount directory being scanned while Include mount directory is disabled ???

maartenlangeveld, 2020-06-16:
While I have disabled scanning of mount directory (/mnt/*) in diepi-cleaner:

```
                    ┌──────────────────────────────────────────────────┤ DietPi-Cleaner ├──────────────────────────────────────────────────┐
                    │ Files: Cleaner options.                                                                                              │
                    │ (NB) The cleaner named "Files" must be enabled for this to work.                                                     │
                    │                                                                                                                      │
                    │                            Include mount directory (/mnt/*) during file scan  : Disabled                             │
                    │                            Modify filenames/extensions to include during scan.                                       │
                    │                                                                                                                      │
                    │                                                                                                                      │
                    │                                  <Ok>                                      <Cancel>                                  │
                    │                                                                                                                      │
                    └──────────────────────────────────────────────────────────────────────────────────────────────────────────────────────┘

```

mount directories seems to be scanned though, I discovered after I had unmounted a mounted NFS share manually:

root@DietPi:~# umount /mnt/rpi-zero-w-backup-xigmanas

`root@DietPi:~# dietpi-cleaner 1`

`find: ‘/mnt/rpi-zero-w-backup-xigmanas’: No such device`

MichaIng, 2020-06-16:
Many thanks for your report. I checked the code and indeed everything is scanned first while the /mnt entries are removed afterwards from results: https://github.com/MichaIng/DietPi/blob/dev/dietpi/dietpi-cleaner#L458-L461

Highly inefficient indeed... Better to exclude and skip /mnt with find in the first place.

MichaIng, 2020-06-16:
#### This should do the job, I'll run some tests: https://github.com/MichaIng/DietPi/commit/81753f4f8eaa0ee0fa32d6ae0f7a7769596485e0

maartenlangeveld, 2020-06-16:
@MichaIng 

**Thanks ! It does the job.**
dietpi-cleaner runs much faster now :-)

Since dietpi-cleaner was so slow I got the impression that the mounts were scanned although I disabled scanning /mnt/ directory (I have some large filesystems mounted). That's why I manually unmounted all /mnt/ mounted filesystems and scanning was still slow... I then noticed `find: ‘/mnt/rpi-zero-w-backup-xigmanas’: No such device
`

It appears to me that it's not worth a changelog entry, but I guess it's up to you to decide.

MichaIng, 2020-06-16:
Lets honour it, especially when long search times force users to take action like unmounting drives/shares as you did: https://github.com/MichaIng/DietPi/commit/590ee877ff863a3ab6120c55ccca7ff021ef78d3

Btw umount can hang on umounting network drives, or it can fail silently if the drive is currently not available/connection is down. To force removal of any linked files use `umount -f` or `umount -l`: https://manpages.debian.org/buster/mount/umount.8.en.html#OPTIONS
I guess this was the reason for the (lucky for us) error.

maartenlangeveld, 2020-06-16:
Thanks for your sound advice on unmounting network drives. Did not know that. I keep learning -)

MichaIng, 2020-06-17:
Okay I mark this as closed/done, feel free to reopen if required.

closed 2020-06-17 by MichaIng