MichaIng/DietPi #3611: Unable to get Bluetooth working on rpi1 with Dietpi 6.28.0

TASSDevon, 2020-06-17:
#### Required Information
- DietPi version 6.28.0
- Distro version buster
- Kernel version Linux skyspeaker 4.19.97+ #1294 Thu Jan 30 13:10:54 GMT 2020 armv6l GNU/Linux
- SBC model rpi1
#### Additional Information (if applicable)

Trying to get a BT dongle from AliExpress working on my rpi1 with Dietpi installed. I used dietpi-config to enable bluetooth. starting hciuart failed. I then tried a couple of suggestions from earlier issues but no luck, I do not get a /dev/ttyAMA0

This is what the hciuart service spews out:

```
● hciuart.service - Configure Bluetooth Modems connected by UART
   Loaded: loaded (/lib/systemd/system/hciuart.service; enabled; vendor preset: enabled)
   Active: inactive (dead)

Jun 17 17:53:27 skyspeaker systemd[1]: Dependency failed for Configure Bluetooth Modems connect
ed by UART.
Jun 17 17:53:27 skyspeaker systemd[1]: hciuart.service: Job hciuart.service/start failed with r
esult 'dependency'.
Jun 17 17:58:05 skyspeaker systemd[1]: Dependency failed for Configure Bluetooth Modems connect
ed by UART.
Jun 17 17:58:05 skyspeaker systemd[1]: hciuart.service: Job hciuart.service/start failed with r
esult 'dependency'.
```

Any ideas?

MichaIng, 2020-06-17:
> hciuart.service - Configure Bluetooth Modems connected by UART

Since you're on RPi1 without internal BT but with a BT dongle, it is connected via USB and not via UART, hence you don't need to enable the UART device or hciuart.service, nor install `pi-bluetooth` which is only for RPi3/4 internal Bluetooth.

Little research: https://rasspberrypi.wordpress.com/2012/09/03/install-bluetooth-dongle-on-raspberry-pi/
```
apt purge pi-bluetooth
apt install bluez
systemctl disable --now hciuart
systemctl enable --now bluetooth
hcitool scan # Probably it already works
```

Probably we need to change dietpi-config to differentiate between RPi devices with and without internal Bluetooth.

TASSDevon, 2020-06-17:
Thanks @MichaIng, tried your suggestion and it works (did a scan and found my phone).

The hciuart service didn't need to be disabled as it was removed (I guess by purging pi-bluetooth package).

Tomorrow I'll see if I can get that BT link up and running with the Jabra speaker :-)

MichaIng, 2020-06-18:
> The hciuart service didn't need to be disabled as it was removed (I guess by purging pi-bluetooth package).

Ah yes that makes sense.

> Tomorrow I'll see if I can get that BT link up and running with the Jabra speaker :-)

Would be great. I'll add generic Bluetooth support then for all devices where we don't know about an internal solution. Long-term also for the ones with internal BT it would be nice to allow usage of a USB adapter, a bid like you can disable internal WiFi but still enable general WiFi.

TASSDevon, 2020-06-18:
Using bluetoothctl I'm not able to connect to the speaker unfortunately:

```
[bluetooth]# connect 10:B7:F6:28:24:7E
Attempting to connect to 10:B7:F6:28:24:7E
Failed to connect: org.bluez.Error.Failed
[bluetooth]# info 10:B7:F6:28:24:7E
Device 10:B7:F6:28:24:7E (public)
	Name: Jabra Solemate Mini v8.0.0
	Alias: Jabra Solemate Mini v8.0.0
	Class: 0x00240404
	Icon: audio-card
	Paired: yes
	Trusted: yes
	Blocked: no
	Connected: no
	LegacyPairing: no
	UUID: Headset                   (00001108-0000-1000-8000-00805f9b34fb)
	UUID: Audio Sink                (0000110b-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
	UUID: Handsfree                 (0000111e-0000-1000-8000-00805f9b34fb)
[bluetooth]# paired-devices
Device 10:B7:F6:28:24:7E Jabra Solemate Mini v8.0.0
[bluetooth]# connect 10:B7:F6:28:24:7E
Attempting to connect to 10:B7:F6:28:24:7E
Failed to connect: org.bluez.Error.Failed
```

Any ideas? I managed to pair it, but it's not establishing a connection

MichaIng, 2020-06-18:
Probably `bluez-tools` can, although they seem to be just wrapper for what you can do within bluetoothctl already... https://packages.debian.org/buster/amd64/bluez-tools/filelist

Do `dmesg` or `journalctl -u bluetooth` give a hint?

TASSDevon, 2020-06-19:
Nothing BT related in dmesg, but lots of errors in journalctl for bt:

```
 for 10:B7:F6:28:24:7E: Protocol not available
Jun 18 18:08:55 skyspeaker bluetoothd[198]: Endpoint registered: sender=:1.22 path=/MediaEndpoint/A2
DPSource
Jun 18 18:08:55 skyspeaker bluetoothd[198]: Endpoint registered: sender=:1.22 path=/MediaEndpoint/A2
DPSink
Jun 18 18:08:55 skyspeaker bluetoothd[198]: RFCOMM server failed for Headset
 Voice gateway: socket(STREAM, RFCOMM): Protocol not supported (93)
Jun 18 18:08:55 skyspeaker bluetoothd[198]: RFCOMM server failed for :1.22/P
rofile/HSPHSProfile/00001108-0000-1000-8000-00805f9b34fb: socket(STREAM, RFCOMM): Protocol not suppo
rted (93)
Jun 18 18:12:31 skyspeaker bluetoothd[198]: Unable to get Headset Voice gate
way SDP record: Host is down
Jun 18 18:13:01 skyspeaker bluetoothd[198]: Endpoint unregistered: sender=:1.22 path=/MediaEndpoint/
A2DPSource
Jun 18 18:13:01 skyspeaker bluetoothd[198]: Endpoint unregistered: sender=:1.22 path=/MediaEndpoint/
A2DPSink
Jun 18 18:14:59 skyspeaker bluetoothd[198]: a2dp-sink profile connect failed
 for 10:B7:F6:28:24:7E: Protocol not available
Jun 18 18:15:07 skyspeaker bluetoothd[198]: a2dp-sink profile connect failed
 for 10:B7:F6:28:24:7E: Protocol not available
Jun 18 18:15:16 skyspeaker bluetoothd[198]: a2dp-sink profile connect failed
 for 10:B7:F6:28:24:7E: Protocol not available
Jun 18 18:15:21 skyspeaker bluetoothd[198]: a2dp-sink profile connect failed
 for 10:B7:F6:28:24:7E: Protocol not available
Jun 18 18:15:43 skyspeaker bluetoothd[198]: a2dp-sink profile connect failed
 for 10:B7:F6:28:24:7E: Protocol not available
Jun 18 18:16:07 skyspeaker bluetoothd[198]: a2dp-sink profile connect failed
 for 10:B7:F6:28:24:7E: Protocol not available
Jun 18 18:16:10 skyspeaker bluetoothd[198]: a2dp-sink profile connect failed
 for 10:B7:F6:28:24:7E: Protocol not available
Jun 18 18:16:11 skyspeaker bluetoothd[198]: a2dp-sink profile connect failed
 for 10:B7:F6:28:24:7E: Protocol not available
Jun 18 18:16:13 skyspeaker bluetoothd[198]: a2dp-sink profile connect failed
 for 10:B7:F6:28:24:7E: Protocol not available
Jun 18 18:19:00 skyspeaker bluetoothd[198]: a2dp-sink profile connect failed
 for 10:B7:F6:28:24:7E: Protocol not available
```

MichaIng, 2020-06-19:
Ah, I am dumb, Bluez does not support ALSA natively since v5, but you can install it: https://github.com/Arkq/bluez-alsa
And see here for ALSA configuration: https://github.com/MichaIng/DietPi/issues/3456

The alternative is indeed to go with PulseAudio and use the related Bluetooth module: https://packages.debian.org/buster/pulseaudio-module-bluetooth

TASSDevon, 2020-06-20:
Thanks for the hint, I'll give it a try today or tomorrow and report back.

MichaIng, 2020-06-21:
> One thing I did notice was that sqeezelite service was not running after a reboot, not sure if it is related but I've never seen it not running before.

Please check: `journalctl -u squeezelite`

> Any reason for the 10s delay in the settings? Seems quite long.

In which settings you mean?

> do you perhaps have an example on what should go into the service file to get this running from boot?

The official docs about it look very complete, including ondemand activation via dbus 👍: https://github.com/Arkq/bluez-alsa/wiki/Systemd-integration

TASSDevon, 2020-06-21:
Nothing out of the ordinary

```
root@skyspeaker:~# journalctl -u squeezelite
-- Logs begin at Sat 2020-06-20 17:17:03 BST, end at Sun 2020-06-21 18:54:03 BST. --
Jun 20 17:44:18 skyspeaker systemd[1]: Started SqueezeLite (DietPi).
Jun 20 17:44:19 skyspeaker systemd[1]: squeezelite.service: Succeeded.
Jun 20 17:45:31 skyspeaker systemd[1]: Started SqueezeLite (DietPi).
```

I did a systemctl enable squeezelite and it created a symlink, for some reason the service was disabled at boot. I guess we can disregard this ;-)

The 10 second setting is from the issue you linked to, your comment: https://github.com/MichaIng/DietPi/issues/3456#issuecomment-608409922

Thanks for the link to the systemd integration, must've missed that one.

MichaIng, 2020-06-21:
> I did a systemctl enable squeezelite and it created a symlink, for some reason the service was disabled at boot. I guess we can disregard this ;-)

Ah actually `dietpi-services` should start all services including `squeezelite` bundled at the end of boot (`dietpi-postboot.service`). To check and test:
```
systemctl status dietpi-postboot
/boot/dietpi/postboot
```
If you see `Active: active (exited)` and `[  OK  ] DietPi-Services | start : squeezelite` then it is not required (but not an issue as well) to have systemd starting it.

TASSDevon, 2020-06-22:
I'm not sure where to see the log line with the "DietPi-Services start" part. The service status is this which looks OK:

```
root@skyspeaker:~# systemctl status dietpi-postboot
● dietpi-postboot.service - DietPi-PostBoot
   Loaded: loaded (/etc/systemd/system/dietpi-postboot.service; enabled; vendor preset: enabled
)
   Active: active (exited) since Sun 2020-06-21 19:03:10 BST; 11h ago
  Process: 518 ExecStart=/boot/dietpi/postboot (code=exited, status=0/SUCCESS)
 Main PID: 518 (code=exited, status=0/SUCCESS)

Jun 21 19:03:10 skyspeaker systemd[1]: Started DietPi-PostBoot.
```

MichaIng, 2020-06-22:
When you run `/boot/dietpi/postboot`, you should see the second output. This is what `dietpi-postboot.service` does. Since with Stretch we cannot log to journal + console yet, the service status does not show the output. Will be possible once we drop Stretch support around next summer 😉.

TASSDevon, 2020-06-22:
For reference, the output ;-)

```
root@skyspeaker:~# /boot/dietpi/postboot
mv: cannot stat '/tmp/dietpi-*.log': No such file or directory

 DietPi-Services
─────────────────────────────────────────────────────
 Mode: start

[  OK  ] DietPi-Services | start : avahi-daemon
[  OK  ] DietPi-Services | start : squeezelite
[  OK  ] DietPi-Services | start : cron

 ─────────────────────────────────────────────────────
 DietPi v6.30.0 : 19:19 - Mon 22/06/20
 ─────────────────────────────────────────────────────
 - LAN IP : 192.168.1.237 (wlan0)

 Please hit <return> to login
```

Strange that it say it's booting it. And maybe a different thing but I notice avahi is started, yet when I do anything with sudo I get the following:

```
root@skyspeaker:~# sudo
sudo: unable to resolve host skyspeaker: Name or service not known
```

MichaIng, 2020-06-24:
> sudo: unable to resolve host skyspeaker: Name or service not known

Very strange, how did you set this hostname? Is it added to loopback address (127.0.1.1) via `/etc/hosts`?

MichaIng, 2020-06-26:
Ah there is actually a bluealsa package available on the RPi repo:
```
# apt show bluealsa
Package: bluealsa
Version: 0.13
Priority: optional
Section: sound
Maintainer: Simon Long <simon@raspberrypi.org>
Installed-Size: 244 kB
Depends: libasound2 (>= 1.0.16), libbluetooth3 (>= 4.91), libc6 (>= 2.28), libdbus-1-3 (>= 1.9.14), libglib2.0-0 (>= 2.30.0), libsbc1, bluez
Homepage: https://github.com/Arkq/bluez-alsa
Download-Size: 57.8 kB
APT-Sources: https://archive.raspberrypi.org/debian buster/main armhf Packages
Description: Bluetooth ALSA Audio backend
 Packaging of the bluez-alsa audio backend to add ALSA support to
 BlueZ version 5.
```
So no need to compile oneself and it ships with a default ALSA config, systemd unit and all this. Even available for all other archs, so we could use this and implement Bluetooth audio into `dietpi-config` natively for all SBC/PC models: http://archive.raspberrypi.org/debian/pool/main/b/bluealsa/
udev activation, systemd unit, all is there, very nice!

TASSDevon, 2020-06-27:
Whoops, didn't see your post there.
I'm not sure how I set the hostname, it's been too long. But no real issue.

For me this ticket can be closed, so far the BT connection and squeezelite for Spotify playback with LMS > Squeezelite > Jabra over BT dongle connection --> works great.

MichaIng, 2020-06-27:
I'll keep the issue open as a request/ToDo to add USB Bluetooth support natively to `dietpi-config` when I find time.

About the hostname, could you paste the output of: `cat /etc/hosts` (mask any public IP/domain info in case)
Since your PS1 prompt (console) shows "`root@skyspeaker`", at least the short (first) part of your hostname is "skyspeaker". Since `sudo` shows "`unable to resolve host skyspeaker`", I guess this hostname is missing in `/etc/hosts`. Usually there should be an entry like:
```
127.0.1.1 skyspeaker
```
or
127.0.1.1 skyspeaker.domain.some

TASSDevon, 2020-06-27:
The file doesn't exist...? Would explain a lot.
```
root@skyspeaker:~# cat /etc/host
cat: /etc/host: No such file or directory
```

I'm 100% sure I didn't delete it though, I'll make one with the entry you said.

MichaIng, 2020-06-27:
Jep, replacing `DietPi` with `skyspeaker` should do it.

MichaIng, 2020-07-15:
This resolves the issue that UART-based Bluetooth is attempted to be enabled on non-onboard WiFi/BT RPi models: https://github.com/MichaIng/DietPi/commit/702b9769b55b7f1243ca216ec6c1dfef8c478dc2
So the option now installs and enables generic BT packages/services now, which does not mean that specific drivers might be requires. The [bluez-firmware](https://packages.debian.org/buster/bluez-firmware) adds them for a certain chip, others might be included in kernel, some others are part of the pre-installed [firmware-misc-nonfree](https://packages.debian.org/buster/firmware-misc-nonfree) package. If its still not recognised, then check/follow manufacturer instructions, same for onboard BT and SBC manufacturer wikis/docs.

Bluetooth audio support is a dedicated question and requires bluealsa as shown above. This is handled now here where some more info has been accumulated: #3456

closed 2020-08-31 by MichaIng

Talk2Giuseppe, 2024-12-09:
Nearly 2025 and BT is still elusive on DietPi... Wow!  

I've spent the last 4 days attempting to get DietPi to work on a Raspberry Pi 3B and a Pi 4.  What an absolute waste of time.  I feel like I've been in hell for the last 4 days.  I hope you guys can fix it cause using a Pi as an audio player is a pretty simple project.  But w/o BT, it's worthless.

Moving on to a different variant.  Good luck getting this working!

MichaIng, 2024-12-12:
@Talk2Giuseppe 
Bluetooth is working perfectly fine on RPi in general. So you have a specific issue with the adapter or application. But please open a new issue, as it is unrelated to this >4 years old one.

If you want to use a Bluetooth speaker/audio device, as it requires additional software, please check this issue: https://github.com/MichaIng/DietPi/issues/3456