MichaIng/DietPi #3613: Can not install from repository

Arioch1973, 2020-06-18:
#### Details:
- Date           | Wed Jun  3 19:48:03 CEST 2020
- DietPi version | v6.30.0 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi 3 Model B (armv7l) (ID=3)
- Kernel version | Linux DietPi 4.19.118-v7+ #1311 SMP Mon Apr 27 14:21:24 BST 2                                                                                                                                                             020 armv7l GNU/Linux
- Distro         | buster (ID=5)
- Command        | apt-get -q update
- Exit code      | 100
- Software title | DietPi-Software
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Get:1 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB]
Get:2 https://archive.raspberrypi.org/debian buster InRelease [32.6 kB]
Reading package lists...
E: Release file for http://raspbian.raspberrypi.org/raspbian/dists/buster/InRele                                                                                                                                                             ase is not valid yet (invalid for another 14d 17h 21min 55s). Updates for this r                                                                                                                                                             epository will not be applied.
E: Release file for https://archive.raspberrypi.org/debian/dists/buster/InReleas                                                                                                                                                             e is not valid yet (invalid for another 13d 19h 12min 10s). Updates for this rep                                                                                                                                                             ository will not be applied.
```

MichaIng, 2020-06-18:
Many thanks for your report.

> Date | Wed Jun 3 19:48:03 CEST 2020

It looks your system clock is wrong? Could you try to run: `/boot/dietpi/func/run_ntpd 1`
Aside of that please simply retry `apt update` in a few hours, it might be simply a temporary server issue.

Arioch1973, 2020-06-18:
Thank you, I completely missed that one. And it was what was wrong.

Joulinar, 2020-06-18:
many thanks for feedback. Does it mean issue was fixed and we can close the topic?

Arioch1973, 2020-06-18:
Yes, please do. I am sorry for making a ticket and missing such a simple fault as the date and time.

closed 2020-06-19 by Joulinar