MichaIng/DietPi #3614: Install of NodeJs failed to install dependency libatomic.so.1

larsno, 2020-06-19:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
- Kernel version | `uname -a`
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
- Power supply used | (EG: 5V 1A RAVpower)
- SDcard used | (EG: SanDisk ultra)

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. ...
2. ...

#### Expected behaviour

- ...

#### Actual behaviour

- ...

#### Extra details

- ...

Joulinar, 2020-06-19:
Hi,

I used a fresh clean DietPi on my RPi3B+ and tried to replicate the issue without success. For me Node.js installation was running fine. As well libatomic1 was present on my empty test system right from the beginning.

```
root@DietPi3:~# dpkg -l libatomic1
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name             Version      Architecture Description
+++-================-============-============-=====================================================
ii  libatomic1:armhf 8.3.0-6+rpi1 armhf        support library providing __atomic built-in functions
root@DietPi3:~#
```
```
 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Installing Node.js: javascript runtime

[  OK  ] DietPi-Software | Checking URL: https://raw.githubusercontent.com/taaem/nodejs-linux-installer/master/node-install.sh
--2020-06-19 11:19:18--  https://raw.githubusercontent.com/taaem/nodejs-linux-installer/master/node-install.sh
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 151.101.112.133
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|151.101.112.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 1943 (1.9K) [text/plain]
Saving to: ‘node-install.sh’

node-install.sh                       100%[=========================================================================>]   1.90K  --.-KB/s    in 0.004s

2020-06-19 11:19:18 (513 KB/s) - ‘node-install.sh’ saved [1943/1943]

Node Linux Installer by www.github.com/taaem
Running as Root User
Get Latest Version Number...
Done
Downloading latest stable Version node-v14.4.0-linux-armv7l.tar.gz...
Done
Installing...
Finished installing!
```
```
root@DietPi3:~# node --version
v14.4.0
root@DietPi3:~#
```

MichaIng, 2020-06-19:
Many thanks for your report and solution. From Debian package dependencies it looks like libatomic is required since Node 10 already, or it depends on build options...

Would be nice to have a shared lib dependency list for the pre-compiled Linux binaries but there is none: https://github.com/nodejs/help/wiki/Installation#how-to-install-nodejs-via-binary-archive-on-linux

MichaIng, 2020-06-19:
Found the discussion around that, seems that this dependency is not fully intended. Many PRs were opened to remove it but all closed: https://github.com/nodejs/node/issues/29718

MichaIng, 2020-06-19:
Fixed with: https://github.com/MichaIng/DietPi/commit/95b8e7e33a1aa46baa088258d0c981673908e222
Changelog: https://github.com/MichaIng/DietPi/commit/3b7712d75c11818cf5b9175653f767b85abc5502

closed 2020-06-30 by MichaIng