MichaIng/DietPi #3616: No sound on AudioOut of NanoPi Neo v1.31

dzubybb, 2020-06-20:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=30
G_DIETPI_VERSION_RC=0
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
stretch
- Kernel version | `uname -a`
Linux DietPi 4.14.0 #1 SMP Mon Dec 18 11:43:22 CST 2017 armv7l GNU/Linux
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
NanoPi NEO (armv7l)
- Power supply used | 5v USB
- SDcard used | Kingston 8GB class 10

#### Steps to reproduce
1. Install DietPi
2. Enable Audio
3. Select Card 2 (H3 Audio Codec) [hw:2,0] as default in dietpi-config
4. Test sound by `speaker-test` or Aplay some_wav_file

#### Expected behaviour
- hear sound

#### Actual behaviour
- on aplay i get error :
```
# aplay /mnt/dietpi_userdata/Music/test.wav
Playing WAVE '/mnt/dietpi_userdata/Music/test.wav' : Signed 16 bit Little Endian, Rate 44100 Hz, Stereo
aplay: pcm_write:2011: write error: Input/output error
```
- `speaker-test` seams to work but i don't hear any sound :
```
# speaker-test

speaker-test 1.1.3

Playback device is default
Stream parameters are 48000Hz, S16_LE, 1 channels
Using 16 octaves of pink noise
Rate set to 48000Hz (requested 48000Hz)
Buffer size range from 256 to 262144
Period size range from 128 to 32768
Using max buffer size 262144
Periods = 4
was set period_size = 32768
was set buffer_size = 262144
 0 - Front Left
Time per period = 10.303499
 0 - Front Left
Time per period = 10.239453
 0 - Front Left
```

#### Extra details
- Same file is playing fine on UbuntuCore 16.04 FriendlyArm image out of the box
- in output of `cat /var/lib/alsa/asound.state` on FriendlyArm image i have sections `state.audiocodec`, `state.sndhdmi` and `state.H33582100` which are not present on DietPi
- in output of `cat /var/lib/alsa/asound.state` on DietPi image in section `state.Codec` control `Line Out Playback Switch` have false values, even when i disable mute (if i'm correct that this is the same control) in `alsamixer` on `Line Out` of card `H3 Audio Codec`

Some debug info:
```
# aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: Dummy [Dummy], device 0: Dummy PCM [Dummy PCM]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
[... 405 lines cut ...]
                value.0 0
                value.1 0
                value.2 0
                value.3 0
                value.4 0
                value.5 0
                comment {
                        access read
                        type INTEGER
                        count 6
                        range '0 - 36'
                }
        }
        control.2 {
                iface PCM
                name ELD
                value '0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000'
                comment {
                        access 'read volatile'
                        type BYTES
                        count 128
                }
        }
}
```

dzubybb, 2020-06-20:
[alsa-info.txt](https://github.com/MichaIng/DietPi/files/4807843/alsa-info.txt)

MichaIng, 2020-06-20:
Note that `/var/lib/alsa/asound.state` is only written to on shutdown via `alsa-restore.service` or when running `alsactl store` manually. It is then used on startup to re-apply last state, as well via `alsa-restore.service`.

To reset current states to default: `alsactl init`
Then run `alsamixer` to adjust them, indeed not all sound devices are initiated well to play sound OOTB, possibly FriendlyCore ships a working `/var/lib/alsa/asound.state`, but this can then be achieved via `alsamixer` as well.
Annoying to have all those dummy and loopback audio devices, indeed only card 2 and card 3 (HDMI) are actual output cards.

We'll not apply special changes for this old Stretch image anymore, but might do that for the new Armbian-based image, based on much newer mainline Linux: https://dietpi.com/downloads/images/testing/DietPi_NanoPiNEO-ARMv7-Buster.7z
Probably it results in better default mixer settings as well.

dzubybb, 2020-06-20:
Thanks for quick response ! 

I run `alsactl init`, then enabled `Line Out` but it was still no sound on output. 
Then i realized that maybe muted DAC can have something to do with it, and viola , it worked !
So all along it was muted DAC.... 🤦 

Thanks for help, i'm closing issue.

closed 2020-06-20 by dzubybb