MichaIng/DietPi #3618: Network | MAC address of onboard Ethernet

MichaIng, 2020-06-24:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:

- On most Armbian-based images, the onboard Ethernet MAC address is assigned randomly on every boot.
- On Odroid XU4 we set it fixed via `boot.ini` for all systems.
- Both ist not optimal.

#### Describe the solution you'd like:

- Best default is to assign a new random MAC persistently on first boot via u-boot/cmdline config file.
- Bonus would be to add a `dietpi-config` option to allow changing the MAC (per-interface) or unset it to intentionally have a new random MAC on every boot.

#### Describe alternatives you've considered:

- There is not really one! Having an identical fixed MAC per image leads to collisions when using multiple devices in the same network. Having a rotation MAC breaks persistent DHCP assignments via router.

#### Additional context

- Generic u-boot variable `ethaddr` (`eth1addr`/...): https://github.com/MichaIng/DietPi/issues/2979#issuecomment-648921133
- XU4 `boot.ini` variable: https://github.com/MichaIng/DietPi/blob/aa033a5300141a0c5c1a3cc135cc02ff64e9f1d9/boot_xu4.ini#L8
- How about other Odroids? Do all u-boot boards support `ethaddr`?

dzubybb, 2020-06-29:
@MichaIng 
I got error on `fw_printenv` :
```
Bad CRC, using default environment
bootcmd=run distro_bootcmd
bootdelay=2
baudrate=115200
stdin=serial,cros-ec-keyb,usbkbd
stdout=serial,vidconsole
stderr=serial,vidconsole
ethaddr=00:00:11:22:33:44
eth1addr=00:00:11:22:33:45
eth3addr=00:00:11:22:33:46
eth5addr=00:00:11:22:33:47
ipaddr=1.2.3.4
....
```
I've added `ethaddr 6a:32:0b:30:91:7c` to `/etc/armbianEnv.txt` but it's not working.
I'm going back to `hwaddress ether 6a:32:0b:30:91:7c` in `/etc/network/interfaces`.

MichaIng, 2020-06-29:
Thanks, that is sad but good to know you found another solution. Best would be to apply MAC on early boot (before systemd even starts), but `/etc/network/interfaces` as least should work in a generic way.