MichaIng/DietPi #3619: Wifi adopters are not working in kali linux dual boot

gurukiran4u, 2020-06-25:
i have installed kali linux in my laptop as dual boot, i am unable to find the wifi adopters ,when i type `lspci` command in terminal under network controller i got this **Realtek Semiconductor Co., Ltd. RTL8723DE 802.11b/g/n PCIe Adapter**
but when i use command `iwconfig` there is no wlan0 and it also shows no wireless extensions 
can anyone please help me with this 
here are the screen shots 
[click here](https://drive.google.com/drive/folders/1kKrKO1kThGV1SPk0CuLrC_9vNB3iAy29?usp=sharing)

Joulinar, 2020-06-25:
Hi,

many thanks for your message. Just for clarification. Do you have the issue with DietPi or with Kali?

gurukiran4u, 2020-06-25:
@Joulinar thank you for your reply i am using Kali Linux, the Wifi adopter works fine with Windows 10 but they won't work with Kali Linux

Joulinar, 2020-06-25:
hi, well not sure if your are in the right GitHub. We are supporting DietPi and not Kali 😃

gurukiran4u, 2020-06-25:
@Joulinar can u tell me where can i find the solution please

Joulinar, 2020-06-25:
maybe you can ask your question at Kali Forum https://forums.kali.org/

gurukiran4u, 2020-06-25:
@Joulinar thank you so much

Joulinar, 2020-06-25:
ok I will close the ticket if you agree.

closed 2020-06-25 by Joulinar

gurukiran4u, 2020-06-25:
@Joulinar ok and thank you