MichaIng/DietPi #3622: DietPi-Software | Netdata - update to latest version

athisun, 2020-06-26:
Netdata has announced a binary repository since releasing v1.16.0 on 2019-07-08: https://www.netdata.cloud/blog/release-1.16/#binary-distributions-via-packagecloud

DietPi is currently installing self-compiled packages on Stretch, but they are now available from the new repo. Newer distro versions install from default repositories, but the packages are outdated.

Version available in default repos: [1.12.0-1+deb10u1](https://packages.debian.org/buster/netdata) which is generated from [v1.12.0](https://github.com/netdata/netdata/releases/tag/v1.12.0) [2019-02-06]. 1 year, 18 releases ago!

New repository: https://packagecloud.io/netdata/netdata

Example binaries:
- https://packagecloud.io/netdata/netdata/packages/debian/jessie/netdata_1.23.0_i386.deb
- https://packagecloud.io/netdata/netdata/packages/debian/stretch/netdata_1.23.0_i386.deb
- https://packagecloud.io/netdata/netdata/packages/debian/buster/netdata_1.23.0_i386.deb

There are repository install instructions available [here](https://packagecloud.io/netdata/netdata/install) that include the GPG key and apt config.

athisun, 2020-06-26:
Doh! I can see now that these are only x86 binaries. I wonder if there is a newer source for arm binaries.

athisun, 2020-06-26:
I can see that updates to the default repository are [in progress](https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=952959#20). Closing this issue in favour of the Debian bug tracker issue: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=952959.

closed 2020-06-26 by athisun

MichaIng, 2020-06-26:
Yes I stumbled over this repo as well. Sadly x86 only and also the install is a VERY heavy one, with all extensions that most will never need. Doesn't make sense to install this on a little SBC to monitor it while it overloads the whole thing: https://github.com/MichaIng/DietPi/issues/3062#issuecomment-527151425
Hopefully Debian will backport the new version to Buster as well.