MichaIng/DietPi #3623: DietPi downgrade possibility

danergo, 2020-06-27:
Hi,

I've a quite fresh DietPi install (v6.30.0) on a RPi4B.

If I check the debian version:

```
# cat /etc/debian_version
10.4
```

What is my best option to revert the whole system to debian version 10.3 (other than completely reinstall from scratch?
(I have very strange bluetooth issue, and I believe this is not DietPi's fault, and now I found someone who also has similar problems as me, and he didn't have these with 10.3)

Joulinar, 2020-06-27:
Hi,

many thanks for your question. I guess that's not possible, because DietPi will call `apt update && apt upgrade` during initial setup. Means your system will always be upgrade to latest version. Following link from Debian Wiki will give a clear answer as well.

https://wiki.debian.org/SystemDowngrade

danergo, 2020-06-27:
Thanks for the quick response. Is it possible to download an older DietPi which was based on 10.3? I can drop its packets in the router, so aptget wont succeed, but still i can get a running system?

MichaIng, 2020-06-27:
As Joulinar said, it doesn't matter which image you download, APT packages will be upgraded to latest version on startup. Generally create `dietpi-backup` regularly and revert with this if you face issues.

Debian 10.3 vs 10.4 barely has any meaning. This version string is set by a single package, while there are hundreds of packages dynamically upgraded without having a Debian version string change. So they basically mark a state of minimum package versions that you can expect, but not the exact or maximum package versions.

If you face issues with Bluetooth, check the related log files and service output, e.g.:
```
journalctl -u bluetooth
journalctl -u hciuart
```
Likely it is a state or config issue, not a package/binary issue, but if so, then you can revert the single offending package instead of the whole system.

danergo, 2020-06-27:
Even if I managed to download an older DietPi, it won't be able to start without internet access?

Joulinar, 2020-06-27:
Correct. DietPi first run required internet access to complete setup. It will always port your system to latest version.

danergo, 2020-06-27:
Thanks a lot!

closed 2020-06-27 by danergo