MichaIng/DietPi #3624: Wireguard (RTNETLINK answers: File exists )

repomanz, 2020-06-28:
### Creating a bug report/issue

#### Required Information
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=30
G_DIETPI_VERSION_RC=0
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
Debian 10.4
Linux wireguard 3.16.82+ #1 SMP PREEMPT Tue Mar 17 18:33:26 CET 2020 aarch64 GNU/Linux
Odroid C2 (aarch64)

I saw there may be a work around in a previous closed issue but was not clear to me what the fix/work around is or what it's doing. Hoping for some clarification / direction.

#### Steps to reproduce

1. install openssh and wiregaurd together.  
2. in allowed ip section of the wireguard peer, add the subnet that the device (my odroid) is on for it's own IP address

#### Expected behaviour
1. wireguard should start the iface

#### Actual behaviour
[#] ip link add wg0 type wireguard
[#] wg setconf wg0 /dev/fd/63
[#] ip -4 address add 10.180.100.1/24 dev wg0
[#] ip link set mtu 1420 up dev wg0
[#] ip -4 route add 10.180.50.0/24 dev wg0
RTNETLINK answers: File exists
[#] ip link delete dev wg0

#### additional info:
[Interface]
Address = 10.180.100.1/24
PrivateKey = <removed>
ListenPort = 41820

PostUp = sysctl net.ipv4.conf.%i.forwarding=1 net.ipv4.conf.$(mawk 'NR==3' /run/dietpi/.network).forwarding=1
PostUp = sysctl net.ipv6.conf.$(mawk 'NR==3' /run/dietpi/.network).accept_ra=2
PostUp = sysctl net.ipv6.conf.%i.forwarding=1 net.ipv6.conf.$(mawk 'NR==3' /run/dietpi/.network).forwarding=1
PostUp = iptables -A FORWARD -i %i -j ACCEPT; iptables -t nat -A POSTROUTING -o $(mawk 'NR==3' /run/dietpi/.network) -j MASQUERADE
PostUp = ip6tables -A FORWARD -i %i -j ACCEPT; ip6tables -t nat -A POSTROUTING -o $(mawk 'NR==3' /run/dietpi/.network) -j MASQUERADE
PostDown = iptables -D FORWARD -i %i -j ACCEPT; iptables -t nat -D POSTROUTING -o $(mawk 'NR==3' /run/dietpi/.network) -j MASQUERADE
PostDown = ip6tables -D FORWARD -i %i -j ACCEPT; ip6tables -t nat -D POSTROUTING -o $(mawk 'NR==3' /run/dietpi/.network) -j MASQUERADE

# Client 1
[Peer]
PublicKey = <removed>
AllowedIPs = 10.180.100.2/32, 10.180.50.0/24

Joulinar, 2020-06-28:
Hi,

many thanks for your report. There is no need to add the local subnet 10.180.50.0/24 to client section on `wg0.conf`. Pls remove and restart WireGuard.

repomanz, 2020-06-28:
When doing so, I'm not able to access anything on 10.180.50.0/24 when a wg client is connected

Joulinar, 2020-06-28:
do you mean your client is not able to access anything on the local network? Maybe something wrong on client setting.

repomanz, 2020-06-28:
Hey guys, thanks.  This issue appears to be a dns issue either with routing rules in wg or the client.  I have a dns server parked in the subnet  I can't seem to access.  any tips here?  We can close the issue

Joulinar, 2020-06-28:
Some more information about client configuration as well as DNS server location/IP would be helpful.

repomanz, 2020-06-28:
mobile client:
[Interface]
Address = 10.180.100.2/32
DNS = 10.180.50.5
PrivateKey = <removed>

[Peer]
AllowedIPs = 0.0.0.0/0, ::/0
Endpoint = 10.180.50.10:41820
PublicKey = <removed>

dietpi is running on an odroid c2 with wireguard running.  
- IP address on the dietpi is 10.180.50.10
- DNS server is 10.180.50.5

Joulinar, 2020-06-28:
ok I checked my `wg0-client.conf`. There the interface is set to `/24` and not `/32` as shown on your config.

```
[Interface]
Address = 10.9.0.2/24
```

repomanz, 2020-06-28:
updated my client for /24.  Still not able to resolve dns.  More information the dns server 10.180.50.5 is a pihole instance.  I can ping this IP and others in the 10.180.50.0/24 network but still cant' resolve dns.

Joulinar, 2020-06-28:
ok did you changed the network configuration on the client as well? because just changing them on DietPi doesn't do anything on the client itself. 

on Pihole you could try to set DNS server interface setting to `Listen on all interfaces, permit all origins`

repomanz, 2020-06-28:
so I can have a reference, can you post up your server / client settings (minus the personal info) as well as the network your dns is at?  I'll compare the two.  I don't know enough about the routing in WG above, but does it cover dns requests as part of the routing?

MichaIng, 2020-06-28:
Jep please try the `Listen on all interfaces, permit all origins` setting in Pi-hole which is indeed required.

Another thing about your configs. Please assure that:
- in `wg0.conf` in the `[Peer]` blocks, `AllowedIPs` must only be the exact client IP with /32 mask, e.g.:
    ```
    AllowedIPs = 10.180.100.2/32
    ```
    Since this is what IP (ranges) the server should send through the tunnel to this peer. Adding `10.180.50.0/24` AFAIK leads to your WireGuard server is pinging the connected client instead of Pi-hole or anything else in the local network.

- in the clients config instead you could `AllowedIPs = 10.180.50.0/24` so that all requests to the local network (e.g. Pi-hole on `10.180.50.5`) are sent through the tunnel by the client, or keep `0.0.0.0/0, ::/0` to tunnel all requests (including IPv6).

The initial error `RTNETLINK answers: File exists` has gone? This actually indicates a doubled route, likely due to the `10.180.50.0/24` entry in `wg0.conf`. When your local network connection is established this route is added automatically (since your DietPi has an IP/mask of `10.180.50.10/24` which indicates that `10.180.50.0/24` is the local network you want to connect to. Now due to wg0.conf entry WireGuard wants to set up the same route on wg0 interface, hence the collision.

Joulinar, 2020-06-28:
The routing is not done on port level. It's done on IP level. Therefore, if you are able to reach PiHole, from your WireGuard client, on `http://10.180.50.5/admin` you should be able to resolve DNS as well. For testing purposes, pls disconnect your Mobile device from local network, to ensure your are using VPN tunnel.

My server `/etc/wireguard/wg0.conf`
```
[Interface]
Address = 10.9.0.1/24
PrivateKey = xxx
ListenPort = 51820
PostDown = ip6tables -D FORWARD -i %i -j ACCEPT; ip6tables -t nat -D POSTROUTING -o $(sed -n 3p /run/dietpi/.network) -j MASQUERADE
PostUp = ip6tables -A FORWARD -i %i -j ACCEPT; ip6tables -t nat -A POSTROUTING -o $(sed -n 3p /run/dietpi/.network) -j MASQUERADE
PostUp = sysctl net.ipv6.conf.%i.forwarding=1 net.ipv6.conf.$(sed -n 3p /run/dietpi/.network).forwarding=1
PostUp = sysctl net.ipv6.conf.$(sed -n 3p /run/dietpi/.network).accept_ra=2
PostUp = sysctl net.ipv4.conf.%i.forwarding=1 net.ipv4.conf.$(sed -n 3p /run/dietpi/.network).forwarding=1

PostUp = iptables -A FORWARD -i %i -j ACCEPT; iptables -t nat -A POSTROUTING -o $(sed -n 3p /run/dietpi/.network) -j MASQUERADE
PostDown = iptables -D FORWARD -i %i -j ACCEPT; iptables -t nat -D POSTROUTING -o $(sed -n 3p /run/dietpi/.network) -j MASQUERADE

# Client 1
[Peer]
PublicKey = xxx
AllowedIPs = 10.9.0.2/32

# Client 2
[Peer]
PublicKey = xxx
AllowedIPs = 10.9.0.3/32

# Client 3
[Peer]
PublicKey = xxx
AllowedIPs = 10.9.0.4/32

# Client 4
[Peer]
PublicKey = xxx
AllowedIPs = 10.9.0.5/32
```
One of my clients `/etc/wireguard/wg0-client.conf`
```
[Interface]
Address = 10.9.0.2/24
PrivateKey = xxx

# Comment the following to preserve the clients default DNS server, or force a desired one.
DNS = 10.9.0.1

# Kill switch: Uncomment the following, if the client should stop any network traffic, when disconnected from the VPN server
# NB: This requires "iptables" to be installed, thus will most likely not work on mobile phones.
#PostUp = iptables -I OUTPUT ! -o %i -m mark ! --mark $(wg show %i fwmark) -m addrtype ! --dst-type LOCAL -j REJECT && ip6tables -I OUTPUT ! -o %i -m mark ! --mark $(wg show %i fwmark) -m addrtype ! --dst-type LOCAL -j REJECT
#PreDown = iptables -D OUTPUT ! -o %i -m mark ! --mark $(wg show %i fwmark) -m addrtype ! --dst-type LOCAL -j REJECT && ip6tables -D OUTPUT ! -o %i -m mark ! --mark $(wg show %i fwmark) -m addrtype ! --dst-type LOCAL -j REJECT

[Peer]
PublicKey = DwVUTC0L7n7zSCzu/EuKeYFtW6v5nzKi6XyfNkFy7y8=
# Tunnel all network traffic through the VPN:
#       AllowedIPs = 0.0.0.0/0, ::/0
# Tunnel access to server-side local network only:
#       AllowedIPs = 192.168.0.0/24
# Tunnel access to VPN server only:
#       AllowedIPs = 192.168.0.11/32
AllowedIPs = 0.0.0.0/0, ::/0
Endpoint = MyDDNS.com:51820

# Uncomment the following, if you're behind a NAT and want the connection to be kept alive.
#PersistentKeepalive = 25
```

I have PiHole running on same DietPi device as WireGuard server.

repomanz, 2020-06-28:
correct, the intial issue (subject) was gone as soon as I removed the 10.180.50.0/24 subnet from the allowed ip section.  

here's what I see in pihole logs:
Jun 28 19:05:49 dnsmasq[73565]: query[A] <removed host name> from 10.180.50.10
Jun 28 19:05:49 dnsmasq[73565]: config <removed host name is NODATA-IPv4

Bizarre as it resolves dns then says nodata. 

I'll keep looking into this guys obviously you guys pointed me to the issue I had created.  Appreciate both of you helping.

MichaIng, 2020-06-28:
You set `Listen on all interfaces, permit all origins` setting? Not sure currently how a response for a non-allowed origin looks like, probably NODATA is the answer then without sending the request to upstream DNS. Did you try other hostnames/domains and compared with how it looks like when resolving a request from the DietPi system (or any other directly attached to the local network) instead?

repomanz, 2020-06-29:
yeah, it's strange.  I see the queries being resolved in pihole but on the client the local host names do not resolve. For external host names, like www.google.com resolve and work.

Joulinar, 2020-06-29:
Do you use `AllowedIPs = 0.0.0.0/0, ::/0` on client side?

repomanz, 2020-06-29:
Yes

Joulinar, 2020-06-29:
And the only problem is to resolve local host names? Global internet addresses are working fine? Correct?

repomanz, 2020-06-29:
correct. Really odd behavior.

Joulinar, 2020-06-29:
inside PiHole5 you have a section called `Local DNS Records`. Pls can you check if all your local systems are listed there?

repomanz, 2020-07-02:
Hi Joulinar; apologies for delay.  Looked into pihole and there wasn't any dns information under local.  I ended just adding them manually which seems to have me fixed up.  Closing the issue, thanks for all the help guys!

closed 2020-07-02 by repomanz