MichaIng/DietPi #3625: 4G/LTE USB dongle not automatically working [with manual solution included]

sphh, 2020-06-28:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`:
```
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=30
G_DIETPI_VERSION_RC=0
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
```
- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`: `buster` or `10.4`
- Kernel version | `uname -a`: `Linux DietPi 4.19.118+ #1311 Mon Apr 27 14:16:15 BST 2020 armv6l GNU/Linux`
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3): `1`

#### Additional Information (if applicable)
- Was the software title installed freshly or updated/migrated? freshly installed
- Can this issue be replicated on a fresh installation of DietPi? yes

#### Steps to reproduce

1. I plugged my 4G/LTE USB dongle HUAWEI E8372 into the RPi.
2. I booted from a freshly etched SD card

#### Expected behaviour

- The Internet should come up.

#### Actual behaviour

- No Internet.

#### Extra details

- I know that this dongle works perfectly well with the newest version of `usb-modeswitch`, but this software package was missing. So here are the steps how I got it going:

1. Download those three files to your local computer:
    - http://raspbian.raspberrypi.org/raspbian/pool/main/j/jimtcl/libjim0.79_0.79+dfsg0-2_armhf.deb
    - http://raspbian.raspberrypi.org/raspbian/pool/main/u/usb-modeswitch/usb-modeswitch_2.6.0-2_armhf.deb
    - http://raspbian.raspberrypi.org/raspbian/pool/main/u/usb-modeswitch-data/usb-modeswitch-data_20191128-3_all.deb
2. Mount the SD card on your local computer.
3. Copy the downloaded Debian packages to the SD card:
    - `cp libjim0.79_0.79+dfsg0-2_armhf.deb usb-modeswitch_2.6.0-2_armhf.deb usb-modeswitch-data_20191128-3_all.deb /media/<user>/rootfs/home/dietpi/`
4. Unmount the SD card.
5. Boot the RPi.
6. Log in as root.
7. Go to folder with downloaded files:
    - `cd /home/dietpi`
8. Install the Debian packages:
    - `dpkg -i 
libjim0.79_0.79+dfsg0-2_armhf.deb usb-modeswitch_2.6.0-2_armhf.deb usb-modeswitch-data_20191128-3_all.deb`
9. Adapt the `/etc/network/interfaces` file, because after switching the mode of the USB dongle, the network will appear as `eth1`:
    - Open `etc/network/interfaces` in your editor.
    - Replace `eth0` with `eth1` OR make a new section similar to `eth0`, but with `eth1`.
10. Reboot
11. Starting from now on you will have your "wired" internet connection under `eth1`.

#### Suggestion
Keep these three packages for the initial start-up and add two `eth` sections in the `interfaces` file. If `eth0` is used, these packages can be removed later.

MichaIng, 2020-06-28:
For reference: https://packages.debian.org/buster/usb-modeswitch
- On Windows some USB dongles mount as flash storage first to allow installing the contained drivers, before being identified/plugged as actual network/whatever adapters.
- On Linux the drivers are part of the kernel, `usb-modeswitch` tells the dongle to be attached as network/whatever adapter directly.

Generally as long as there is onboard network available I see no strong reason to ship this package by default so far. But at least we could offer to install it via network options. On the other hand the package is small. It depends a bid how regular this case is. I added it to network rework project and will have a closer look and through after general support for multiple network adapters (switching between them or use concurrently) has been added.

sphh, 2020-06-28:
For me it was a bit of Catch 22: This dongle is my only internet access. "Luckily" I had a usb-modeswitch related problem on the previous Raspbian release (where you had to recompile it), so I already knew what to look for. Before I installed DietPi I had the current Raspbian Buster running and there the dongle was working flawlessly, this also helped to pinpoint the problem.

I would have thought this might happen more often with people switching from landline to mobile internet ... Thanks for looking into it!

MichaIng, 2020-06-28:
Ah, it was an RPi Zero then? Jep indeed if there is no onboard network at all, this is a strong argument. We could add it to all boards where we know that they have no onboard network, but in case of RPi we ship a shared image. However as said I'll think about it.

Many thanks for this request and info in any case! 👍

sphh, 2020-06-28:
Yes, for the RPi Zero, the argument is even stronger! In my case it is a RPi 1 which I wanted to set up as a router: LTE USB dongle to access the Internet, a USB WiFi dongle to serve the WiFi devices at home...

MichaIng, 2020-06-28:
Ah a LTE dongle, jep another good case. Lol I was misreading the package docks and read "LAN" instead of "WAN", so yours is the exact usual use case then. Now I remember this being the case for a mobile internet dongle that I used on Windows notebook in Indonesia a while ago 👍.

Interesting that Linux identifies/declares them as Ethernet interface.

sphh, 2020-06-28:
Yes, never thought about that. Maybe it's, because the interface behaves exactly like an Ethernet cable plugged into a router: The dongle dishes out the IP number and everything else via DHCP, even to the computer it is plugged in (it can also act as a WiFi access point if configured as such).