MichaIng/DietPi #3626: DietPi-Software | Install software without system upgrade

AlexanderShniperson, 2020-06-28:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:
- Yes, when i install software by `dietpi-software` this tool always upgrade system and break compatibility of LIRC

#### Describe the solution you'd like:
- abbility to skip system upgrade and just install needed software

#### Describe alternatives you've considered:
- ...

#### Additional context
- ...

AlexanderShniperson, 2020-06-28:
will be nice if you can share workaround solution to disable upgrade by editing some script file

Joulinar, 2020-06-28:
Hi,

many thanks for your request. Doing `apt update && apt upgrade` is an essential part of `dietpi-soiftware`. It's implemented to ensure an up-to-date system before software installation. I don't thinks it's possible to remove on an easy way. I guess @MichaIng could explain the WHY 😄 

As an alternative, you could requested software yourself/manuell following software provider installation instruction.

MichaIng, 2020-06-28:
Hmm, in theory it would be possible to skip the APT upgrades. Generally we need to know the minimum package versions we can expect, but this is assure via `dietpi-update` doing APT upgrades.

Generally if a specific package version is required, try `apt-mark hold <package>` which should prevent `apt upgrade` from upgrading this particular one package.

Remove or comment this line otherwise, if the the hold mark above does not work for some reason: https://github.com/MichaIng/DietPi/blob/dev/dietpi/dietpi-software#L14840

AlexanderShniperson, 2020-06-28:
ok thank will try comment line, because i don't know what exact dependency broke LIRC

Joulinar, 2020-06-28:
Keep in mind that this would need to be done every time you update DietPi core. Because it will override all DietPi scripts.

MichaIng, 2021-04-29:
Ah, this has been added with DietPi v7.0 already 🙂. Marking as closed.

closed 2021-04-29 by MichaIng