MichaIng/DietPi #3627: mopidy installs Python2.7, which is not needed

sphh, 2020-06-28:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`:
```
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=30
G_DIETPI_VERSION_RC=0
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
```
- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`: `buster` or `10.4`
- Kernel version | `uname -a`: `Linux DietPi 4.19.118+ #1311 Mon Apr 27 14:16:15 BST 2020 armv6l GNU/Linux`
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3): `1`

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud): `mopidy`
- Was the software title installed freshly or updated/migrated? freshly installed

#### Steps to reproduce

1. Install `mopidy` from `dietpi-software`

#### Expected behaviour

- Install `python3`, `python3-pip` and associates.

#### Actual behaviour

- `python2.7` and `python-pip` are installed.

#### Extra details

As it was pointed in other places, Python2.7 is already past its shelf life and `mopidy` runs perfectly fine on `python3`, there is probably no need to install `python2.7`.

This might be due to the line 13106/13111 in PR #3486: Maybe this line should have been replaced by the following line and not be there anymore.

Q: Can I uninstall `python2.7` & Co without any side effect from the command line?

MichaIng, 2020-06-28:
Many thanks for your report. I just checked, it is since we do not separate between Python2 pip and Python3 pip but install both versions when any pip is required. Mopidy on Stretch btw still uses Python2.

I wanted to separate between both Python version in the past. Since Python2.7 support has officially ended, probably we should even ban Python2 completely and remove/disable any software titles that still depend on it, i.e. disable Mopidy on Stretch (existing installs will stay in any case of course). All those software titles are most likely not maintained anymore anyway.

I'll have a closer look on next DietPi version, lets see how many titles would be affected.

MichaIng, 2020-07-06:
I mark this as closed in favour of: #3656

closed 2020-07-06 by MichaIng