MichaIng/DietPi #3628: Open Beta v6.31 | Please help testing and hardening the upcoming release

MichaIng, 2020-06-28:
RC version | v6.31.0
---------- | -------
Changelog | https://github.com/MichaIng/DietPi/blob/beta/CHANGELOG.txt
Code changes | https://github.com/MichaIng/DietPi/compare/master...beta
How to apply | https://github.com/MichaIng/DietPi/blob/master/BRANCH_SYSTEM.md
Release planned | End June/Beginning Juli

#### Important testing cases:
- If you have an RPi2 v1.2 RPi3 or RPi4, you could help test a major RPi-related change of this version by testing our new RPi 64-bit image (alpha stage): https://github.com/MichaIng/DietPi/issues/3570
- If you ever wanted to install WireGuard but it is not available for your device via `dietpi-software`, retry it with this new DietPi version as we added Linux builtin module detection which is e.g. true for new Armbian-based images and probably other Linux 5.X images.
- If you use Firefox, you can now run your own FF sync server: https://dietpi.com/phpbb/viewtopic.php?p=24713#p24713
______________
#### Related/solved issues: https://github.com/MichaIng/DietPi/issues?q=is%3Aissue+milestone%3Av6.31
_______________
### Known issues
- DietPi-Config | Enabling WiFi + Ethernet adapters, both on different subnets, breaks WiFi connection in some cases: https://github.com/MichaIng/DietPi/issues/2103
- RPi | On TigerVNC virtual desktop, LXAppearance hangs on dbus-launch: https://github.com/MichaIng/DietPi/issues/1791
- Odroid C2 | Some WiFi adapters do no work as hotspot: https://github.com/MichaIng/DietPi/issues/1955
- DietPi-Software | Node-RED: Pre-installed modules cannot be updated via web UI: https://github.com/MichaIng/DietPi/issues/2073
- DietPi-Software | Raspimjpeg: With Lighttpd, streaming mjpeg does not work: https://github.com/MichaIng/DietPi/issues/1747
- DietPi-Software | MATE desktop: When logging in as root, desktop items and right-click context menu is missing: https://github.com/MichaIng/DietPi/issues/3160
- DietPi-Software | Sonarr/Mono: With current Mono version 6, import to a file system without UNIX permissions support (exFAT, FAT32/vfat, CIFS mounts and NTFS without "permissions" option) fails, regardless of user/umask mount options: https://github.com/MichaIng/DietPi/issues/3179
- DietPi-Software | Transmission: On Raspbian/Debian Stretch, RAM usage raises unlimited over time: https://github.com/MichaIng/DietPi/issues/2413

Joulinar, 2020-06-28:
@MichaIng anything special to test?

MichaIng, 2020-06-28:
Yes indeed, although I am not sure if/which scenarios are always important to test 🤔. I mean we could go through the most critical things:
- WiFi and Ethernet connection on first run, each with DHCP and with static IP.
- Time sync
- Firstrun setup without any software title selected
- Automated firstrun setup, including pre-v6.31 images which include a dietpi-update (all currently, actually always all when we're in beta phase).
- dietpi-backup + restore
- Network setup after finished firstrun setup, switching DHCP <> static IP, WiFi <> Ethernet
- `dietpi-drive_manager` mounting and unmounting an external drive, check `/etc/fstab` that all entries have been (re)created/preserved correctly.

I'd say the above is what really must work, everything else is secondary, e.g. can be tested on demand when related scripts have been touched (more then just the comments), e.g.: https://github.com/MichaIng/DietPi/pull/3621/files#diff-620191171f47b4763a532e984acd2e8b (skip .meta scripts and .conf content)
Often all change at least a bid, so more effective would it be go through the changelog and verify that larger changes/mentionable fixes do of course work but as well do not imply any undiscussed regressions/downsides: https://github.com/MichaIng/DietPi/blob/beta/CHANGELOG.txt

Probably when touching any parts of the code, we should maintain a list of affected software titles, exact menus/selections or script call in general. I mean `dietpi-config` is huge and it hence does not make sense to test all its menus and options manually, but e.g. with this release the RPi Overscan menu, RPi screen resolution menus and RPi sound card selection have been touched + on x86 AMD display drivers have been added. So one can test these specific menus and related options and have an eye on format, spelling, visual glitches and such, and of course the new options in case.

camdenorrb, 2020-06-29:
Raspberry pi 4

Update logs: https://pastebin.com/WV2SGRtT

After install lscpu run:
```
dietpi@DietPi:~$ lscpu
Architecture:        armv7l
Byte Order:          Little Endian
CPU(s):              4
On-line CPU(s) list: 0-3
Thread(s) per core:  1
Core(s) per socket:  4
Socket(s):           1
Vendor ID:           ARM
Model:               3
Model name:          Cortex-A72
Stepping:            r0p3
CPU max MHz:         1500.0000
CPU min MHz:         600.0000
BogoMIPS:            43.20
Flags:               half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
```

For some reason it's not showing ARM8, is there another way to verify that it's running a 64bit OS?

Possibly-Matt, 2020-06-30:
Having some issues with running `G_CONFIG_INJECT 'DEV_GITBRANCH=' 'DEV_GITBRANCH=beta' /DietPi/dietpi.txt`
Seems like the file /DietPi/dietpi.txt doesn't exist for me. 
Issue seems to be covered here https://github.com/MichaIng/DietPi/pull/3629 by using /boot/dietpi.txt instead :smile: which worked great. 
Since the changes by @Joulinar only got merged into dev and you've linked to the master. 

Other than that the update went smooth and all seems well for my pihole / pivpn, great stuff! 
I shall be retrying to use MineOS as I've spent a while trying to get it working pre-beta but never got great results with it, even without using dietpi's software installer.

MichaIng, 2020-06-30:
Ah yes a remain from DietPi-RAMdisk time: https://github.com/MichaIng/DietPi/commit/f9232db99a995e7640d58e9ce7aa0159d749c856
Please use: `G_CONFIG_INJECT 'DEV_GITBRANCH=' 'DEV_GITBRANCH=beta' /boot/dietpi.txt`

MichaIng, 2020-07-02:
#### New Beta v6.31.1 has been merged: #3639

Joulinar, 2020-07-03:
@MichaIng 
Running some test now. So most probably I will spam you soon. 😉 

1. Tried to run PREP Script towards beta branch
I used `https://raw.githubusercontent.com/MichaIng/DietPi/beta/PREP_SYSTEM_FOR_DIETPI.sh` and selected `BETA` branch

```
[ INFO ] DietPi-PREP | -----------------------------------------------------------------------------------
[  OK  ] DietPi-PREP | Step 2: Downloading and installing DietPi source code
[ INFO ] DietPi-PREP | -----------------------------------------------------------------------------------
[  OK  ] DietPi-PREP | Checking URL: https://github.com/MichaIng/DietPi/archive/beta.tar.gz
[  OK  ] DietPi-PREP | Downloading DietPi sourcecode
[  OK  ] DietPi-PREP | Extracting DietPi sourcecode
[ INFO ] DietPi-PREP | Moving kernel and boot configuration to /boot
[  OK  ] DietPi-PREP | mv DietPi-beta/config.txt /boot/
[  OK  ] DietPi-PREP | mv DietPi-beta/dietpi.txt /boot/
[  OK  ] DietPi-PREP | mv DietPi-beta/README.md /boot/dietpi-README.md
[FAILED] DietPi-PREP | mv DietPi-beta/LICENSE.txt /boot/dietpi-LICENSE.txt
```

MichaIng, 2020-07-03:
Many thanks! Right, the license file has no file ending indeed: https://github.com/MichaIng/DietPi/commit/1cf3b08dae2078a235d6bf0703b75219b88583ac

Joulinar, 2020-07-03:
2. Run unattended installation of Chromium, accept the LICENSE and set auto-start in `dietpi.txt`.
- was working fine and Chromium was starting right after initial setup finished and system rebooted. 
- There was just one small error message at the end but I was not fast enough before my system rebooted. Will try to capture on next run and update this post.

Joulinar, 2020-07-03:
3. I continued playing around with `AUTO_SETUP_ACCEPT_LICENSE`.
- I used the PREP beta script + beta branch to get the `AUTO_SETUP_ACCEPT_LICENSE` in
- Once PREP finished I set `AUTO_SETUP_ACCEPT_LICENSE=1` within `dietpi.txt` and rebooted my RPi
- The RPi automatically starts the initial setup without user login. Means, no chance to connect via SSH anymore because of the running initial setup
`[INFO] DietPi first run setup: Currently running on another screen`
- Without keyboard and monitor, you are lost 🤔 

Is that the intention of the `AUTO_SETUP_ACCEPT_LICENSE` feature? I thought it was just to accept the `LICENSE` and not to start the initial setup. If yes, you would need to set `AUTO_SETUP_AUTOMATED=1` same time. Otherwise you will be blocked on the first window asking for user input if you are running headless.

MichaIng, 2020-07-03:
> The RPi automatically starts the initial setup without user login. Means, no chance to connect via SSH anymore because of the running initial setup

That is strange and not intended without `AUTO_SETUP_AUTOMATED=1`. The license prompt happens already in the login script, hence the automated login already happened then. Only place where we touch it is here: https://github.com/MichaIng/DietPi/blob/dev/rootfs/var/lib/dietpi/services/dietpi-firstboot.bash#L163

Does the following contain anything?
```
ls -Al /etc/systemd/system/getty@tty1.service.d
```

Joulinar, 2020-07-03:
4. Running initial setup, using fresh img 6.30 and beta branch as target
- initial setup was working fine
- system completed first run and finished on `6.31.1 (beta)`
- no auto setup used, standard first run procedure followed

MichaIng, 2020-07-03:
Ah now I see your previous test with Chromium auto install. I guess `/etc/systemd/system/getty@tty1.service.d/dietpi-autostart` was still present from the last attempt. We do not remove this during PREP (only `/etc/systemd/system/getty@tty1.service.d/autostart` known from other images) but we should. I'll also have a look at this Chromium desktop entry creation. Probably this is expected when no desktop is installed, but then we should not attempt to create the symlink 😉.

Joulinar, 2020-07-03:
ok I have a new copy burned to my SD card. Will do an initial run on 6.30 and PREP to beta afterwards. Let's see how it behave.

Joulinar, 2020-07-03:
Ok seems you are right. Looks like some leftover from Chromium installation causing the auto login. On a fresh installation, this did not happen.

MichaIng, 2020-07-04:
Okay any drop-in config with `autologin` inside its name on any local or serial TTY is removed now: https://github.com/MichaIng/DietPi/commit/836b04f9c89850f5df8f1763ac442b622c9b564d
In theory a console on tty5 (getty@tty5.service) or serial console ttyS2 (serial-getty@ttyS2.service) or such could be enabled with autologin override, which would lead to firstrun setup (automated or prompt) on a console that might not even attached, just like in your case but in case of a serial console more difficult to resolve and in case of tty > 1 one would need to know to hit `F<ID>` on keyboard to attach the screen to this TTY.

We cannot check/set for all possible things on a pre-image but for this one we have two cases (previous DietPi image with any autologin autostart option enabled + RPi OS autologin on first boot), so lets cover all possible ones.

closed 2020-07-05 by MichaIng

MichaIng, 2020-07-05:
#### v6.31 has been released 🚀