MichaIng/DietPi #3630: Dietpi-Backup error over samba (unsupported file system)

FrostyMisa, 2020-06-28:
### Creating a bug report/issue

#### Required Information
- DietPi version | `G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=30
G_DIETPI_VERSION_RC=0
G_GITBRANCH='master'
G_GITOWNER='MichaIng' `
- Distro version | `buster`
- Kernel version | `Linux PiZero 4.19.118+ #1311 Mon Apr 27 14:16:15 BST 2020 armv6l GNU/Linux `
- SBC model | `RPi Zero W (armv6l)`
- Power supply used | (EG: 5V 2A it’s official power supply)
- SDcard used | (16 Gb brand I don’t know now)

#### Steps to reproduce

1. I set and install Samba inside Dietpi-Software in File Server line. 
2. I settings everything and I can connect from windows 10, from my phone, everything with. 
3. I do this in another raspberry pi 4. Everything works like on zero. Win 10, phone connect. 
4. Zero - dietpi-launcher - dietpi-drive_manager - add network drive - I add my raspberry 4 here, like I can access everything, in raspberry 4 path = /
5. In zero I go dietpi-backup - location - manual - /mnt/samba/mnt/xxx/zero_backup (samba is mount directory for raspberry 4, second mnt is file in raspberry 4, xxx is flash card inserted in raspberry 4)
6. Get error Has filesystem of: 4, and is not supported. 

#### Expected behaviour
Expecting it will make backup in raspberry 4 flash card, where I store raspberry 4 backup. File system is ok when I make backup directly from raspberry 4.

So I think zero can’t check what file system is in samba directory I write for backup and it’s why I get this error. 

tl;dr- I only want to make backup from one raspberry to another raspberry flash card over samba. But I get error because of unsupported file system.

MichaIng, 2020-06-28:
Many thanks for your report.

Samba cannot be used for backups indeed. The problem is that it does not support UNIX permissions, hence all owner/group/modes info on all files are lost, which would lead to a broken system when restoring.

If you want to backup from DietPi to another UNIX system, consider NFS, which natively supports UNIX permissions as long as on the server-side the share is on a UNIX permissions-supporting file system.

Since rsync has an integrated network transfer feature, I actually aim to integrate this as well, so skip the network share mount part.

FrostyMisa, 2020-06-29:
Thank you very much for your quick response! Now I understand you. And sorry for wasting your time🙃
Looking forward for everything new and improved what you want to bring to Dietpi, great , i really looks to work with it! Easy backup is one of the best things, but all menus with every settings is great as well!
Thank you again 😉

MichaIng, 2020-06-29:
Great, I'll mark this issue as closed then, feel free to reopen if required 🙂.

closed 2020-06-29 by MichaIng