MichaIng/DietPi #3633: upd 6.26.3 to 6.30.0 makes PRI hang after awhile

Thidsa, 2020-06-30:
cat /boot/dietpi/.version
#!/bin/bash
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=26
G_DIETPI_VERSION_RC=3
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

Distro version | 10.4 | Buster
Kernel version | Linux DietPi 4.19.118-v7+ #1311 SMP Mon Apr 27 14:21:24 BST 2020 armv7l GNU/Linux
SBC model | RPI3
Power supply used | Tried several, using 5v 2.5A
SDcard used | Sandisk Ultra 32gb, tried another but same thing happens

To reproduce i just update to 6.30.0

#### Actual behaviour

Rpi hangs after some time. Have not timed it, but within 24 hours

MichaIng, 2020-06-30:
Many thanks for your report.

What exactly hangs?
Does the system fully crash or can you still login via SSH or at least ping it?
Did you monitor CPU temperature and RAM usage?
```
cpu
free -m
```
Which services/software are you using?
```
dietpi-services status
```

And a few general logs to have a look at to check for kernel or service errors:
```
dmesg | tail -10
journalctl
```

Ah, and you could send a bug report so I can have a closer look at those outputs as well:
```
dietpi-bugreport 1
```
and tell me the UUID:
```
echo $G_HW_UUID
```

Thidsa, 2020-06-30:
Im sorry but i just noticed that its not been caused by updating dietpi, but by apt-get update/upgrade...
As dietpi-update do it i was tricked :) It has been like this for months so i got tired of it and didnt update dietpi this time but used apt-get update/upgrade and when it lasted longer than 24 hours i thought i was in the clear.
Thats why i wrote this to say that it happen with dietpi update.
But now its stuck again.
Im pretty sure ive pinged it before without result, but now i can
Everything else networkbased is dead as far as i can see.
I dont know? Should we proceed or?
Everything else you asked about i have to do after a restart..

MichaIng, 2020-07-01:
Probably the OOM killer kills SSH and other network related stuff while ping (ICMP echo) is a kernel feature. Please restart and do the other steps, so we see if something is generally wrong. Monitor CPU temperature and RAM usage to see how it develops and if any of both reaches a critical limit which causes the crash.

Thidsa, 2020-07-01:
[dietpi_michalng.txt](https://github.com/MichaIng/DietPi/files/4858313/dietpi_michalng.txt)

Joulinar, 2020-07-01:
Just some observation: 
- you basically have 2 web server running. `apache2` as well as `lighttpd`. Both are trying to LISTEN on `port 80`. As this is not possible, `lighttpd` is failing. You would need to decide between both or ensure that one of the is running on another port.
- config file of pihole-FTL changed, pls run `systemctl daemon-reload` to reload all service file
- you are using `DHCP` on a system that is running `PiHole`. Best practice is to use `STATIC IP` to ensure that `PiHole` will be reachable on same IP address always
- in general it would be interesting to see how system memory behaves over time. Right now it seems to be fine.

MichaIng, 2020-07-01:
Aside from what Joulinar said, also the Ethernet interface fails to be configured. I guess the most important part (DHCP) is done, so network is up regardless, but there is an ifup script that tries to send a mail which fails:
```
Jul 01 12:43:45 DietPi sh[354]: msmtp: authentication failed (method PLAIN)
Jul 01 12:43:45 DietPi sh[354]: msmtp: server message: 454 4.7.0 Too many login attempts, please try again later. r13sm2074956lf
p.80 - gsmtp
Jul 01 12:43:45 DietPi sh[354]: msmtp: could not send mail (account default from /etc/msmtprc)
Jul 01 12:43:45 DietPi sh[354]: ERROR: Email could not be sent, please check your logs
Jul 01 12:43:45 DietPi sh[354]: run-parts: /etc/network/if-up.d/berryio_email_ip exited with return code 1
Jul 01 12:43:45 DietPi sh[354]: ifup: failed to bring up eth0
Jul 01 12:43:45 DietPi systemd[1]: ifup@eth0.service: Main process exited, code=exited, status=1/FAILURE
```
____________
Since Apache2 is currently the active webserver (because it is started before Lighttpd), to keep things as they are (regarding Pi-hole web UI and in case you have other websites), you could purge Lighttpd and PHP-FPM:
```
apt purge lighttpd php7.3-fpm
apt autopurge
```
Of course you could purge Apache2 instead, but this means a different webserver will serve your sites as before, so I would test first if everything still works fine (or even better):
```
systemctl stop apache2
systemctl start lighttpd
```
Check websites, and if everything works fine/better:
```
apt purge apache2
apt autopurge
```

Thidsa, 2020-07-01:
> * you basically have 2 web server running. `apache2` as well as `lighttpd`. Both are trying to LISTEN on `port 80`. As this is not possible, `lighttpd` is failing. You would need to decide between both or ensure that one of the web server is running on another port like 81.
Yes i have noticed, but since everything has been working ok ive let it be

>pls run `systemctl daemon-reload` to reload all service file
done

> * you are using `DHCP` on a system that is running `PiHole`. Best practice is to use `STATIC IP` to ensure that `PiHole` will be reachable on same IP address always

Ip is enforced through the router, its always the same.

> * in general it would be interesting to see how system memory behaves over time. Right now it seems to be fine.

heh, yes but within a few hours i guess it stops again as usual.
Have a system backup that i use from late december, and that runs as long as i want it too, if i dont update

Thidsa, 2020-07-01:
> Jul 01 12:43:45 DietPi sh[354]: msmtp: authentication failed (method PLAIN)
> Jul 01 12:43:45 DietPi sh[354]: msmtp: server message: 454 4.7.0 Too many login attempts, please try again later. [...]

This also has been like that along time, cant rember if i configured berryio to send mail.
ill see if i can disable the mail thing

> Since Apache2 is currently the active webserver (because it is started before Lighttpd), to keep things as they are (regarding Pi-hole web UI and in case you have other websites), you could purge Lighttpd and PHP-FPM:
> apt purge lighttpd php7.3-fpm
> apt autopurge
 ok ill remove it

MichaIng, 2020-07-01:
> heh, yes but within a few hours i guess it stops again as usual.

Yeah the unused PHP-FPM service takes a few unnecessary resources but not that much, so while it makes sense to fix the mentioned things, I also don't see a reason why those should cause a crash or resolving them should resolve it.

Keep an eye on `htop` and/or `free -m` for memory usage (for the latter "available" memory is what you need to look at) and `dmesg | tail -10` if any yellow or red lines appear (kernel warnings/errors).

> Ip is enforced through the router, its always the same.

Generally as well for network establish performance it makes sense to assign a static IP in such cases, as long as there is not a changing DNS nameserver, changing gateway or similar that you need to regularly re-sync. However there might be routers which do not handle devices well that use an IP inside the DHCP range but to not request the IP via DHCP, not sure. I do it both ways:
- On first boot of a new device, I let it receive an IP via DHCP, which assures that the router as well stores its hostname, allows to reserve the IP etc and do block/reserve the IP.
- Then on the device I switch to static IP by copying IP+mask+gateway that was assigned via DHCP before.
- So the router has a reserved entry but the new device will bring up network much faster and does not need to run a DHCP client.

Thidsa, 2020-07-01:
> Keep an eye on `htop` and/or `free -m` for memory usage (for the latter "available" memory is what you need to look at) and `dmesg | tail -10` if any yellow or red lines appear (kernel warnings/errors).

Anyway i can log this? Cant sit here and watch it until it hangs :)

> * On first boot of a new device, I let it receive an IP via DHCP, which assures that the router as well stores its hostname, allows to reserve the IP etc and do block/reserve the IP.
> * Then on the [...]

Ok fixed

Thidsa, 2020-07-01:
ok ill have something running here. See what happens
Thx so far

MichaIng, 2020-07-01:
```
nohup cat /proc/kmsg > dmesg.log & disown
cat << _EOF_ > ./monitor.bash
#!/bin/bash
while :
do
free -m
cat /sys/class/thermal/thermal_zone0/temp
sleep 5
done
_EOF_
chmod +x ./monitor.bash
nohup ./monitor.bash > monitor.log & disown
```
Not beautiful but the quickest idea I have currently 😄.

Joulinar, 2020-07-01:
if needed I can guide you to a full blown monitoring solution using Telegraf, Grafana and InfluxDB. It might be oversized by you will get everything with nice graphs 🤣

Thidsa, 2020-07-01:
ill see where this brings me first :)

Thidsa, 2020-07-02:
https://nextcloud.thidsa.net:446/s/NFBtJno7H4zi2JS
Ok there the network went poof. Cant connect to anything. 2 of the running ssh sessions went blank but your script and htop is still running. What now?

MichaIng, 2020-07-03:
Hmm, since four shown SSH sessions are still up, it is neither a system crash nor a complete network crash. Were the two disconnected (?) SSH sessions from a different client?
Of course you can quick the running htop and check the content of the produced log files, memory usage is obsolete since shown perfectly fine in htop.

Is it probably Pi-hole and/or router/DHCP server which do not play well together and break connection from local network? From the running SSH session (htop or another one of those), check:
```
ip r
ip a
```
Check if there is a route for the local network and that the IP(s) are as expected. Additionally check for recent kernel/sys logs which might be related: `journalctl` (or since you have rsyslogd running /var/log/syslog /var/log/daemon.log and such).

Generally do you use Pi-hole only for DNS resolving or as DHCP server as well? DHCP disabled in router? Ah actually deriving from above posts the router is the DHCP server, so DHCP functionality in Pi-hole is disabled, right?

Thidsa, 2020-07-03:
Yes i see that now, but thought it was since i could not connect to anything. No same client
Remember this does not happen if i put the backup from late december on sdcard, so something gets updated through apt-get and cause this.
Running ssh's are unusable, see pictures....cant run any command it seems
Yes router is dchp and pihole is not, also this config is the same on the mentioned sdcard backup i revert to when i get tired of this :) Difficult for me this since im not very good with linux things....
Maybe you get a clue from pictures.....actually its one picture
[image]

Ill see if i cant find the logs when i reboot

MichaIng, 2020-07-03:
So SSH connection works fine. Since the commands are not found, I guess your PATH variable is wrong. Can you please paste from a broken SSH session:
```
echo $PATH
```
And fix it for the current session with:
```
export PATH='/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin'
```

Thidsa, 2020-07-03:
have not rebooted yet, still the old ssh's from last reboot.
Patch command get executed, but it did no difference
[image]

MichaIng, 2020-07-03:
Okay then there is something seriously broken. Does the following show something?
```
command -v ls
```
Does calling the binary manually work (does the file exist)?
```
/bin/ls -l
```
Any difference when logging in with root user?

Joulinar, 2020-07-03:
could it be that the RootFS disappears or get corrupted?

Thidsa, 2020-07-03:
[image]
cant login as anything right now, remember...
if i reboot all goes back to normal until next occurence of this..
Dont want to take to much of your time with this either, i can reboot try to gather my configs and do a clean install and see what happens. Or restore my workable backup and use it for as long as i can without using apt-get.

Thidsa, 2020-07-03:
> could it be that the RootFS disappears or get corrupted?

if you ask me i have no clue, dont know what it is :)

Joulinar, 2020-07-03:
I guess something like `lsblk` is not working as well isn't it?

Thidsa, 2020-07-03:
[image]

Joulinar, 2020-07-03:
what about `/bin/lsblk`

Thidsa, 2020-07-03:
dietpi@DietPi:~$ /bin/lsblk
-bash: /bin/lsblk: Input/output error

Joulinar, 2020-07-03:
Found this one fitting your issue. https://unix.stackexchange.com/questions/542554/got-input-output-error-when-execute-any-commands

Thidsa, 2020-07-03:
yes, seems about the same. was only hdd focus there. if there was something with my sdcard, why do the backup image work like it should. I dont have the linux knowledge to fiddle with this.....

Joulinar, 2020-07-03:
Just guessing. Maybe the `apt upgrade` will write new files to bad or damaged areas on your SD card. 

@MichaIng any other ideas?

Thidsa, 2020-07-03:
yes it could be, i have tried another sdcard too brand new. same happens
This time it too about 30 hours for it to be like this. Time varies, but usually around 20 +- maybe. Cant says for sure

Thidsa, 2020-07-03:
i dont know what the script Michal gave me did, maybe i should reboot and try to find the log.

MichaIng, 2020-07-03:
Ah yes totally forgot about the `dmesg.log` and also last CPU temperature would be intersting. Please reboot and have a look at those:
```
cat ~/dmesg.log
tail -50 ~/monitor.log
```

Thidsa, 2020-07-03:
i dont know if it indicate something but the green light (read write) on the pi was steady on for 5 sec off for one steady for 5 etc
ive seen it before when this had happened.

dietpi@DietPi:~$ sudo cat ~/dmesg.log
cat: /proc/kmsg: Permission denied

Joulinar, 2020-07-03:
Login as user `root` or use `sudo`

Thidsa, 2020-07-03:
root@DietPi:~# sudo cat ~/dmesg.log
cat: /root/dmesg.log: No such file or directory

ive looked at the dmesg.log without root and all that was in it was this: cat: /proc/kmsg: Permission denied
was only 35bytes
i musta  executed the script wrong....
just did nano file paste
bash file
it was executed and it didnt release cursor in ssh
please tell me how to run it and ill wait one more time until the pi acts up again

Thidsa, 2020-07-03:
monitor.log was there tho
[monitor.log](https://github.com/MichaIng/DietPi/files/4871244/monitor.log)

Joulinar, 2020-07-03:
Probably it would need to be executed as user `root`. I guess you did as user `dietpi`

Thidsa, 2020-07-03:
tried sudo bash file now and the dmesg.log stays at 0 and the monitor.log file grows, is that right?

Thidsa, 2020-07-03:
maybe and yes, i didnt know :)

MichaIng, 2020-07-03:
Ah makes sense, at least `dmesg` is not permitted for non-root users by default. Yes `cat /proc/kmsg` at best prints all output a single time only, but when running it repeatedly I don't get that again. Instead only new kernel messages are printed as fast as they arrive.

Thidsa, 2020-07-03:
are you interested in looking at it through teamviewer? i know you developers are curious at finding causes :)
just say when, its ready. Atleast tell me how to run the script again

MichaIng, 2020-07-03:
```
sudo nohup cat /proc/kmsg > dmesg.log & disown
```

Thidsa, 2020-07-03:
as root or?

MichaIng, 2020-07-03:
When using `sudo`, root is invoked. Alternative is to login as root and skip `sudo`.

Thidsa, 2020-07-03:
yes, i kinda knew but wasnt sure. ok thx :)
and does it have to be exited when the rpi acts up again?

Thidsa, 2020-07-03:
dietpi@DietPi:~$ sudo nohup cat /proc/kmsg > dmesg.log & disown
[1] 27891
dietpi@DietPi:~$ nohup: ignoring input and redirecting stderr to stdout

running, got this result, later and thx again

Thidsa, 2020-07-04:
if the ssh the command was started from exited, does that kill the nohup cat command?

MichaIng, 2020-07-04:
The way it is started prevents it from being killed when the SSH session closes. It is completely detached from shell.

Thidsa, 2020-07-11:
Strange...its been up over 7 days now. Right now im updating to 6.31.2

Thidsa, 2020-07-20:
Updated and rebooted and its been up for almost 4 new days.....something has changed for the better.
Only thing i did before the 7 days up, was correct the timezone it was 3 hours off. Cant imagine that it had something to do with this getting better.
If it continues like this ill close this.

MichaIng, 2020-07-21:
Great to hear 😃.

Thidsa, 2020-07-27:
Still going strong so ill close this. Thx

closed 2020-07-27 by Thidsa