MichaIng/DietPi #3634: dietpi-arr_to_RAM: command not found

BramvdHoek, 2020-06-30:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=30
G_DIETPI_VERSION_RC=0
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
buster

- Kernel version | `uname -a`
Linux DietPi 4.19.118+ #1311 Mon Apr 27 14:16:15 BST 2020 armv6l GNU/Linux

- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
RPi Zero W (armv6l)

- Power supply used | (EG: 5V 1A RAVpower)
- SDcard used | (EG: SanDisk ultra)

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. Installed the latest version of Dietpi for the RPI,
2. Did a minimal install of Dietpi itself (no software installed yet)

#### Expected behaviour

Wanted to check if I could enable Dietpi log into the ram for Radarr, Sonarr and Jackett

#### Actual behaviour

Command not found

`root@DietPi:~# dietpi-arr_to_RAM
-bash: dietpi-arr_to_RAM: command not found`

`root@DietPi:~# dietpi-arr_to_RAM enable
-bash: dietpi-arr_to_RAM: command not found`

Joulinar, 2020-06-30:
Hi,

many thanks for your report. Pls can you use full path to `dietpi-arr_to_RAM` like

`/boot/dietpi/misc/dietpi-arr_to_RAM enable`

MichaIng, 2020-06-30:
Do you have Sonarr, Radarr or Lidarr installed already (via `dietpi-software`)? Some aliases are only available when the related software is installed.
If you installed them but skipped the reboot, you need to reload globals:
```
. /boot/dietpi/func/dietpi-globals
dietpi-arr_to_RAM
```
If not, use the full bath as mentioned by Joulinar, although without Sonarr/Radarr/Lidarr installed there is no point in using the script 😉.

BramvdHoek, 2020-06-30:
> Hi,
> 
> many thanks for your report. Pls can you use full path to `dietpi-arr_to_RAM` like
> 
> `/boot/dietpi/misc/dietpi-arr_to_RAM enable`

This seems to work. Even without Jackett, Sonarr, Radarr installed!

> Do you have Sonarr, Radarr or Lidarr installed already (via `dietpi-software`)? Some aliases are only available when the related software is installed.
> If you installed them but skipped the reboot, [...]

I know, wanted to reinstall some things and did update to the latest Dietpi version.
Then this command suddenly didn't work.

Maybe update this post:
https://dietpi.com/phpbb/viewtopic.php?f=8&t=5828

Hopefully you guys support Bazarr soon too!

BramvdHoek, 2020-06-30:
* ment lidarr instead of jackett

MichaIng, 2020-06-30:
What you mean by "it does not work"? Nothing has changed recently so it should work as before. Else please paste error messages or which step exactly fails.

BramvdHoek, 2020-07-01:
> What you mean by "it does not work"? Nothing has changed recently so it should work as before. Else please paste error messages or which step exactly fails.
> 
> Ah or due you mean the missing alias/shortcut? As said it is by design (right from the beginning) that it is not used or better not easily accessible as long as there is no sonarr/radarr/lidarr installed. Enabling the service and such would result in an unexpected situation I guess and why would one use it without those installed?

Even with one of them (Radarr) it didn't work. 
The error messages are in my first post. 
It was actually a fresh install of Dietpi, no other software was installed. Even after installing Radarr I couldn't execute the command. 

Anyways:
`/boot/dietpi/misc/dietpi-arr_to_RAM enable` seems to work.

closed 2020-07-01 by BramvdHoek

MichaIng, 2020-07-01:
Did you do a reboot or ran "`. /boot/dietpi/func/dietpi-globals`" after installing Radarr?
This exists? `ls -l $G_FP_DIETPI_USERDATA/radarr`

This directory (G_FP_DIETPI_USERDATA == /mnt/dietpi_userdata) is created during Radarr install for its config and data files. Is checked if this or /sonarr or /lidarr is present before the alias is added during sourcing dietpi-globals: https://github.com/MichaIng/DietPi/blob/dev/dietpi/func/dietpi-globals#L488

MichaIng, 2020-07-01:
Just tested and the alias is available afterwards as expected 🤔:
```
 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Installation completed

[ SUB1 ] DietPi-Services > restart
[  OK  ] DietPi-Services | restart : radarr
[  OK  ] DietPi-Services | restart : cron
2020-07-01 15:54:57 root@VM-Buster:~# . /boot/dietpi/func/dietpi-globals
2020-07-01 17:35:49 root@VM-Buster:~# dietpi-arr_to_RAM
2020-07-01 17:35:52 | DietPi-Arr_to_RAM: [ERROR] Invalid input command (<empty>). Aborting...

Usage: dietpi-arr_to_RAM <command> [<program>]
Available commands:
  1 [<program>]         Link (program) database(s) to RAM
  2 [<program>]         Update (program) database backup(s)
  0 [<program>]         Store (program) database(s) back to disk
  enable                Enable Link to RAM on boot
  disable               Disable Link to RAM on boot
Supported programs:
  <empty>               Apply to all supported and installed programs
  sonarr                Apply to Sonarr database only
  radarr                Apply to Radarr database only
  lidarr                Apply to Lidarr database only
```