MichaIng/DietPi #3636: LED control revert back to original after reboot

FrostyMisa, 2020-06-30:
### Creating a bug report/issue

#### Required Information
- DietPi version | `G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=30
G_DIETPI_VERSION_RC=0
G_GITBRANCH='master'
G_GITOWNER='MichaIng' `
- Distro version | `buster`
- Kernel version | `Linux Pi4 4.19.118-v7l+ #1311 SMP Mon Apr 27 14:26:42 BST 2020 armv7l GNU/Linux`
- SBC model | `RPi 4 Model B (armv7l) `
- Power supply used | original stronger one
- SDcard used | USB SSD Verbatim store 'n' go portable 120GB

#### Additional Information (if applicable)
- Can this issue be replicated on a fresh installation of DietPi? I think yes with USB boot. 
#### Steps to reproduce

1. Clean install, change both LED for different behavior, I like green heartbeat and red panic, but after reboot it’s get back to default. Change again, works, after reboot default. 

#### Expected behaviour

-it will stay the same after reboot. Before with SD card it’s working. When I switch back SD card where I have same system it’s working. Manually set it after reboot it’s change immediately, but after reboot jump back to default. 

#### Actual behaviour

- jump back to default settings

#### Extra details

- maybe some problem with USB boot?

MichaIng, 2020-06-30:
Many thanks for your report. LED triggers should be applied via udev rules on early boot once those are recognised. Can you please do the following after a reboot with LEDs back to default:
```
cat /etc/udev/rules.d/dietpi-led_control.rules
cat /sys/class/leds/led{0,1}/trigger
udevadm trigger -c add /sys/class/leds/led0
udevadm trigger -c add /sys/class/leds/led1
cat /sys/class/leds/led{0,1}/trigger
```

Joulinar, 2020-06-30:
@MichaIng 
I have seen 2 similar reports on the forum
https://dietpi.com/phpbb/viewtopic.php?f=11&t=7830
https://dietpi.com/phpbb/viewtopic.php?f=11&t=7792

FrostyMisa, 2020-07-01:
Here is all you want. Hope it help to you. And many thanks for your time!
I search only here on GitHub, so next time I’ll try forum as well!

```

root@Pi4:~# cat /etc/udev/rules.d/dietpi-led_control.rules
cat: /etc/udev/rules.d/dietpi-led_control.rules: No such file or directory

root@Pi4:~# cat /sys/class/leds/led{0,1}/trigger
none rc-feedback kbd-scrolllock kbd-numlock kbd-capslock kbd-kanalock kbd-shiftlock kbd-altgrlock kbd-ctrllock kbd-altlock kbd-shiftllock kbd-shiftrlock kbd-ctrlllock kbd-ctrlrlock timer oneshot heartbeat backlight gpio cpu cpu0 cpu1 cpu2 cpu3 default-on input panic [mmc0] 
none rc-feedback kbd-scrolllock kbd-numlock kbd-capslock kbd-kanalock kbd-shiftlock kbd-altgrlock kbd-ctrllock kbd-altlock kbd-shiftllock kbd-shiftrlock kbd-ctrlllock kbd-ctrlrlock timer oneshot heartbeat backlight gpio cpu cpu0 cpu1 cpu2 cpu3 [default-on] input panic mmc0 

root@Pi4:~# udevadm trigger -c add /sys/class/leds/led0

root@Pi4:~# udevadm trigger -c add /sys/class/leds/led1

root@Pi4:~# cat /sys/class/leds/led{0,1}/trigger
none rc-feedback kbd-scrolllock kbd-numlock kbd-capslock kbd-kanalock kbd-shiftlock kbd-altgrlock kbd-ctrllock kbd-altlock kbd-shiftllock kbd-shiftrlock kbd-ctrlllock kbd-ctrlrlock timer oneshot heartbeat backlight gpio cpu cpu0 cpu1 cpu2 cpu3 default-on input panic [mmc0] 
none rc-feedback kbd-scrolllock kbd-numlock kbd-capslock kbd-kanalock kbd-shiftlock kbd-altgrlock kbd-ctrllock kbd-altlock kbd-shiftllock kbd-shiftrlock kbd-ctrlllock kbd-ctrlrlock timer oneshot heartbeat backlight gpio cpu cpu0 cpu1 cpu2 cpu3 [default-on] input panic mmc0

```

MichaIng, 2020-07-01:
> root@Pi4:~# cat /etc/udev/rules.d/dietpi-led_control.rules
cat: /etc/udev/rules.d/dietpi-led_control.rules: No such file or directory

That is strange, the script should create this file. Can you please run `dietpi-led_control`, select your desired LED triggers, exit and paste the result of what has been printed to console, especially if there is any error message?

Joulinar, 2020-07-01:
@FrostyMisa 
It's fine to use GitHub. no problem on this. Basically both channels are worked on. 😉

FrostyMisa, 2020-07-01:
Yes, no problem. But I often change LED with launcher - config - display - LED. 

Here is what I see behind when I scroll up:
```

root@Pi4:~# dietpi-launcher
[ INFO ] DietPi-LED_Control | Applying trigger panic to LED led1
[FAILED] DietPi-LED_Control | /etc/udev/rules.d/dietpi-led_control.rules
[ INFO ] DietPi-LED_Control | Applying trigger heartbeat to LED led0
[FAILED] DietPi-LED_Control | /etc/udev/rules.d/dietpi-led_control.rules

```

And cat give me same answer:
```

cat: /etc/udev/rules.d/dietpi-led_control.rules: No such file or directory

```

FrostyMisa, 2020-07-01:
And I don’t want to open another ticket with same hardware and system settings, but before I think I saw what memory split is set, now I can’t see it but I think it’s working, because I set 320 for Kodi and I think default is 16 and with this Kodi won’t run. 
Here is picture what I see:
[image]

MichaIng, 2020-07-01:
Confirmed the has been a bug introduced with v6.29.

Fixed: https://github.com/MichaIng/DietPi/commit/1cada641595ec78421bef2561fdde890b1f88e2e
Changelog: https://github.com/MichaIng/DietPi/commit/aca6f8ae32e7a1f9c1bba69470b8ea21a4dd9c5d

FrostyMisa, 2020-07-01:
Thank you very much for help! Looking forward for Dietpi 6.31 and any other future version, I love your work! Best is backup, because I can experiment with everything and if something get broken, I can restore and try again! No longer copy whole SD card. 
Thanks again for help!
Do I have to open new issue for the ram split problem?

MichaIng, 2020-07-01:
Ah good second find. We check that current memory split with a method that does not work anymore on a >1GiB RAM model: https://github.com/MichaIng/DietPi/blob/dev/dietpi/dietpi-config#L2045
I'll fix it for RPi4.

MichaIng, 2020-07-01:
Fixed: https://github.com/MichaIng/DietPi/commit/1a2d514caa471308d590fac8f605fc213164e43d
Changelog: https://github.com/MichaIng/DietPi/commit/3a0853dbe427c653d93fef51cf78fd601a632915

closed 2020-07-02 by MichaIng