MichaIng/DietPi #3637: Lower VMWare Image Hardware Version

solarmicrobe, 2020-07-01:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:

The current VMWare images are being created with a very recent version of the hardware, version 16 to be exact (https://kb.vmware.com/s/article/1003746). Users who are not on the latest versions of VMWare products may have issues importing and have to make manual changes.

#### Describe the solution you'd like:

Lower the version of the hardware family to alleviate this.  This would depend on how the vmx is getting created.  In ovftool there is a switch "--maxVirtualHardwareVersion=XX" that can be used to set the hardware version.

#### Describe alternatives you've considered:

- ...

#### Additional context
If there are specific features utilized from the hardware family there may be a minimum. https://kb.vmware.com/s/article/2051652

#### Vote for this feature on FeatHub: https://feathub.com/MichaIng/DietPi/

MichaIng, 2020-07-01:
Many thanks about the information and good to know.

The `vmx` is just taken as-is from the VM on which the PREP script ran before: https://github.com/MichaIng/DietPi/wiki/How-to-create-a-DietPi-image-for-VMware#image-export-and-packaging

But just to clarify, do older VMware players (or ESXI) simply deny to open it the the version string is newer than the program, or does it depend on the enabled hardware features only? If so it looks doable to assure that SATA adapters are the newest feature used, meaning compatibility with hardware version 10.

solarmicrobe, 2020-07-13:
I was able to work past it but circling back a bit, I'm trying a few things again today.  It was pretty late when I made this suggestion and I'm not sure it makes as much sense as it did when I was up late and frustrated with my process.  I'll put some information in this ticket but it could probably be closed with no work and might have even been things that were considered already but I'll put my thoughts for posterity or further discussion.

To answer your direct question when I convert the VMX/VMDK to upload to ESXi (for ease of import for my use case to ESXi and storing it on a NAS) it gives me an error and won't let me proceed.
`Line 25: Unsupported hardware family 'vmx-16'.`
So you can see it does carry over some interesting information from the original VM.  This could be solved when creating the VM by choosing a different compatibility.  In ESXi 6.7u3 I can still choose to set this compatibility to "ESXi 5.5 Virtual Machine". In Workstation 15.5 I can do this by choosing a custom new VM and selecting the compatibility of "Workstation 10.x". Both of these would give hardware version 10, per your note above and the links I provided. This would only be a documentation note, would solve my personal immediate need, and might make it easier for others for the same reasons from a download perspective.

On a broader note, the original download is a 7zip of the VMX(machine description) and VMDK(machine disk).  The VMX is specific to VMWare (I believe) and are "harder" to upload than the simple OVF or OVA files (albeit... that depends on your environment and is really minor but skips my normal tooling as I don't enable ssh on ESXi or VCenter).  So to take a step back the higher order suggestion would be to use OVF.  It is an open format supported by both VMWare products and Virtualbox but not Hyper-V (boo Microsoft). I saw some notes that OVF files use a specific version of the VMDK (version 5) that is block compressed so this may do some similar things to 7zip and might take a step out of the end-user process.

I see you also have downloads for Virtualbox and Hyper-V and switching to OVF  might be a worse user experience from your current set up but depends on the amount of work to create all those images (and keep all of the software around to do the authoring).  VMWare and Virtualbox users should just be able to import the OVF and off they go.  Hyper-V is a different story and still needs a conversion.  However, if this simplified authoring you could reduce your tool-set down to Virtualbox and whatever converters you might need.  Several free ones (including from Microsoft) for Hyper-V, another for Qemu, etc.

Thanks for hearing me out.  I'm glad I found DietPi and look forward to using it no matter the conversion.

MichaIng, 2020-07-13:
Btw I tried to import the VirtualBox ova with VMware and it gives me:
```
The import failed because C:\RamDisk\Downloads\DietPi_VirtualBox-x86_64-Buster.ova did not pass OVF specification conformance or virtual hardware compliance checks.

Click Retry to relax OVF specification and virtual hardware compliance checks and try the import again, or click Cancel to cancel the import. If you retry the import, you might not be able to use the virtual machine in VMware Player.
```
So while the format is generally supported, I bet it is the individual (emulated) hardware components that are different between WMware and VirtualBox. Hitting "retry" to see what happens:
- Looks good moreless
- Guest operating system "unknown", likely because on VirtualBox its "Debian (64-bit)" while on WMware its e.g. "Debian 10.x 64-bit", hence versioned. But I'm pretty sure that has zero effect but is only cosmetic + defines defaults when creating a new VM.

Basically it works: https://docs.vmware.com/en/VMware-Workstation-Pro/15.0/com.vmware.ws.using.doc/GUID-DDCBE9C0-0EC9-4D09-8042-18436DA62F7A.html
Probably we can find a way that the exported ova is 100% compatible with all without error message. However I fear that we might then live the one or other limitation, e.g. not being able to contain the `virtio-net` paravirtualized Ethernet adapter for VirtualBox which should perform better then the default Intel PRO/1000 emulation. So to preserve better than default hardware choices for each platform I don't see a way then shipping different archives.

I now manually set `virtualhw.version = "10"` (from default `virtualhw.version = "16"`). In WMware player it shows me `Version: Workstation 10.x` now, but aside of that I don't see downsides. Indeed some feature are gone, e.g. `Virtualize IOMMU` has gone.

Checking again the restricutions:
- Maximum memory `1011 GiB` only, okay that is a real issues.
- Only 64 logical CPUs. Highest CPU count every reported by our users was 48, now (since 2020) 24: https://dietpi.com/survey/#cores
- Regardless if v10 or v16 I can always select USB3.1, not sure about that.
- PCI passthough 6 vs 16.
- Serial ports 4 vs 32
- No NVDIMM, and the mentioned virtualization feature.
- Nothing very serious, although the question is if we get more reports about one of these missing or about too high hardware revision. Its always a compromise.

The feature jump was with hardware v14 (ESXI 6.7), @solarmicrobe this matches your version, hence hardware v14 works OOTB? Choosing this would preserve most of mentioned features.

MichaIng, 2020-07-31:
New image has been uploaded with hardware version 10 (ESXi 5.5 compatible). I mark this as closed.

closed 2020-07-31 by MichaIng