MichaIng/DietPi #3638: iso and vmware corrupted?

nnovaes, 2020-07-01:
### Creating a bug report/issue

#### Actual behaviour

- i've downloaded both the vmware version and the .dmg file from dietpi.com and both show an error. the first can't be initiated on two different systems (esxi and mac os) and the second can't be opened on both windows and mac. i'm not sure if this is an issue with my side but testing the three different OSs in different machines give me these results.

#### Extra details

- ...

MichaIng, 2020-07-01:
Many thanks for your report.

I'll check but actually nothing has changed for a while so those should work as before. Probably a version incompatibility? #3637
Did you compare checksum/hash to assure nothing got corrupted during download or unpacking?

And what is the exact error you face on which step?

> and the .dmg file

You mean `.img` right? But which? For WMware there should be an archive with .vmdk and .vmx file, no .img.
Ah or you mean to have the native PC UEFI iso on a Mac? Not sure if the ever worked: #1755

nnovaes, 2020-07-01:
i just checked the hash for the Dietpi_NativePC-BIOS-x86_64_Buster.img and it matches. however when i open on mac it says no mountable file systems on mac (https://www.evernote.com/l/AV0FChQJIAhI8KttBlEEq0j2rGPSup8BoP8) and on windows it says the file is corrupted.

i'll check the vmware hash when rebuilt  and report back.

nnovaes, 2020-07-01:
about #3637 it shouldn't be an issue since i'm running the latest fusion version

MichaIng, 2020-07-01:
On a Mac from what I know you need to use the UEFI image, not the BIOS image.
And the BIOS image for sure cannot be opened on Windows since it is an ext4 file system with is not supported by Windows.

So I'm still not sure how/where you want to run DietPi actually:
- If you want to run it on WMware, download the VMware image, which contains a vmdk and a vmx. Execute the vmx should start the VM in WMware workstation player, if present, on ESXi it should be as well possible to import it.
- If you want to run DietPi natively on a PC that does not support UEFI (or you simply don't want it), download the DietPi PC BIOS image and use a flasher like [Rufus](https://rufus.ie/) to flash it onto the target drive. It is not the most elegant solution, I know, since you need to in case take out the internal drive and plug it onto another Windows machine to flash the image. Another possibility is to boot the target system via live OS (or even DietPi) from USB stick and download + flash the image from there.
- If you want to run DietPi natively on a UEFI-capable PC or Mac, download the DietPi PC UEFI image and flash it onto a USB stick (again e.g. via [Rufus](https://rufus.ie/), assure that GPT native UEFI is selected). Plug and boot that stick on the target machine, which will allow you to install DietPi on the internal drive.

MichaIng, 2020-08-29:
Marking as closed due to missing reply. Feel free to reopen if required.

closed 2020-08-29 by MichaIng