MichaIng/DietPi #3642: How to prevent sdcard of corruption?

AlexanderShniperson, 2020-07-03:
Hi
Folks tell me please how did you achieved that when the power randomly lost, the sd card is not corrupted?
what settings and steps need to be done for this, I need to configure my raspbian stretch lite image like you did in dietpi.
i need that because your dietpi-stretch image is contain updated kernel 4.19 that do break compatibility with LIRC software.
i checked yesterday that raspbian stretch lite is have kernel 4.14 and LIRC working great there.

MichaIng, 2020-07-03:
To associate with the initial LIRC issue with Linux 4.19: #3481
Your patched packages do not work? https://github.com/AlexanderShniperson/rpi-lirc

> Folks tell me please how did you achieved that when the power randomly lost, the sd card is not corrupted?

There is not really a way to achieve this. When power loss happens during actual SDcard write, corruption is the result, but one can reduce the writes to reduce that change that power loss happens while a write is in progress:
- Having `async` disk writes enabled (default as long a sync mount option has not been added: https://manpages.debian.org/buster/mount/mount.8.en.html#FILESYSTEM-INDEPENDENT_MOUNT_OPTIONS) reduces the change since small disk writes are hold in memory and written in larger blocks together.
- Add the `noatime` + `lazytime` mount options disables access time writes and reduces as well modification/creation timestamp writes.
- Generally reduce SDcard writes where possible, use external drives for persistent software data/configs and tmpfs mounts for temporary files and in case logs.

Ah and enable automated fsck + repair on boot to have smaller corruption fixed immediately before it becomes worse: In `/boot/cmdline.txt` add/assure `fsck.repair=yes`.

AlexanderShniperson, 2020-07-03:
> Your patched packages do not work?

it's stop working and i don't know why, the signal come to IR-receiver, but can not be decoded, that can be seen at  `journalctl -fu lircd`
tired to fight with that problem

closed 2020-09-27 by MichaIng