MichaIng/DietPi #3646: ZeroPi with 6.30 - dietpi-config are not working for ethernet adapter

mattischer, 2020-07-03:
### Creating a bug report/issue

#### Required Information
- DietPi version | 6.30.0
- Distro version | buster
- Kernel version Linux DietPi 5.4.45-sunxi #20.05.3 SMP Wed Jun 10 12:09:20 CEST 2020 armv7l GNU/Linux
- SBC model | ZeroPi (armv7l)
- Power supply used | 5V, 2,4A
- SDcard used | SanDisk ultra

#### Additional Information (if applicable)
hash.txt image:
FILE:	DietPi_NanoPiZeroPi-ARMv7-Buster.img
DATE:	Tue Jun  2 16:32:08 CEST 2020
MD5:	f6698bd534e676db5b4d36db91256a51
SHA1:	ad58ef75e0f934fffbd510c75e9afa5eddceafdd
SHA256:	a73a37c6a5ab7f4bb8e76e695bbf466ea31e1232fb7b0095687b47bf97dd0e27

#### Steps to reproduce
1. Flash image 6.30 for new ZeroPi
2. Try to configure static IP using dietpi-config

#### Expected behaviour
- Comfiguration of static IP is working.

#### Actual behaviour
- Configuration of static IP is not possible

Ethernet    : Not Found | [On] | Disconnected

No Ethernet hardware was detected on your device.

#### Extra details
- Initial setup is completed with one software installation and reboot. ZeroPi is connected with ethernet adapter and dynamic IP.

- Subdirectory /run/dietpi doesn't exists. Following errors are thrown:

/boot/dietpi/func/obtain_network_details: line 143: /run/dietpi/.network: No such file or directory
 ─────────────────────────────────────────────────────
 DietPi v6.30.0 : 21:08 - Fri 07/03/20
 ─────────────────────────────────────────────────────
 - Device model : ZeroPi (armv7l)
 - CPU temp : 41'C : 105'F (Optimal temperature)
 - LAN IP : mawk: cannot open /run/dietpi/.network (No such file or directory)
/boot/dietpi/func/dietpi-banner: line 222: /run/dietpi/.dietpi_motd: No such file or directory

- ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo                                                                                                                                                                                   
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000
    link/ether fa:f8:66:f3:6f:43 brd ff:ff:ff:ff:ff:ff
    inet 192.168.1.227/24 brd 192.168.1.255 scope global dynamic eth0
       valid_lft 40945sec preferred_lft 40945sec
    inet6 fd63:50fa:98da:0:f8f8:66ff:fef3:6f43/64 scope global dynamic mngtmpaddr
       valid_lft forever preferred_lft forever
    inet6 fe80::f8f8:66ff:fef3:6f43/64 scope link
       valid_lft forever preferred_lft forever

MichaIng, 2020-07-03:
Many thanks for your report. Can you please paste the output of the following:
```
cat /etc/tmpfiles.d/dietpi.conf
```

mattischer, 2020-07-03:
`d /run/dietpi 0777`

MichaIng, 2020-07-03:
Okay so actually this file should create `/run/dietpi` automatically at boot.
Can you please paste the following:
```
systemctl status systemd-tmpfiles-setup
journalctl -u systemd-tmpfiles-setup
```

mattischer, 2020-07-03:
It seems that /boot/dietpi.txt is not working too. Static IP address is not active after reboot.

```
root@DietPi:~# systemctl status systemd-tmpfiles-setup
● systemd-tmpfiles-setup.service - Create Volatile Files and Directories
   Loaded: loaded (/lib/systemd/system/systemd-tmpfiles-setup.service; static; vendor preset: enabled)
   Active: active (exited) since Fri 2020-07-03 20:31:23 BST; 1h 52min ago
     Docs: man:tmpfiles.d(5)
           man:systemd-tmpfiles(8)
  Process: 343 ExecStart=/bin/systemd-tmpfiles --create --remove --boot --exclude-prefix=/dev (code=exited, status=73)
 Main PID: 343 (code=exited, status=73)

Warning: Journal has been rotated since unit was started. Log output is incomplete or unavailable.
root@DietPi:~# journalctl -u systemd-tmpfiles-setup
-- Logs begin at Fri 2020-07-03 22:09:05 BST, end at Fri 2020-07-03 22:24:07 BST. --
-- No entries --
```

MichaIng, 2020-07-03:
`/boot/dietpi.txt` network entries are only effective on first boot. Afterwards use `dietpi-config` or `/etc/network/interfaces` directly.

>  Main PID: 343 (code=exited, status=73)

It failed at some stage. The following should work (besides manual creation, but lets test `systemd-tmpfiles`):
```
systemd-tmpfiles --create /etc/tmpfiles.d/dietpi.conf
```
If this succeeds and `/run/dietpi` does exist then, you should check where the service fails at boot. Run the following, when there is no important service currently running on the machine, as this might break services as it removes clears some subdirs in /run/ to prepare for a fresh boot:
```
systemd-tmpfiles --create --remove --boot --exclude-prefix=/dev
```
I hope it produces meaningful output.

mattischer, 2020-07-04:
Now it looks fine for me. / was not owned by root. 

```
root@DietPi:~# systemd-tmpfiles --create /etc/tmpfiles.d/dietpi.conf
Detected unsafe path transition / → /run during canonicalization of /run.
root@DietPi:~# stat /
  File: /
  Size: 1024            Blocks: 2          IO Block: 1024   directory
Device: b301h/45825d    Inode: 2           Links: 21
Access: (0775/drwxrwxr-x)  Uid: ( 1023/ UNKNOWN)   Gid: ( 1023/ UNKNOWN)
Access: 2020-05-14 20:53:20.000000000 +0100
Modify: 2020-05-28 17:06:03.000000000 +0100
Change: 2020-05-28 17:06:03.000000000 +0100
 Birth: -
root@DietPi:~# chown root:root /
root@DietPi:~# systemd-tmpfiles --create /etc/tmpfiles.d/dietpi.conf
root@DietPi:~# stat /run/dietpi
  File: /run/dietpi
  Size: 40              Blocks: 0          IO Block: 4096   directory
Device: 15h/21d Inode: 2203729     Links: 2
Access: (0777/drwxrwxrwx)  Uid: (    0/    root)   Gid: (    0/    root)
Access: 2020-07-04 09:33:06.109260079 +0100
Modify: 2020-07-04 09:33:06.109260079 +0100
Change: 2020-07-04 09:33:06.109260079 +0100
 Birth: -

root@DietPi:~# systemctl status systemd-tmpfiles-setup
● systemd-tmpfiles-setup.service - Create Volatile Files and Directories
   Loaded: loaded (/lib/systemd/system/systemd-tmpfiles-setup.service; static; vendor preset: enabled)
   Active: active (exited) since Sat 2020-07-04 09:37:32 BST; 3min 41s ago
     Docs: man:tmpfiles.d(5)
           man:systemd-tmpfiles(8)
  Process: 345 ExecStart=/bin/systemd-tmpfiles --create --remove --boot --exclude-prefix=/dev (code=exited, status=0/SUCCESS)
 Main PID: 345 (code=exited, status=0/SUCCESS)

Jul 04 09:37:32 DietPi systemd[1]: Starting Create Volatile Files and Directories...
Jul 04 09:37:32 DietPi systemd-tmpfiles[345]: [/usr/lib/tmpfiles.d/fail2ban-tmpfiles.conf:1] Line references path below legacy directory /var/run/, updating /var/run/fail2ban → /run/fail2ba
n; please update the tmpfiles.d/ drop-in file accordingly.
Jul 04 09:37:32 DietPi systemd[1]: Started Create Volatile Files and Directories.
```

Joulinar, 2020-07-04:
Thats strange, how could it be that `RootFS` is not owned by user `roor`?  Even more strange that it was owned by an unknown user (id 1023). Did you placed the SD card into another system?

mattischer, 2020-07-04:
No, but I used my tablet and app EtchDroid to flash SD card. Maybe there something went wrong. Currently I only have my tablet with me. Thanks for your support.

MichaIng, 2020-07-04:
I wanted to check if it is something with the image, but I have no idea how. When opening in archiver, the parent/root dir is not visible. When mounting, the parent/root dir takes owner+mode from the mount point, same when chrooting into it. Not sure if this information is even contained as image content. Probably when flashing the image, the root dir takes the owner+mode of the image file or the mount point of the SDcard?

Probably UID 1023 is the login user on the tablet which flashed the image, or it was the owner of the mount point when the image got created, which would be bad of course. No idea how to check without ZeroPi device here currently 😅, but I guess we would have gotten more reports when the image itself was affected.

mattischer, 2020-07-04:
From my point of view the image is fine and it was the wrong way to flash it to sdcard. I found out that uid 1023 (user media_rw) is used by virtual sdcard daemon on android devices.
But maybe it makes sense to check permissions of rootfs during first boot. Only an idea.

MichaIng, 2020-07-04:
Great find!

I am still wondering how/where this owner is stored on the SDcard and how to check it without actually booting the image. Something for superuser.com 😄.

> But maybe it makes sense to check permissions of rootfs during first boot. Only an idea.

Since it is obviously possible to affect this owner without actually editing the img file, just the way/user it is flashed with (although the exact flash method for sure plays a role as well), this makes sense.

closed 2020-08-27 by MichaIng