MichaIng/DietPi #3647: RootFS is currently Read Only (R/O) mounted.

eunito, 2020-07-04:
#### Details:
- Date           | Sat Jul  4 16:01:16 BST 2020
- DietPi version | v6.30.0 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspberry Pi OS (32-bit) Lite
- Hardware       | RPi B (armv6l) (ID=1)
- Kernel version | Linux DietPi 4.19.118+ #1311 Mon Apr 27 14:16:15 BST 2020 armv6l GNU/Linux
- Distro         | buster (ID=5)
- Command        | mount -v -o rw,remount /
- Exit code      | 32
- Software title | DietPi-Drive_Manager
#### Steps to reproduce:

1. Every time I log into my rpi this happens... Can't write into files, can't use mkdir commands, can't reboot and this keeps happening. not even diet-bugreport works:

[FAILED] DietPi-Bugreport | RootFS is currently Read Only (R/O) mounted. Aborting...
[ INFO ] DietPi-Bugreport | DietPi requires RootFS to be Read/Write (R/W) mounted. Please run "dietpi-drive_manager" to re-enable
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
mount: /: cannot remount /dev/mmcblk0p2 read-write, is write-protected.
```

Joulinar, 2020-07-04:
Hi,

many thanks for your report. Pls can you do following to mount RootFS r/w

1. login as `root`
2. `mount -o remount,rw /`
3. `nano /etc/fstab`
4. add the line for rootFS > `PARTUUID=<YOUR-ID>-02 / ext4 noatime,lazytime,rw 0 1`
5. `PARTUUID` needs to be adjusted, should be same like `/boot` if you use SD card. Be careful and use the right partition `-02`
6. save and exit
7. `reboot`

To get/verify the ID just use `blkid`. You can identify the rootFS by the lable `LABEL="rootfs"`

```
root@DietPi3:~# blkid
/dev/mmcblk0p1: LABEL_FATBOOT="boot" LABEL="boot" UUID="5203-DB74" TYPE="vfat" PARTUUID="6c586e13-01"
/dev/mmcblk0p2: LABEL="rootfs" UUID="2ab3f8e1-7dc6-43f5-b0db-dd5759d51d4e" TYPE="ext4" PARTUUID="6c586e13-02"
/dev/mmcblk0: PTUUID="6c586e13" PTTYPE="dos"
```

It should looks like this afterwards (just with your PARTUUID)
```
#----------------------------------------------------------------
# PHYSICAL DRIVES
#----------------------------------------------------------------
PARTUUID=6c586e13-02 / ext4 noatime,lazytime,rw 0 1
PARTUUID=6c586e13-01 /boot vfat noatime,lazytime,rw 0 2
```

strange that you are hit by this. Maybe a leftover from DietPi v6.29 issue. See #3511

eunito, 2020-07-04:
Forgot to mention that if I turn the power off and on everything is fine as if no problem ever happened. as soon as I get access to it I will report back

MichaIng, 2020-07-04:
RootFS is as well remounted R/O in some cases of file system errors. Please also check `dmesg | tail -10` when facing the issue, and watch out for red lines (= kernel errors).

Joulinar, 2020-07-04:
My first post shouldn't be applicable if everything is working fine after reboot. Maybe kernel massages will give a hint once you are hit by the issue.

eunito, 2020-07-05:
Here we go again! 
Here's the output for _`dmesg --level=err,crit,alert,emerg`_:

```
[    0.841774] vc_vchi_sm_init: failed to open VCHI service (-1)
[    0.841789] [vc_sm_connected_init]: failed to initialize shared memory service
[   15.247777] vc_sm_cma_vchi_init: failed to open VCHI service (-1)
[   15.247794] [vc_sm_connected_init]: failed to initialize shared memory service
```

Can't reboot, dmesg | less (less is not recognized when this happens), etc... only hard reset removing the power plug get's it fixed...

Joulinar, 2020-07-05:
ok these kernel massages can be ignored as they are not relevant for the issue. 

can you do `mount -o remount,rw /` and check if your get RootFS back into r/w mode

MichaIng, 2020-07-05:
`less` is indeed not necessarily installed 😉. Or use `pager` which is always linked to the currently installed default pager.

Just to be sure, directly after boot, the SD card is fully writeable, i.e. you can do something like `echo content > /testfile`? And then after a while it is not possible anymore but `dmesg` doesn't show any related error message? Does `journalctl -r` show any message instead (`-r` leads to newest messages are at the top)?

eunito, 2020-07-08:
Curious as after the latest update it never happened. Will reopen if occurs again!

closed 2020-07-08 by eunito