MichaIng/DietPi #3650: DietPi-Software | Node.js: teeam installer now completely broken

MichaIng, 2020-07-05:
https://github.com/taaem/nodejs-linux-installer/blob/master/node-install.sh#L13

This curl check is now doomed to fail since nodejs.org has HTTPS redirect enabled (now) which makes curl output being empty without `-L` option.

My PR fixes this by using HTTPS, however the `-L` open additionally makes sense as file structure might change.

For now I'll switch to my fork, probably integrate this completely into dietpi-software, if taaem does not maintain it anymore.

MichaIng, 2020-07-05:
MOTD live patch is up:
```
[[ -w '/boot/dietpi/dietpi-software' ]] && grep -q taaem /boot/dietpi/dietpi-software && sed -i '/taaem/{s/taaem/MichaIng/;q}' /boot/dietpi/dietpi-software
```

marius86ar, 2020-07-05:
not working

MichaIng, 2020-07-05:
Bug in my MOTD hack, please fix with:
```
curl -sSfL https://raw.githubusercontent.com/MichaIng/DietPi/master/dietpi/dietpi-software > /boot/dietpi/dietpi-software
sed -i 's/taaem/MichaIng/' /boot/dietpi/dietpi-software
```

MichaIng, 2020-07-09:
I'll mark this as closed. Own script works fine, I'll give taaem some time to react before merging code into `dietpi-software`.

closed 2020-07-09 by MichaIng