MichaIng/DietPi #3652: Dietpi-Software didn’t open

FrostyMisa, 2020-07-05:
### Creating a bug report/issue

#### Required Information
- DietPi version | `G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=31
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'`
- Distro version | `buster`
- Kernel version | `Linux Pi4 4.19.118-v7l+ #1311 SMP Mon Apr 27 14:26:42 BST 2020 armv7l GNU/Linux`
- SBC model | `RPi 4 Model B (armv7l)`
- Power supply used | Original 5,1V 3A
- SDcard used | (USB SSD Verbatim Store 'n' Go 120GB)

#### Steps to reproduce

```

root@Pi4:~# dietpi-software
/boot/dietpi/dietpi-software: line 6594: syntax error: unexpected end of file

```

#### Expected behaviour

Command will run Dietpi software

#### Actual behaviour

Syntax error 

#### Extra details

Update to v6.31.2, reboot and this happened

MichaIng, 2020-07-05:
Bug in my MOTD hack, please fix with:
```
curl -sSfL https://raw.githubusercontent.com/MichaIng/DietPi/master/dietpi/dietpi-software > /boot/dietpi/dietpi-software
sed -i 's/taaem/MichaIng/' /boot/dietpi/dietpi-software
```

FrostyMisa, 2020-07-05:
Good job, it’s working again 🙃
You are the best!

MichaIng, 2020-07-05:
Please also remove the stored MOTD to refresh on next banner print: `rm /run/dietpi/.dietpi-motd`

marius86ar, 2020-07-06:
working now

FrostyMisa, 2020-07-06:
```

root@Pi4:~# rm /run/dietpi/.dietpi-motd
rm: cannot remove '/run/dietpi/.dietpi-motd': No such file or directory

```

MichaIng, 2020-07-06:
All fine if the file is not present 🙂.

MichaIng, 2020-07-08:
I mark this as closed, as there shouldn't be any more effected users.

closed 2020-07-08 by MichaIng