MichaIng/DietPi #3655: DietPi Upgrade to v6.30.1 Failed

mprince28117, 2020-07-06:
#### Details:
- Date           | Mon Jul  6 10:11:45 EDT 2020
- Bug report     | ed987da7-8247-4ac6-ac0e-f58a33c89e02
- DietPi version | v6.30.0 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi 3 Model B+ (armv7l) (ID=3)
- Kernel version | Linux DIETPI 4.19.42-v7+ #1219 SMP Tue May 14 21:20:58 BST 2019 armv7l GNU/Linux
- Distro         | stretch (ID=4,RASPBIAN=1)
- Command        | apt-get -qq --allow-change-held-packages install ./36.deb
- Exit code      | 100
- Software title | DietPi-Software
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
E: Unable to correct problems, you have held broken packages.
```

Joulinar, 2020-07-06:
Hi,

many thanks for your report. Looks like you have not finished package installations from the past. The error message is created by `dietpi-software`, pointing to an `squeezelite` (software ID 36) installation. Are you sure this is happening during `dietpi-update`? Pls can you run `apt update && apt upgrade`

mprince28117, 2020-07-06:
Yes, I will.  I first encountered this error during a dietpi-upgrade command.  I had another sd card with the image installed and could not complete the upgrade there either.  I've gone back and loaded an "original" image from the dietpi forum.  Let me try loading that SD card and see if I can upgrade to the latest version from there.

MichaIng, 2020-07-06:
> Are you sure this is happening during dietpi-update?

During dietpi-update we reinstall/upgrade SqueezeLite, so this is thinkable. Lets check the full output:
```
cd /tmp
wget https://dietpi.com/downloads/binaries/stretch/squeezelite_armv7l.deb
apt install ./squeezelite_armv7l.deb
```

Joulinar, 2020-07-06:
@MichaIng 

> During dietpi-update we reinstall/upgrade SqueezeLite, so this is thinkable. Lets check the full output:

But than I would expect the error message above created bei `dietpi-update`. But it's stating `dietpi-software`

`Software title | DietPi-Software`

MichaIng, 2020-07-06:
We do software updates through dietpi-software reinstalls. Would be a huge overhead to add the whole procedure to the updater patch file.

The prints to not contain parent script names, although this would be actually nice, basically an error stack trace.

Joulinar, 2020-07-06:
ok understood.

mprince28117, 2020-07-06:
Apologies for slow response.  My pi stopped responding, and I am formatting and re-imaging an SD card so that I can carry forward with this.  It will have whatever version of dietpi is in the image from the dietpi website.

MichaIng, 2020-07-06:
On first boot, it will be auto-updated to v6.32.1, so when you install SqueezeLite it will be the new version right from the start. However if there were some general package incompatibilities, the error message might look the same, like the suggested debug steps: https://github.com/MichaIng/DietPi/issues/3655#issuecomment-654272427

I'll check in on my Pi as well now.

MichaIng, 2020-07-06:
Found it:
```
The following packages have unmet dependencies:
 squeezelite : Depends: libmpg123-0 (>= 1.23.8-1+b1) but 1.23.8-1 is to be installed
E: Unable to correct problems, you have held broken packages.
```
This is the problem when building ARMv7 packages on Debian, which are pulled on ARMv7 RPi models. 99,9% match, but this `+b1` version was never ported over to Raspbian:
- http://raspbian.raspberrypi.org/raspbian/pool/main/m/mpg123/
- https://deb.debian.org/debian/pool/main/m/mpg123/
- https://packages.debian.org/stretch/libmpg123-0

I'll remove the part from the dependency version string, hang on.

MichaIng, 2020-07-06:
Fixed it, package repacked, install/update should now succeed.

mprince28117, 2020-07-06:
Excellent, many thanks.  I will boot up my pi in a bit (SD still formatting) and make sure everything comes up OK.

mprince28117, 2020-07-06:
Was able to install upgrade without errors, now to set everything up again as I had to reload the image...:)

closed 2020-07-06 by mprince28117